ev3dev-lang-java/ev3dev-lang-java #483: mindsensors MotorMultiplexer

aivars-sablis, 2018-04-25:
Hi,

I am exploring a way to add mindsensors MotorMultiplexer [http://www.mindsensors.com/ev3-and-nxt/21-multiplexer-for-nxtev3-motors](url) to ev3dev-lang-java. It has hardware driver for ev3dev OS, so motors on multiplexer are initialized in the filesystem. Therefore I suspect this is mostly port addressing issue. But maybe any general advise how to address this in the best way?

jabrena, 2018-04-25:
Good morning @aivars-sablis,

I reviewed the technical documentation about the EV3Dev support and it seems that it is possible:
http://docs.ev3dev.org/projects/lego-linux-drivers/en/ev3dev-jessie/muxs.html

We should begin the analysis executing some statements from the terminal and later, try to automate in a Java class.

In few hours, I will post some ideas.

aivars-sablis, 2018-04-25:
Yes, there is EV3Dev support for this multiplexer. What we get are two tacho-motors in /sys/class/ with addresses in2:i2c3:M1 and in2:i2c3:M2 . So it is possible to control these motors more or less in the same way (though, by documentation some functions is limited). However, I try to understand how I could make assigning those motors in java through MotorPort and respective classes.

aivars-sablis, 2018-04-25:
Yes, except it is the NxtMMX driver that loads two [tacho-motor] class devices. So these ports are already visible in /sys/class/lego-ports and motors are in /sys/class/tacho-motors. Therefore, I suspect rest of process to use this motors could be as usual if we can add additional port addresses support. 

P.S. I can provide additional information from Terminal, if needed.

jabrena, 2018-04-25:
Yes please send me all calls by the terminal.
I have an idea about the solution. We could create a new Class for the new ports.

jabrena, 2018-04-25:
If you observe the UML Class diagram:

[image]

All motors  inherit from class `BaseRegulatedMotor`, example:

``` java
    @Test
    public void forwardTest() throws Exception {

        final FakeBattery fakeBattery = new FakeBattery(EV3DevPlatform.EV3BRICK);
        final FakeEV3LargeRegulatedMotor fakeMotor = new FakeEV3LargeRegulatedMotor(EV3DevPlatform.EV3BRICK);

        EV3LargeRegulatedMotor motor = new EV3LargeRegulatedMotor(MotorPort.A);
        motor.forward();
        motor.stop();
    }
```

For a tactical solution, if you duplicate the class BaseRegulatedMotor

``` java
    /**
     * Constructor
     * @param motorPort motor port
     * @param moveP moveP
     * @param moveI moveI
     * @param moveD moveD
     * @param holdP holdP
     * @param holdI holdI
     * @param holdD holdD
     * @param offset offset
     * @param maxSpeed maxSpeed
     */
    public BaseRegulatedMotor(final Port motorPort, float moveP, float moveI, float moveD,
                              float holdP, float holdI, float holdD, int offset, int maxSpeed) {

        List<RegulatedMotorListener> list = new ArrayList<>();
        listenerList = Collections.synchronizedList(list);

        if(log.isInfoEnabled())
            log.info("Configuring motor connected on Port: {}", motorPort.getName());

        MAX_SPEED_AT_9V = maxSpeed;
        final String port = EV3DevPlatforms.getMotorPort(motorPort);

        if(log.isDebugEnabled())
            log.debug("Detecting motor on port: {}", port);
        this.detect(LEGO_PORT, port);
        if(log.isDebugEnabled())
            log.debug("Setting port in mode: {}", TACHO_MOTOR);
        this.setStringAttribute(MODE, TACHO_MOTOR);
        Delay.msDelay(500);
        this.detect(TACHO_MOTOR, port);
        Delay.msDelay(500);
        this.setStringAttribute(COMMAND, RESET);
        if(log.isDebugEnabled())
            log.debug("Motor ready to use on Port: {}",motorPort.getName());
    }
```

and we change this line:

```
 final String port = EV3DevPlatforms.getMotorPort(motorPort);
```

With a new internal method, we could solve problem translating:

- Motor.A -> in2:i2c3:M1
- Motor.A -> in2:i2c3:M2
- Motor.A -> in2:i2c3:M1
- Motor.A -> in2:i2c3:M2

The new class, has 2 ports, the native port: in1, in2, in3 & in4 and the connection in the Multiplexer.

We could refine the solution once you have the class running.
I would like to help you.

In the following days, I will review how LeJOS solved this issue in one example and later, we will try to adapt the solution.

Juan Antonio

jabrena, 2018-04-29:
Any progress with your local tests?

jabrena, 2019-08-18:
No feedback

closed 2019-08-18 by jabrena