ev3dev-lang-java/ev3dev-lang-java #489: Build OpenJDK 10 for Debian Jessie

jabrena, 2018-05-02:
Many users continue installing `Debian Jessie` because it is the `EV3Dev` stable image.

It could be interesting to have a build of `OpenJDK 10` for Debian Jessie.

JakubVanek, 2018-05-03:
More details in ev3dev-lang-java/openjdk-ev3#23

jabrena, 2018-05-05:
Hi @JakubVanek, 
I am going to test if compilation for `Debian Stretch` is compatible with `Debian Jessie`.
Crossing fingers.

jabrena, 2018-05-05:
In a preliminar way, it is working:

```
robot@ev3dev:/opt/jri-ev3/bin$ ./java -version
openjdk version "10" 2018-03-20
OpenJDK Runtime Environment (build 10+46-ev3)
OpenJDK Client VM (build 10+46-ev3, mixed mode)
``` 

I will do more serious tests

jabrena, 2018-05-05:
``` 
# Run complete. Total time: 00:23:32

Benchmark                                       Mode  Cnt        Score   Error  Units
DynamicLanguages.groovy                        thrpt             0.097          ops/s
ExceptionsBenchMark.throwRuntimeException      thrpt           743.047          ops/s
GeneratePoiWorkBook.generatePoiXLSWorkBook     thrpt             2.346          ops/s
GeneratePoiWorkBook.generatePoiXLSXWorkBook    thrpt             0.088          ops/s
GeneratePrimeNumbersWithForLoop.generatePrime  thrpt          2493.930          ops/s
Jackson.jsonMashal                             thrpt             2.373          ops/s
Jackson.jsonUnmashal                           thrpt             1.925          ops/s
Jackson.xmlMashalJackson                       thrpt             1.167          ops/s
Jackson.xmlMashalJacksonStatic                 thrpt            59.888          ops/s
Jackson.xmlUnmashalJackson                     thrpt             0.546          ops/s
Jackson.xmlUnmashalJacksonStatic               thrpt            12.139          ops/s
MicroBenchMarks.arrayListForeach               thrpt         89920.757          ops/s
MicroBenchMarks.arrayListIterator              thrpt        122118.860          ops/s
MicroBenchMarks.calculateLogPi                 thrpt         61120.985          ops/s
MicroBenchMarks.calculateLogPiCodeElimination  thrpt         61750.151          ops/s
MicroBenchMarks.deadCodeElimination            thrpt       8581360.870          ops/s
MicroBenchMarks.loopThatCanBeUnrolled          thrpt          1208.797          ops/s
MicroBenchMarks.manualUnrollOfLoop             thrpt          1218.728          ops/s
MicroBenchMarks.noDeadCodeElimination          thrpt         40636.199          ops/s
Reflection.doReflection                        thrpt          2970.825          ops/s
Reflection.withoutReflection                   thrpt       3267992.990          ops/s
RegularExpressions.validateEmail               thrpt            27.185          ops/s
```

jabrena, 2018-05-05:
Certified current OpenJDK 10 build on Debian Jessie without any problem.

Any comment @JakubVanek ?

jabrena, 2018-05-23:
I reviewed the installer and it doesn´t have any check with Stretch, so it is not necessary to change any line from Develop branch.
https://github.com/ev3dev-lang-java/installer/blob/develop/modules/java.sh