ev3dev-lang-java/ev3dev-lang-java #494: new display driver in kernel 4.14.x

jabrena, 2018-05-04:
new ev3dev kernel on ev3dev-stretch that updates from 4.9.x to 4.14.x. This comes with a new display driver for the EV3 from the mainline kernel.

The new driver still provides a fbdev device (/dev/fb0) however, the memory layout is different. Because of limitations in the mainline kernel, there is no monochome. Instead it uses a 32-bit XRGB pixel format.

Related issue:
https://github.com/ev3dev/ev3dev-lang-python/issues/455

jabrena, 2018-05-04:
hi @pambrose,

Do you think that the upgrade is complex?

Juan Antonio

jabrena, 2018-05-04:
Ohh, it is a pity but in the other hand, It is nice that you moved to ROS.
I am working in ROS too but I go a bit slow.

If you like, in few months, we could collaborate.
Do you like the idea?

pambrose, 2018-05-04:
Sure!

JakubVanek, 2018-05-04:
Hmm, we might want to implement a DRM client or use the GRX 3 library, which might be capable of that.

jabrena, 2018-05-04:
Hi @JakubVanek, i didn´t say you anything because you are busy in `JVM` side, but it could be lovely.

For Debian Jessie, we have the solution but for Stretch, GRX could the final solution.

Can you do a `POC` to integrate with `GRX`?
GRX library is packaged with Debian Stretch so we could link with that library using JNI.

jabrena, 2018-05-19:
Moving out of the release

jabrena, 2018-08-05:
Related question:
https://github.com/ev3dev/ev3dev/issues/1114

jabrena, 2018-08-19:
The initial tests failed:

```
ev3dev#6|Caused by: java.lang.RuntimeException: Unable to draw the LCD
ev3dev#6|	at ev3dev.utils.Sysfs.writeBytes(Sysfs.java:132)
ev3dev#6|	at ev3dev.actuators.LCD.flush(LCD.java:135)
ev3dev#6|	at ev3dev.actuators.LCD.refresh(LCD.java:297)
ev3dev#6|	at ev3dev.actuators.LCD.init(LCD.java:123)
ev3dev#6|	at ev3dev.actuators.LCD.<init>(LCD.java:50)
ev3dev#6|	at ev3dev.actuators.LCD.getInstance(LCD.java:40)
ev3dev#6|	at example.LCDDrawLinesTest.<clinit>(LCDDrawLinesTest.java:9)
ev3dev#6|Caused by: java.io.IOException: File too large
ev3dev#6|	at java.base/java.io.FileOutputStream.writeBytes(Native Method)
ev3dev#6|	at java.base/java.io.FileOutputStream.write(Unknown Source)
ev3dev#6|	at java.base/java.io.DataOutputStream.write(Unknown Source)
ev3dev#6|	at java.base/java.io.FilterOutputStream.write(Unknown Source)
ev3dev#6|	at ev3dev.utils.Sysfs.writeBytes(Sysfs.java:128)
ev3dev#6|	... 6 more
```

JakubVanek, 2018-08-19:
What will print this command?
```sh
sudo cat /dev/fb0 | wc -c
```

JakubVanek, 2018-08-19:
Example on my computer and what is an expected result:
```
$ sudo cat /dev/fb0 | wc -c
4096000
# which is equal to 1280 pixels wide × 800 pixels tall × 4 bytes per pixel
```

jabrena, 2018-08-19:
and now with OpenJDK 11 JRI:

``` 
robot@ev3dev:~$ java -version
openjdk version "11" 2018-09-25
OpenJDK Runtime Environment (build 11+26-ev3)
OpenJDK Client VM (build 11+26-ev3, mixed mode)
``` 

I checked the display behaviour and the program clean the display but later, the program crash.

In relation to your question, the answer is:

``` 
robot@ev3dev:~$ sudo cat /dev/fb0 | wc -c
6144
```

JakubVanek, 2018-08-19:
OK, that is a surprise. This really does not look like XRGB 32bit framebuffer.
If we divide it by 128, we get 48. So, it looks like we take the original 24 byte stride and multiply it by two.
Could you please send me the output of fbset command?
```sh
sudo apt-get install fbset
sudo fbset -fb /dev/fb0 -i
```

JakubVanek, 2018-08-19:
Maybe this is the direct 2bit representation from the display, but that would be quite strange.

jabrena, 2018-08-19:
This is an Unknown world... everything is possible :p

jabrena, 2018-08-20:
```
robot@ev3dev:~$ sudo apt-get install fbset
[sudo] password for robot: 
Reading package lists... Done
Building dependency tree        
Reading state information... Done
``` 

Tomorrow, I will see the results.
The brick seem a bit zombie.
If the process don´t finish, I will install the package manually.
https://packages.debian.org/stretch/admin/fbset

jabrena, 2018-08-20:
Output:

```
robot@ev3dev:~$ sudo fbset -fb /dev/fb0 -i
[sudo] password for robot: 

mode "178x128"
    geometry 178 128 178 128 1
    timings 0 0 0 0 0 0 0
    grayscale true
    rgba 1/0,1/0,1/0,0/0
endmode

Frame buffer device information:
    Name        : LEGO EV3 LCD
    Address     : 0xc3a9a000
    Size        : 6144
    Type        : PACKED PIXELS
    Visual      : MONO10
    XPanStep    : 0
    YPanStep    : 0
    YWrapStep   : 0
    LineLength  : 24
    Accelerator : No
```

JakubVanek, 2018-08-20:
Hi, have you got the new kernel installed? This looks like the old framebuffer. I have to implement a different detection strategy, as Stretch has non-4.14 kernels.

Jakub

jabrena, 2018-08-20:
I am using Debian Stretch for the tests but maybe it is a old Snapshot.
To not have any doubt, I will install last image:
https://www.ev3dev.org/downloads/
https://github.com/ev3dev/ev3dev/releases/download/ev3dev-stretch-2018-08-06/ev3dev-stretch-ev3-generic-2018-08-06.zip

jabrena, 2018-08-20:
``` java
public class LCDDrawLinesTest {

    public static GraphicsLCD lcd = LCD.getInstance();

    public static void main(final String[] args){

        System.out.println("EV3 LCD Example");

        clear();
        lcd.setColor(Color.BLACK);
        lcd.drawLine(0, 0, 50, 50);
        lcd.drawLine(0, 0, 30, 60);
        lcd.refresh();

        Delay.msDelay(5000);
    }

    public static void clear(){
        lcd.setColor(Color.WHITE);
        lcd.fillRect(0,0, lcd.getWidth(), lcd.getHeight());
    }

}
```

```
robot@ev3dev:~$ sudo cat /dev/fb0 | wc -c
91136
robot@ev3dev:~$ sudo fbset -fb /dev/fb0 -i

mode "178x128"
    geometry 178 128 178 128 32
    timings 0 0 0 0 0 0 0
    accel true
    rgba 8/16,8/8,8/0,0/0
endmode

Frame buffer device information:
    Name        : 
    Address     : 0xc4882000
    Size        : 91136
    Type        : PACKED PIXELS
    Visual      : TRUECOLOR
    XPanStep    : 1
    YPanStep    : 1
    YWrapStep   : 0
    LineLength  : 712
    Accelerator : No
robot@ev3dev:~$ 
```

JakubVanek, 2018-08-20:
Hmm, I'll have to add better framebuffer mode switching. I'll also have to review pixel order -- it looks that pixels are not stored in this order:
```
X R G B X R G B ...
```
but in this
```
B G R X B G R X ...
```
I'm not sure though, I'll have to look in the kernel driver.

JakubVanek, 2018-08-21:
Could you please try running the tests for the library again? I have added better xrgb handling, framebuffer memory mapping and automatic xrgb/bitplane detection.

jabrena, 2018-08-21:
Yes Sir!

I will test it today. Crossing fingers :)

jabrena, 2018-08-21:
With latest implementation it continue failing:

```
java -jar /home/robot/ev3dev-lang-java-2.4.0-SNAPSHOT-all.jar 

ev3dev#1|2018-08-21 22:03:31 [main] DEBUG ev3dev.utils.Shell - Command: cat /etc/os-release
ev3dev#1|2018-08-21 22:03:32 [main] DEBUG ev3dev.hardware.EV3DevDistros - Debian Stretch detected
ev3dev#1|2018-08-21 22:03:33 [main] DEBUG ev3dev.utils.Shell - Command: cat /etc/os-release
ev3dev#1|2018-08-21 22:03:33 [main] DEBUG ev3dev.hardware.EV3DevDistros - Debian Stretch detected
ev3dev#1|2018-08-21 22:03:34 [main] DEBUG ev3dev.hardware.EV3DevFileSystem - Root Path: /sys/class
ev3dev#1|2018-08-21 22:03:34 [main] TRACE ev3dev.utils.Sysfs - ls /sys/class/power_supply/lego-ev3-battery
ev3dev#1|2018-08-21 22:03:34 [main] TRACE ev3dev.hardware.EV3DevPlatformsImpl - Detected platform: EV3BRICK
ev3dev#1|2018-08-21 22:03:34 [main] TRACE ev3dev.utils.Sysfs - cat /sys/class/graphics/fb0/bits_per_pixel
ev3dev#1|2018-08-21 22:03:34 [main] TRACE ev3dev.utils.Sysfs - value: 32
ev3dev#1|Exception in thread "main" java.lang.ExceptionInInitializerError
ev3dev#1|Caused by: java.lang.RuntimeException: Unable to map the framebuffer
ev3dev#1|	at ev3dev.actuators.LCD.init(LCD.java:159)
ev3dev#1|	at ev3dev.actuators.LCD.<init>(LCD.java:56)
ev3dev#1|	at ev3dev.actuators.LCD.getInstance(LCD.java:46)
ev3dev#1|	at example.LCDDrawLinesTest.<clinit>(LCDDrawLinesTest.java:10)
ev3dev#1|Caused by: java.io.IOException: Invalid argument
ev3dev#1|	at java.base/sun.nio.ch.FileDispatcherImpl.truncate0(Native Method)
ev3dev#1|	at java.base/sun.nio.ch.FileDispatcherImpl.truncate(Unknown Source)
ev3dev#1|	at java.base/sun.nio.ch.FileChannelImpl.map(Unknown Source)
ev3dev#1|	at ev3dev.actuators.LCD.initFramebuffer(LCD.java:102)
ev3dev#1|	at ev3dev.actuators.LCD.init(LCD.java:157)
ev3dev#1|	... 3 more
``` 

Why that invalid argument exception?

JakubVanek, 2018-08-22:
Hmm, I don't know, but I have googled a bit and found that `FileChannel.map()` is not suitable for block devices. I therefore put back the old `Sysfs.writeBytes()` method of writing the data.

jabrena, 2018-08-22:
I will test it again in the morning. :)

jabrena, 2018-08-22:
Now, the example doesn´t crash but we are not interacting with the Display, I don´t appreciate any operation:

```
java -jar /home/robot/ev3dev-lang-java-2.4.0-SNAPSHOT-all.jar 

ev3dev#5|2018-08-22 05:35:39 [main] DEBUG ev3dev.utils.Shell - Command: cat /etc/os-release
ev3dev#5|2018-08-22 05:35:40 [main] DEBUG ev3dev.hardware.EV3DevDistros - Debian Stretch detected
ev3dev#5|2018-08-22 05:35:40 [main] DEBUG ev3dev.utils.Shell - Command: cat /etc/os-release
ev3dev#5|2018-08-22 05:35:41 [main] DEBUG ev3dev.hardware.EV3DevDistros - Debian Stretch detected
ev3dev#5|2018-08-22 05:35:41 [main] DEBUG ev3dev.hardware.EV3DevFileSystem - Root Path: /sys/class
ev3dev#5|2018-08-22 05:35:41 [main] TRACE ev3dev.utils.Sysfs - ls /sys/class/power_supply/lego-ev3-battery
ev3dev#5|2018-08-22 05:35:41 [main] TRACE ev3dev.hardware.EV3DevPlatformsImpl - Detected platform: EV3BRICK
ev3dev#5|2018-08-22 05:35:41 [main] TRACE ev3dev.utils.Sysfs - cat /sys/class/graphics/fb0/bits_per_pixel
ev3dev#5|2018-08-22 05:35:41 [main] TRACE ev3dev.utils.Sysfs - value: 32
ev3dev#5|EV3 LCD Example
```

JakubVanek, 2018-08-23:
Hi, I added back the byte-oriented XRGB operation. It works on my Linux notebook when I switch to the console. Could you please test it again?

If this fails too (it is quite possible), we may have to go for some JNI code and do the same thing [grx linux framebuffer driver](https://github.com/ev3dev/grx/blob/master/plugins/video_driver/linuxfb/vd_lnxfb.c) does.

jabrena, 2018-08-24:
Good morning,

I was thinking this morning, where is Jakub? :p but here appeared your reply :)
I will test again when I return home.

If the test fails, take a look this Pull Rest for Go:
https://github.com/ev3go/ev3/pull/3/files
https://github.com/ev3go/ev3dev/tree/master/fb

I think that without the grx drivers, we should solve the issue.

Juan Antonio

JakubVanek, 2018-08-24:
Hmmm, quick look tells me that ev3go only `mmap`s the framebuffer and then operates on that. But writing to framebuffer using `write()` should do the same thing.

jabrena, 2018-08-24:
Oki, crossing fingers. In few hours, I will test it.

jabrena, 2018-08-25:
It doesn´t crash but it doesn´t clear the display and later draw the lines. 

``` 
java -jar /home/robot/ev3dev-lang-java-2.4.0-SNAPSHOT-all.jar 

ev3dev#2|2018-08-24 16:32:10 [main] DEBUG ev3dev.utils.Shell - Command: cat /etc/os-release
ev3dev#2|2018-08-24 16:32:11 [main] DEBUG ev3dev.hardware.EV3DevDistros - Debian Stretch detected
ev3dev#2|2018-08-24 16:32:11 [main] DEBUG ev3dev.utils.Shell - Command: cat /etc/os-release
ev3dev#2|2018-08-24 16:32:11 [main] DEBUG ev3dev.hardware.EV3DevDistros - Debian Stretch detected
ev3dev#2|2018-08-24 16:32:12 [main] DEBUG ev3dev.hardware.EV3DevFileSystem - Root Path: /sys/class
ev3dev#2|2018-08-24 16:32:12 [main] TRACE ev3dev.utils.Sysfs - ls /sys/class/power_supply/lego-ev3-battery
ev3dev#2|2018-08-24 16:32:12 [main] TRACE ev3dev.hardware.EV3DevPlatformsImpl - Detected platform: EV3BRICK
ev3dev#2|2018-08-24 16:32:12 [main] TRACE ev3dev.utils.Sysfs - cat /sys/class/graphics/fb0/bits_per_pixel
ev3dev#2|2018-08-24 16:32:12 [main] TRACE ev3dev.utils.Sysfs - value: 32
ev3dev#2|EV3 LCD Example
```

dlech, 2018-08-25:
I don't think you can use `write` anymore, I think you need to use `mmap`.

JakubVanek, 2018-08-25:
Wow, that's it! Thanks!
This works:
```c
#include ...

int main() {
  char *data = malloc(91136);
  int fd = open("/home/robot/fb0", O_RDONLY);
  read(fd, data, 91136);
  close(fd);
  fd = open("/dev/fb0", O_RDWR);
  char *dst = mmap(NULL, 91136, PROT_READ | PROT_WRITE, MAP_SHARED, fd, 0);
  close(fd);
  memcpy(dst, data, 91136);
  munmap(dst, 91136);
}
```
And this does not:
```c
#include ...

int main() {
  char *data = malloc(91136);
  int fd = open("/home/robot/fb0", O_RDONLY);
  read(fd, data, 91136);
  close(fd);
  fd = open("/dev/fb0", O_RDWR);
  write(fd, data, 91136);
  close(fd);
}
```

jabrena, 2018-08-31:
```
java -jar /home/robot/ev3dev-lang-java-2.4.0-SNAPSHOT-all.jar 

ev3dev#2|2018-08-31 05:51:17 [main] INFO  ev3dev.actuators.LCD - Framebuffer '%s' is compatible
ev3dev#2|2018-08-31 05:51:18 [main] DEBUG ev3dev.utils.Shell - Command: cat /etc/os-release
ev3dev#2|2018-08-31 05:51:18 [main] DEBUG ev3dev.hardware.EV3DevDistros - Debian Jessie detected
ev3dev#2|2018-08-31 05:51:18 [main] DEBUG ev3dev.utils.Shell - Command: cat /etc/os-release
ev3dev#2|2018-08-31 05:51:18 [main] DEBUG ev3dev.hardware.EV3DevDistros - Debian Jessie detected
ev3dev#2|2018-08-31 05:51:18 [main] DEBUG ev3dev.hardware.EV3DevFileSystem - Root Path: /sys/class
ev3dev#2|2018-08-31 05:51:18 [main] TRACE ev3dev.utils.Sysfs - ls /sys/class/power_supply/legoev3-battery
ev3dev#2|2018-08-31 05:51:18 [main] TRACE ev3dev.utils.Sysfs - ls /sys/class/power_supply/pistorms-battery
ev3dev#2|2018-08-31 05:51:18 [main] TRACE ev3dev.utils.Sysfs - ls /sys/class/power_supply/brickpi-battery
ev3dev#2|Exception in thread "main" java.lang.ExceptionInInitializerError
ev3dev#2|Caused by: java.lang.NullPointerException
ev3dev#2|2018-08-31 05:51:18 [main] TRACE ev3dev.utils.Sysfs - ls /sys/class/power_supply/brickpi3-battery
ev3dev#2|	at java.base/sun.nio.fs.UnixFileSystem.getPath(Unknown Source)
ev3dev#2|	at java.base/java.nio.file.Path.of(Unknown Source)
ev3dev#2|	at java.base/java.nio.file.Paths.get(Unknown Source)
ev3dev#2|	at ev3dev.hardware.EV3DevPlatformsImpl.batteryTest(EV3DevPlatformsImpl.java:24)
ev3dev#2|	at ev3dev.hardware.EV3DevPlatformsImpl.lambda$new$0(EV3DevPlatformsImpl.java:49)
ev3dev#2|	at java.base/java.util.stream.ReferencePipeline$2$1.accept(Unknown Source)
ev3dev#2|	at java.base/java.util.Spliterators$IteratorSpliterator.tryAdvance(Unknown Source)
ev3dev#2|	at java.base/java.util.stream.ReferencePipeline.forEachWithCancel(Unknown Source)
ev3dev#2|	at java.base/java.util.stream.AbstractPipeline.copyIntoWithCancel(Unknown Source)
ev3dev#2|	at java.base/java.util.stream.AbstractPipeline.copyInto(Unknown Source)
ev3dev#2|	at java.base/java.util.stream.AbstractPipeline.wrapAndCopyInto(Unknown Source)
ev3dev#2|	at java.base/java.util.stream.FindOps$FindOp.evaluateSequential(Unknown Source)
ev3dev#2|	at java.base/java.util.stream.AbstractPipeline.evaluate(Unknown Source)
ev3dev#2|	at java.base/java.util.stream.ReferencePipeline.findFirst(Unknown Source)
ev3dev#2|	at ev3dev.hardware.EV3DevPlatformsImpl.<init>(EV3DevPlatformsImpl.java:50)
ev3dev#2|	at ev3dev.hardware.EV3DevPlatforms.<init>(EV3DevPlatforms.java:12)
ev3dev#2|	at ev3dev.hardware.EV3DevDevice.<init>(EV3DevDevice.java:43)
ev3dev#2|	at ev3dev.actuators.LCD.<init>(LCD.java:66)
ev3dev#2|	at ev3dev.actuators.LCD.<init>(LCD.java:59)
ev3dev#2|	at ev3dev.actuators.LCD.getInstance(LCD.java:111)
ev3dev#2|	at example.LCDDrawLinesTest.<clinit>(LCDDrawLinesTest.java:10)

> Task :remoteRun FAILED
Failed command ev3dev#2 with status 1: java -jar /home/robot/ev3dev-lang-java-2.4.0-SNAPSHOT-all.jar 
``` 

It continue failing

jabrena, 2018-08-31:
We have to remove this line:

```
test {
    jacoco {
        append = false
        destinationFile = file("$buildDir/jacoco/jacocoTest.exec")
        classDumpDir = file("$buildDir/jacoco/classpathdumps")
    }
    //systemProperty "EV3DEV_DISTRO", project.getProperty("EV3DEV_DISTRO")
}
``` 

The distro is an internal query. It is not necessary to appear anything on `Build.gradle`.

jabrena, 2018-08-31:
If you like, focus on Stretch and later, I will review the part about `EV3DevDistros.java`

jabrena, 2018-08-31:
For Dev, purposes, the library returns Stretch, but the example continue failing:

``` 
java -jar /home/robot/ev3dev-lang-java-2.4.0-SNAPSHOT-all.jar 

ev3dev#6|2018-08-31 06:01:17 [main] INFO  ev3dev.actuators.LCD - Framebuffer '%s' is compatible
ev3dev#6|2018-08-31 06:01:17 [main] DEBUG ev3dev.hardware.EV3DevDistros - Debian Stretch detected
ev3dev#6|2018-08-31 06:01:17 [main] DEBUG ev3dev.hardware.EV3DevDistros - Debian Stretch detected
ev3dev#6|2018-08-31 06:01:18 [main] DEBUG ev3dev.hardware.EV3DevFileSystem - Root Path: /sys/class
ev3dev#6|2018-08-31 06:01:18 [main] TRACE ev3dev.utils.Sysfs - ls /sys/class/power_supply/lego-ev3-battery
ev3dev#6|2018-08-31 06:01:18 [main] TRACE ev3dev.hardware.EV3DevPlatformsImpl - Detected platform: EV3BRICK
ev3dev#6|Exception in thread "main" java.lang.ExceptionInInitializerError
ev3dev#6|Caused by: java.lang.IllegalArgumentException: Width (0) and height (0) must be > 0
ev3dev#6|	at java.desktop/java.awt.image.SampleModel.<init>(Unknown Source)
ev3dev#6|	at java.desktop/java.awt.image.MultiPixelPackedSampleModel.<init>(Unknown Source)
ev3dev#6|	at ev3dev.utils.display.ImageUtils.createBWImage(ImageUtils.java:148)
ev3dev#6|	at ev3dev.utils.display.BitFramebuffer.createCompatibleBuffer(BitFramebuffer.java:38)
ev3dev#6|	at ev3dev.utils.display.LinuxFramebuffer.createCompatibleBuffer(LinuxFramebuffer.java:79)
ev3dev#6|	at ev3dev.utils.display.BitFramebuffer.createCompatibleBuffer(BitFramebuffer.java:12)
ev3dev#6|	at ev3dev.utils.display.LinuxFramebuffer.createCompatibleBuffer(LinuxFramebuffer.java:71)
ev3dev#6|	at ev3dev.utils.display.BitFramebuffer.createCompatibleBuffer(BitFramebuffer.java:12)
ev3dev#6|	at ev3dev.actuators.LCD.<init>(LCD.java:75)
ev3dev#6|	at ev3dev.actuators.LCD.<init>(LCD.java:59)
ev3dev#6|	at ev3dev.actuators.LCD.getInstance(LCD.java:111)
ev3dev#6|	at example.LCDDrawLinesTest.<clinit>(LCDDrawLinesTest.java:10)
```

JakubVanek, 2018-08-31:
I've ran it on my notebook and I fixed some bugs. Now it _should_ work. :smile:

jabrena, 2018-08-31:
In next hours, I will run on the Brick.

jabrena, 2018-08-31:
We start drawing something...

```
java -jar /home/robot/ev3dev-lang-java-2.4.0-SNAPSHOT-all.jar 

ev3dev#9|2018-08-31 16:13:27 [main] INFO  ev3dev.actuators.LCD - Framebuffer 'BitFramebufferProvider' is not compatible
ev3dev#9|2018-08-31 16:13:27 [main] INFO  ev3dev.actuators.LCD - Framebuffer 'RGBFramebufferProvider' is compatible
ev3dev#9|2018-08-31 16:13:27 [main] DEBUG ev3dev.hardware.EV3DevDistros - Debian Stretch detected
ev3dev#9|2018-08-31 16:13:27 [main] DEBUG ev3dev.hardware.EV3DevDistros - Debian Stretch detected
ev3dev#9|2018-08-31 16:13:28 [main] DEBUG ev3dev.hardware.EV3DevFileSystem - Root Path: /sys/class
ev3dev#9|2018-08-31 16:13:28 [main] TRACE ev3dev.utils.Sysfs - ls /sys/class/power_supply/lego-ev3-battery
ev3dev#9|2018-08-31 16:13:28 [main] TRACE ev3dev.hardware.EV3DevPlatformsImpl - Detected platform: EV3BRICK
ev3dev#9|2018-08-31 16:13:35 [main] TRACE ev3dev.actuators.LCD - flushing framebuffer
ev3dev#9|EV3 LCD Example
ev3dev#9|2018-08-31 16:13:35 [main] TRACE ev3dev.actuators.LCD - flushing framebuffer
```

The execution of the program: LCDDrawLinesTest only put the display in Black, but it is a progress.
Any idea why?

```java
package example;

import ev3dev.actuators.LCD;
import lejos.hardware.lcd.GraphicsLCD;
import lejos.robotics.Color;
import lejos.utility.Delay;

public class LCDDrawLinesTest {

    public static GraphicsLCD lcd = LCD.getInstance();

    public static void main(final String[] args){

        System.out.println("EV3 LCD Example");

        clear();
        lcd.setColor(Color.BLACK);
        lcd.drawLine(0, 0, 50, 50);
        lcd.drawLine(0, 0, 30, 60);
        lcd.refresh();

        Delay.msDelay(5000);
    }

    public static void clear(){
        lcd.setColor(Color.WHITE);
        lcd.fillRect(0,0, lcd.getWidth(), lcd.getHeight());
    }

}
```

jabrena, 2018-08-31:
Maybe the Color.WHITE and Color.BLACK?

jabrena, 2018-08-31:
One thing: Good progress!!! @JakubVanek

JakubVanek, 2018-08-31:
Hmmm, not all colors are the same... :smile:
I've tried it on the brick and the framebuffer was full of pixels BGRX:060000FF, and it is because `lejos.robotics.Color.* are integers.
I've fixed it in the example and in the LCD documentation.

JakubVanek, 2018-08-31:
Graphical output now works, but there is one caveat. Once you print new line on the console, this triggers redraw and the graphical output is ovewritten with text.

dlech, 2018-08-31:
The way the Linux virtual console works, there is a graphical mode and a text mode. If you want to draw stuff on the screen graphically, you need to put it in graphics mode, otherwise you will get conflicts as you have seen. See https://github.com/ev3dev/ev3dev-lang-python/blob/ev3dev-stretch/ev3dev2/display.py for an example of how to do this.

jabrena, 2018-08-31:
@JakubVanek, Congratulations!

- Backgrounds: Black or Write: Ok
- Basic Geometry: Lines | Rectangles: Ok
- Images: Ok
- Fonts
- Test

As @dlech says, When we execute any Java process, Brickman is working in the background. For Jessie, we developed `BrickMan.java` but maybe exist others ways to interact in a better way. 

The most important thing is: **We have EV3 Display support again.**

@JakubVanek, tomorrow, I will add some notes for the code review. Maybe the package about the display we should move to hardware package and we have pending the Distros review too.

Congrats again.

**Note:** @dlech Thanks for the technical support :)

Juan Antonio

JakubVanek, 2018-08-31:
Thanks :)

I noticed that Brickman is not a problem when the jar is executed through [openjdk-system](http://github.com/ev3dev-lang-java/openjdk-system) shjar.

Jakub

jabrena, 2018-08-31:
hehehehe, ` openjdk-system shjar.` the technology is coming...

Well, maybe we should **rename** with an exotic name or something... 

In OpenJDK exist:

- Amber
- Graal
- Kona
- Kulla
- Lambda
- Metropolis

http://openjdk.java.net/

Find a name about some cool place from `Cech republic`. In `Spring Cloud` for example, the team chose in the last releases, names of streets located at London:
http://projects.spring.io/spring-cloud/

jabrena, 2018-08-31:
a name is ASCII characters please :)

JakubVanek, 2018-09-01:
I have just finished support for avoiding Brickman-app and graphics-text collisions. It works from both from Brickman menu (through ioctls) and from SSH (through stopping Brickman).

@dlech thanks for the link, but how does it the python implementation do? I have inspired myself in GRX3 [linuxfb](https://github.com/ev3dev/grx/blob/master/plugins/video_driver/linuxfb/vd_lnxfb.c) driver, which does it through ioctls and `ioctl(fd, KDSETMODE, KD_TEXT/KD_GRAPHICS)` in particular.

dlech, 2018-09-01:
it *should* do it the same as GRX3. So just go with what GRX3 does.

jabrena, 2018-09-04:
Testing latest code again

jabrena, 2018-09-04:
KO on the first example: `LCDDrawIconsTest`

``` 
ev3dev#2|2018-09-04 19:24:47 [main] DEBUG e.hardware.display.SystemDisplay - initializing new real display
ev3dev#2|2018-09-04 19:24:47 [main] TRACE ev3dev.hardware.display.OwnedDisplay - Initialing system console
ev3dev#2|2018-09-04 19:24:47 [main] TRACE ev3dev.hardware.display.OwnedDisplay - Opening TTY
ev3dev#2|Exception in thread "main" java.lang.ExceptionInInitializerError
ev3dev#2|Caused by: java.lang.NullPointerException
ev3dev#2|	at ev3dev.hardware.display.OwnedDisplay.initialize(OwnedDisplay.java:103)
ev3dev#2|	at ev3dev.hardware.display.OwnedDisplay.<init>(OwnedDisplay.java:53)
ev3dev#2|	at ev3dev.hardware.display.SystemDisplay.initializeRealDisplay(SystemDisplay.java:34)
ev3dev#2|	at ev3dev.hardware.display.SystemDisplay.initializeRealFramebuffer(SystemDisplay.java:55)
ev3dev#2|	at ev3dev.hardware.display.SystemDisplay.initializeRealLCD(SystemDisplay.java:67)
ev3dev#2|	at example.LCDDrawIconsTest.<clinit>(LCDDrawIconsTest.java:12)
```

jabrena, 2018-09-04:
Another point for the code review: `OwnedDisplay`

```
import sun.misc.Signal;
import sun.misc.SignalHandler;
```

JakubVanek, 2018-09-05:
Hmmm, regarding issue 1: it seems that construction of ttyfd failed and I forgot the null check in the finally block.
Regarding issue 2: yes, this code is there for terminal switching - the display manager needs to catch an arriving signal, so it can pause/resume drawing on the display. I haven't been able to test it through Ctrl+Alt+Fx, maybe it is done from the app (GRX has a chvt function called from input handling routines). This feature isn't that important, so I think we can remove it.

JakubVanek, 2018-09-05:
These two issues should be fixed now.

jabrena, 2018-09-05:
Today, I will do the tests.

JakubVanek, 2018-09-05:
Hmm, you might need to do
```
sudo adduser robot video
```
or there might be problem with permissions:
```
$ ls -la /dev/fb0
crw-rw---- 1 root video 29, 0 Sep  5 14:56 /dev/fb0
```

jabrena, 2018-09-05:
KO: LCDDrawLinesTest

``` 
java -jar /home/robot/ev3dev-lang-java-2.4.0-SNAPSHOT-all.jar 

ev3dev#4|2018-09-05 18:35:39 [main] DEBUG e.hardware.display.SystemDisplay - initializing new real display
ev3dev#4|Exception in thread "main" java.lang.ExceptionInInitializerError
ev3dev#4|Caused by: com.sun.jna.LastErrorException: [6] No such device or address
ev3dev#4|2018-09-05 18:35:40 [main] TRACE ev3dev.hardware.display.OwnedDisplay - Initialing system console
ev3dev#4|2018-09-05 18:35:40 [main] TRACE ev3dev.hardware.display.OwnedDisplay - Opening TTY
ev3dev#4|	at ev3dev.utils.io.NativeLibc.open(Native Method)
ev3dev#4|	at ev3dev.utils.io.NativeFile.open(NativeFile.java:114)
ev3dev#4|	at ev3dev.utils.io.NativeFile.<init>(NativeFile.java:79)
ev3dev#4|	at ev3dev.utils.io.NativeDevice.<init>(NativeDevice.java:53)
ev3dev#4|	at ev3dev.utils.io.NativeTTY.<init>(NativeTTY.java:62)
ev3dev#4|	at ev3dev.hardware.display.OwnedDisplay.initialize(OwnedDisplay.java:69)
ev3dev#4|	at ev3dev.hardware.display.OwnedDisplay.<init>(OwnedDisplay.java:46)
ev3dev#4|	at ev3dev.hardware.display.SystemDisplay.initializeRealDisplay(SystemDisplay.java:34)
ev3dev#4|	at ev3dev.hardware.display.SystemDisplay.initializeRealFramebuffer(SystemDisplay.java:55)
ev3dev#4|	at ev3dev.hardware.display.SystemDisplay.initializeRealLCD(SystemDisplay.java:67)
ev3dev#4|	at example.LCDDrawLinesTest.<clinit>(LCDDrawLinesTest.java:9)
```

JakubVanek, 2018-09-05:
Hmmm, are you running it from the template project? If so, that might be the problem - maybe SSH doesn't allocate a pseudo terminal when doing so. Try running it from real SSH prompt, that was working on my brick.
I can add detection for the gradle-run case too - in SystemDisplay, it is just necessary to add a check for ENXIO next to the check for ENOTTY.

[12 more replies]

closed 2018-10-28 by jabrena