ev3dev-lang-java/ev3dev-lang-java #501: Getting start ROS  on EV3 Brick

YuHanChen04, 2018-05-14:
Getting start !!!

jabrena, 2018-05-15:
**Note:** @YuHanChen04 for future issues, try to write a proper description. Remember that the issues could help other users. :)

jabrena, 2018-05-15:
@YuHanChen04, Don´t worry, I will guide you.
Exist many things to do.

YuHanChen04, 2018-05-19:
Hi @jabrena 
I am busy at the last days
And I will try to install ROS ... in the future.
Give me more time !! Thanks !

jabrena, 2018-05-19:
# 2. The solution architecture

## 2.1 Introduction

This project provides a solution to manage Sensors/Actuators from the platform Lego Mindstorms EV3 & Raspberry Pi boards compatible with EV3 Sensors like BrickPi & PiStorms.

## 2.2 Study of alternatives

The solution has been designed to broadcast sensor data to `ROS` using `ROSBridge`. So, the communication is `Botton-up`

In the JVM ecosystem exists 2 alternatives to explore:

+ ROSJava
+ ROSBridge

## 2.1 ROSJava from the Bricks

I tested the usage of `ROSJava` from `EV3` but I could connect due to possible concurrency issues.

Related issues:
https://github.com/ev3dev-lang-java/ev3dev-lang-java/issues/210
https://github.com/rosjava/rosjava_core/issues/251

## 2.2 ROS Bridge

ROSBridge provides a JSON API to ROS functionality for non-ROS programs.
http://wiki.ros.org/rosbridge_suite

Using ROSBridge, it is possible to broadcast information to ROS

## 2.3 ROSJava from ROS computer

Not tested

## 2.4 Summary

Using ROSBridge, it is possible to publish Sensor/Data to ROS.

jabrena, 2018-05-19:
# 3. ROS Client

## 3.1 Introduction 
The project provides a library to publish & subscribe Topics from ROS.
https://github.com/ev3dev-lang-java/ros-client

## 3.2 Topics supported
Using this library, the user could publish & subscribe information to ROS for the following topics:

**Topics Published**

- battery (std_msgs.Float32)
- scan (sensor_msgs.LaserScan)

**Topics Subscribed**

- cmd_vel (geometry_msgs/Twist)

jabrena, 2018-05-19:
# 4. ROS

## 4.1 Running ROS from Vagrant 

Using the following repo:
https://github.com/ev3dev-lang-java/ROS-Kinetic_vagrantfile

Execute the following commands to run ROS Kinetic in your computer:

```
git clone https://github.com/ev3dev-lang-java/ROS-Kinetic_vagrantfile
cd ROS-Kinetic_vagrantfile
vagrant up
```

Inside of the environment, open a terminal and run the following commands:

```
source /opt/ros/kinetic/setup.bash
cd catkin_ws
source install_isolated/setup.bash
roslaunch rosbridge_server rosbridge_websocket.launch
rosrun rviz rviz
```

Related issue:
https://github.com/ev3dev-lang-java/ev3dev-lang-java/issues/393

jabrena, 2018-05-19:
# 5. Getting Started with ROS for EV3

## 5.1 Teleoperate your EV3 Brick with ROS

### 5.1.1 EV3

Create a project to run the following example:

``` 
public class TeleoperationDemo {

    public static void main(String[] args) throws InterruptedException {

        System.out.println("Running a TeleOperation Demo");

        Ros ros = new Ros("192.168.1.198");
        ros.connect();

        System.out.println("Connected");

        PilotConfig pilotConfig = new PilotConfig();

        TeleOperation teleoperation = new TeleOperation(ros, pilotConfig.getPilot());
        teleoperation.subscribe();

        //EV3IRRange irSensor = new EV3IRRange(ros, SensorPort.S1, "base_link");
        EV3UltrasonicRange ultrasonicSensor = new EV3UltrasonicRange(ros, SensorPort.S2, "base_link");

        boolean flag = true;
        while (flag == true){
            //Empty
            //irSensor.publish();
            ultrasonicSensor.publish();
        }

        ros.disconnect();
    }

}
```

### 5.1.1 ROS

Run in ROS the following ROS node to teleoperate your Brick with the keyboard

```
rosrun teleop_twist_keyboard teleop_twist_keyboard.py
```