ev3dev-lang-java/ev3dev-lang-java #502: Support for the usage of Multiple webcams

jabrena, 2018-05-14:
From this conversation on `LeJOS Forum`:
https://lejos.sourceforge.io/forum/viewtopic.php?f=18&t=9271

jabrena, 2018-05-14:
Installing OpenCV on Debian Stretch:

```
robot@ev3dev:~$ sudo apt-get install --no-install-recommends --no-upgrade libopencv2.4-java
[sudo] password for robot: 
Reading package lists... Done
Building dependency tree        
Reading state information... Done
The following additional packages will be installed:
  fontconfig-config libavcodec57 libavformat57 libavutil55 libbluray2 libchromaprint1 libdc1394-22 libfontconfig1 libgme0 libgsm1 libgtk2.0-0 libgtk2.0-common
  libilmbase12 libilmbase23 libmp3lame0 libmpg123-0 libopencv-calib3d2.4v5 libopencv-contrib2.4v5 libopencv-core2.4v5 libopencv-features2d2.4v5 libopencv-flann2.4v5
  libopencv-highgui2.4-deb0 libopencv-imgproc2.4v5 libopencv-ml2.4v5 libopencv-objdetect2.4v5 libopencv-photo2.4v5 libopencv-video2.4v5 libopencv2.4-jni
  libopenexr22 libopenjp2-7 libopenmpt0 libraw1394-11 libshine3 libsnappy1v5 libsoxr0 libspeex1 libssh-gcrypt-4 libswresample2 libswscale4 libtheora0 libtwolame0
  libv4l-0 libv4lconvert0 libva-drm2 libva-x11-2 libva2 libvdpau1 libvorbis0a libvorbisenc2 libvorbisfile3 libvpx5 libwavpack1 libwebpmux3 libx264-152 libx265-146
  libxvidcore4 libzvbi-common libzvbi0
Suggested packages:
  libbluray-bdj gvfs libraw1394-doc speex
Recommended packages:
  libaacs0 libgail-common libgtk2.0-bin va-driver-all | va-driver vdpau-driver-all | vdpau-driver
The following NEW packages will be installed:
  libavcodec57 libavformat57 libavutil55 libbluray2 libchromaprint1 libdc1394-22 libgme0 libgsm1 libgtk2.0-0 libgtk2.0-common libilmbase12 libilmbase23 libmp3lame0
  libmpg123-0 libopencv-calib3d2.4v5 libopencv-contrib2.4v5 libopencv-core2.4v5 libopencv-features2d2.4v5 libopencv-flann2.4v5 libopencv-highgui2.4-deb0
  libopencv-imgproc2.4v5 libopencv-ml2.4v5 libopencv-objdetect2.4v5 libopencv-photo2.4v5 libopencv-video2.4v5 libopencv2.4-java libopencv2.4-jni libopenexr22
  libopenjp2-7 libopenmpt0 libraw1394-11 libshine3 libsnappy1v5 libsoxr0 libspeex1 libssh-gcrypt-4 libswresample2 libswscale4 libtheora0 libtwolame0 libv4l-0
  libv4lconvert0 libva-drm2 libva-x11-2 libva2 libvdpau1 libvorbisfile3 libvpx5 libwavpack1 libwebpmux3 libx264-152 libx265-146 libxvidcore4 libzvbi-common libzvbi0
The following packages will be upgraded:
  fontconfig-config libfontconfig1 libvorbis0a libvorbisenc2
4 upgraded, 55 newly installed, 0 to remove and 348 not upgraded.
Need to get 19.3 MB of archives.
After this operation, 68.9 MB of additional disk space will be used.
Do you want to continue? [Y/n] y
```

jabrena, 2018-05-19:
**Dev note:**
Unfortunately, I don´t have internet at home and I have to wait some days to test it.

jabrena, 2018-05-19:
Solved with a Bluetooth connection.
http://www.ev3dev.org/docs/tutorials/connecting-to-the-internet-via-bluetooth/

jabrena, 2018-05-19:
https://mvnrepository.com/artifact/nu.pattern/opencv/2.4.9-4

jabrena, 2018-05-19:
```
wget http://ftp.us.debian.org/debian/pool/main/o/opencv/libopencv2.4-java_2.4.9.1+dfsg1-2_all.deb
dpkg -i libopencv2.4-java_2.4.9.1+dfsg1-2_all.deb
```

jabrena, 2018-05-19:
Unfortunately for Debian Stretch, I found a runtime error on the dependency:

```
robot@ev3dev:~$ java -Djava.library.path=/usr/lib/jni/ -jar opencv-0.1.0-SNAPSHOT-all.jar 
2018-05-19 20:44:17 [main] INFO  opencv.HelloWorldCV - Welcome to OpenCV 2.4.9.0
2018-05-19 20:44:17 [main] INFO  opencv.HelloWorldCV - /usr/lib/jni/
Segmentation fault
```

jabrena, 2018-05-19:
Testing with OpenCV 3.3:

``` 
sudo apt-get install -t experimental --no-install-recommends --no-upgrade libopencv3.3-java
```

jabrena, 2018-05-19:
Related issues with scenarios about "Segmentation fault":
https://github.com/opencv/opencv/issues/10080
https://github.com/opencv/opencv/issues/10157

jabrena, 2018-05-19:
I tested with Oracle JRE 8 without success:

``` 
robot@ev3dev:~$ sudo update-alternatives --config java
[sudo] password for robot: 
There are 9 choices for the alternative java (providing /usr/bin/java).

  Selection    Path                                         Priority   Status
------------------------------------------------------------
  0            /usr/lib/jvm/java-10-openjdk-armel/bin/java   1094      auto mode
  1            /opt/ejdk1.8.0/linux_arm_sflt/jre/bin/java    1         manual mode
  2            /opt/jdk-10-build-46-v2/bin/java              1         manual mode
  3            /opt/jdk-10-build-46/bin/java                 1         manual mode
  4            /opt/jdk-9-build-46/bin/java                  1         manual mode
  5            /opt/jri-10-build-050/bin/java                10        manual mode
* 6            /opt/jri-10-build-46-v2/bin/java              1         manual mode
  7            /opt/jri-10-build-46-zero/bin/java            1         manual mode
  8            /opt/jri-10-build-46/bin/java                 1         manual mode
  9            /usr/lib/jvm/java-10-openjdk-armel/bin/java   1094      manual mode

Press <enter> to keep the current choice[*], or type selection number: 1
update-alternatives: using /opt/ejdk1.8.0/linux_arm_sflt/jre/bin/java to provide /usr/bin/java (java) in manual mode
robot@ev3dev:~$ java -Djava.library.path=/usr/lib/jni/ -cp /home/robot/opencv-0.1.0-SNAPSHOT-all.jar:/usr/share/OpenCV/java/opencv-249.jar example.opencv.HelloWorldCV
2018-05-19 22:00:57 [main] INFO  example.opencv.HelloWorldCV - Welcome to OpenCV 2.4.9.1
2018-05-19 22:00:58 [main] INFO  example.opencv.HelloWorldCV - /usr/lib/jni/
Segmentation fault
robot@ev3dev:~$ java -version
java version "1.8.0"
Java(TM) SE Embedded Runtime Environment (build 1.8.0-b132, headless)
Java HotSpot(TM) Embedded Client VM (build 25.0-b70, mixed mode)
robot@ev3dev:~$ 
```

jabrena, 2018-05-19:
```
sudo apt-get install gdb
```

jabrena, 2018-05-19:
```
1875.246s][info][class,load] java.util.IdentityHashMap$KeySet source: jrt:/java.base
java.lang.UnsatisfiedLinkError: no opencv_java249 in java.library.path: [/usr/lib/jni/libopencv_java249.so]
[1882.125s][info][class,load] java.lang.StackTraceElement$HashedModules source: jrt:/java.base
        at java.base/java.lang.ClassLoader.loadLibrary(Unknown Source)
        at java.base/java.lang.Runtime.loadLibrary0(Unknown Source)
        at java.base/java.lang.System.loadLibrary(Unknown Source)
        at example.opencv.HelloWorldCV.main(HelloWorldCV.java:14)
[1899.510s][info][class,load] java.lang.Shutdown source: jrt:/java.base
[1901.771s][info][class,load] java.lang.Shutdown$Lock source: jrt:/java.base
[Thread 0xb19f4460 (LWP 4290) exited]
[Thread 0xb1b64460 (LWP 4286) exited]
[Thread 0xb1f58460 (LWP 4283) exited]
[Thread 0xb1972460 (LWP 4291) exited]
[Thread 0xb1a44460 (LWP 4289) exited]
[Thread 0xb1a94460 (LWP 4288) exited]
[Thread 0xb1b14460 (LWP 4287) exited]
[Thread 0xb1e86460 (LWP 4285) exited]
[Thread 0xb1ed6460 (LWP 4284) exited]
[Thread 0xb65ff460 (LWP 4282) exited]
[Inferior 1 (process 4279) exited with code 01]
(gdb) bt
No stack.
(gdb) bt
No stack.
(gdb) Killed
```

JakubVanek, 2018-05-20:
Hi, you've captured it:
```
java.lang.UnsatisfiedLinkError: no opencv_java249 in java.library.path: [/usr/lib/jni/libopencv_java249.so]
```

```
kuba@linuxtardis-x200:~$ apt-file search libopencv_java249.so
libopencv2.4-jni: /usr/lib/jni/libopencv_java249.so
```

BUT, there's a catch... the package should be already installed. https://packages.debian.org/stretch/libopencv2.4-java (libopencv2.4-java depends on libopencv2.4-jni). Just in case, you can try to install it.
```
apt-get install libopencv2.4-jni
```

Does the file `/usr/lib/jni/libopencv_java249.so` exist? If it does, could you please send the output of `ldd /usr/lib/jni/libopencv_java249.so`? That lists all runtime dependencies of the library. If something is missing, that could be the problem.

Regards,

Jakub Vaněk

jabrena, 2018-05-20:
Hi Januk,

This is the output:

```
robot@ev3dev:~$ ldd /usr/lib/jni/libopencv_java249.so
	libopencv_contrib.so.2.4 => /usr/lib/arm-linux-gnueabi/libopencv_contrib.so.2.4 (0xb6e2f000)
	libopencv_photo.so.2.4 => /usr/lib/arm-linux-gnueabi/libopencv_photo.so.2.4 (0xb6e06000)
	libdl.so.2 => /lib/arm-linux-gnueabi/libdl.so.2 (0xb6df3000)
	libpthread.so.0 => /lib/arm-linux-gnueabi/libpthread.so.0 (0xb6dc8000)
	librt.so.1 => /lib/arm-linux-gnueabi/librt.so.1 (0xb6db1000)
	libopencv_calib3d.so.2.4 => /usr/lib/arm-linux-gnueabi/libopencv_calib3d.so.2.4 (0xb6d24000)
	libopencv_features2d.so.2.4 => /usr/lib/arm-linux-gnueabi/libopencv_features2d.so.2.4 (0xb6c91000)
	libopencv_flann.so.2.4 => /usr/lib/arm-linux-gnueabi/libopencv_flann.so.2.4 (0xb6c29000)
	libopencv_ml.so.2.4 => /usr/lib/arm-linux-gnueabi/libopencv_ml.so.2.4 (0xb6ba8000)
	libopencv_objdetect.so.2.4 => /usr/lib/arm-linux-gnueabi/libopencv_objdetect.so.2.4 (0xb6b3b000)
	libopencv_highgui.so.2.4 => /usr/lib/arm-linux-gnueabi/libopencv_highgui.so.2.4 (0xb6af2000)
	libopencv_video.so.2.4 => /usr/lib/arm-linux-gnueabi/libopencv_video.so.2.4 (0xb6a9f000)
	libopencv_imgproc.so.2.4 => /usr/lib/arm-linux-gnueabi/libopencv_imgproc.so.2.4 (0xb6862000)
	libopencv_core.so.2.4 => /usr/lib/arm-linux-gnueabi/libopencv_core.so.2.4 (0xb6685000)
	libstdc++.so.6 => /usr/lib/arm-linux-gnueabi/libstdc++.so.6 (0xb6539000)
	libm.so.6 => /lib/arm-linux-gnueabi/libm.so.6 (0xb6476000)
	libgcc_s.so.1 => /lib/arm-linux-gnueabi/libgcc_s.so.1 (0xb6446000)
	libc.so.6 => /lib/arm-linux-gnueabi/libc.so.6 (0xb62f3000)
	/lib/ld-linux.so.3 (0xb6f87000)
	libz.so.1 => /lib/arm-linux-gnueabi/libz.so.1 (0xb62cc000)
	libjpeg.so.62 => /usr/lib/arm-linux-gnueabi/libjpeg.so.62 (0xb6282000)
	libpng16.so.16 => /usr/lib/arm-linux-gnueabi/libpng16.so.16 (0xb6246000)
	libtiff.so.5 => /usr/lib/arm-linux-gnueabi/libtiff.so.5 (0xb61c7000)
	libImath-2_2.so.12 => /usr/lib/arm-linux-gnueabi/libImath-2_2.so.12 (0xb61a2000)
	libIlmImf-2_2.so.22 => /usr/lib/arm-linux-gnueabi/libIlmImf-2_2.so.22 (0xb5eee000)
	libIex-2_2.so.12 => /usr/lib/arm-linux-gnueabi/libIex-2_2.so.12 (0xb5ec7000)
	libHalf.so.12 => /usr/lib/arm-linux-gnueabi/libHalf.so.12 (0xb5e74000)
	libIlmThread-2_2.so.12 => /usr/lib/arm-linux-gnueabi/libIlmThread-2_2.so.12 (0xb5e5d000)
	libgtk-x11-2.0.so.0 => /usr/lib/arm-linux-gnueabi/libgtk-x11-2.0.so.0 (0xb5a85000)
	libgdk-x11-2.0.so.0 => /usr/lib/arm-linux-gnueabi/libgdk-x11-2.0.so.0 (0xb59dd000)
	libpangocairo-1.0.so.0 => /usr/lib/arm-linux-gnueabi/libpangocairo-1.0.so.0 (0xb59c0000)
	libatk-1.0.so.0 => /usr/lib/arm-linux-gnueabi/libatk-1.0.so.0 (0xb5992000)
	libcairo.so.2 => /usr/lib/arm-linux-gnueabi/libcairo.so.2 (0xb5889000)
	libgdk_pixbuf-2.0.so.0 => /usr/lib/arm-linux-gnueabi/libgdk_pixbuf-2.0.so.0 (0xb5853000)
	libgio-2.0.so.0 => /usr/lib/arm-linux-gnueabi/libgio-2.0.so.0 (0xb56e7000)
	libpangoft2-1.0.so.0 => /usr/lib/arm-linux-gnueabi/libpangoft2-1.0.so.0 (0xb56c4000)
	libpango-1.0.so.0 => /usr/lib/arm-linux-gnueabi/libpango-1.0.so.0 (0xb5671000)
	libgobject-2.0.so.0 => /usr/lib/arm-linux-gnueabi/libgobject-2.0.so.0 (0xb5614000)
	libglib-2.0.so.0 => /lib/arm-linux-gnueabi/libglib-2.0.so.0 (0xb5507000)
	libfontconfig.so.1 => /usr/lib/arm-linux-gnueabi/libfontconfig.so.1 (0xb54c1000)
	libfreetype.so.6 => /usr/lib/arm-linux-gnueabi/libfreetype.so.6 (0xb5424000)
	libgthread-2.0.so.0 => /usr/lib/arm-linux-gnueabi/libgthread-2.0.so.0 (0xb5412000)
	libdc1394.so.22 => /usr/lib/arm-linux-gnueabi/libdc1394.so.22 (0xb538c000)
	libv4l1.so.0 => /usr/lib/arm-linux-gnueabi/libv4l1.so.0 (0xb5376000)
	libavcodec.so.57 => /usr/lib/arm-linux-gnueabi/libavcodec.so.57 (0xb40b1000)
	libavformat.so.57 => /usr/lib/arm-linux-gnueabi/libavformat.so.57 (0xb3e8e000)
	libavutil.so.55 => /usr/lib/arm-linux-gnueabi/libavutil.so.55 (0xb3ded000)
	libswscale.so.4 => /usr/lib/arm-linux-gnueabi/libswscale.so.4 (0xb3d66000)
	liblzma.so.5 => /lib/arm-linux-gnueabi/liblzma.so.5 (0xb3d34000)
	libjbig.so.0 => /usr/lib/arm-linux-gnueabi/libjbig.so.0 (0xb3d17000)
	libgmodule-2.0.so.0 => /usr/lib/arm-linux-gnueabi/libgmodule-2.0.so.0 (0xb3d03000)
	libX11.so.6 => /usr/lib/arm-linux-gnueabi/libX11.so.6 (0xb3bd9000)
	libXcomposite.so.1 => /usr/lib/arm-linux-gnueabi/libXcomposite.so.1 (0xb3bc6000)
	libXdamage.so.1 => /usr/lib/arm-linux-gnueabi/libXdamage.so.1 (0xb3bb1000)
	libXfixes.so.3 => /usr/lib/arm-linux-gnueabi/libXfixes.so.3 (0xb3b9c000)
	libXrender.so.1 => /usr/lib/arm-linux-gnueabi/libXrender.so.1 (0xb3b82000)
	libXinerama.so.1 => /usr/lib/arm-linux-gnueabi/libXinerama.so.1 (0xb3b6f000)
	libXi.so.6 => /usr/lib/arm-linux-gnueabi/libXi.so.6 (0xb3b50000)
	libXrandr.so.2 => /usr/lib/arm-linux-gnueabi/libXrandr.so.2 (0xb3b37000)
	libXcursor.so.1 => /usr/lib/arm-linux-gnueabi/libXcursor.so.1 (0xb3b1e000)
	libXext.so.6 => /usr/lib/arm-linux-gnueabi/libXext.so.6 (0xb3aff000)
	libpixman-1.so.0 => /usr/lib/arm-linux-gnueabi/libpixman-1.so.0 (0xb3a47000)
	libxcb-shm.so.0 => /usr/lib/arm-linux-gnueabi/libxcb-shm.so.0 (0xb3a34000)
	libxcb.so.1 => /usr/lib/arm-linux-gnueabi/libxcb.so.1 (0xb3a02000)
[... 53 lines cut ...]
	libgcrypt.so.20 => /lib/arm-linux-gnueabi/libgcrypt.so.20 (0xb2599000)
	libgssapi_krb5.so.2 => /usr/lib/arm-linux-gnueabi/libgssapi_krb5.so.2 (0xb254f000)
	libmpg123.so.0 => /usr/lib/arm-linux-gnueabi/libmpg123.so.0 (0xb24ed000)
	libvorbisfile.so.3 => /usr/lib/arm-linux-gnueabi/libvorbisfile.so.3 (0xb24d4000)
	libxml2.so.2 => /usr/lib/arm-linux-gnueabi/libxml2.so.2 (0xb233c000)
[... cut ...]
	libk5crypto.so.3 => /usr/lib/arm-linux-gnueabi/libk5crypto.so.3 (0xb1fcc000)
	libcom_err.so.2 => /lib/arm-linux-gnueabi/libcom_err.so.2 (0xb1fb9000)
	libkrb5support.so.0 => /usr/lib/arm-linux-gnueabi/libkrb5support.so.0 (0xb1fa0000)
	libkeyutils.so.1 => /lib/arm-linux-gnueabi/libkeyutils.so.1 (0xb1f8d000)
	libicui18n.so.57 => /usr/lib/arm-linux-gnueabi/libicui18n.so.57 (0xb1d72000)
	libicuuc.so.57 => /usr/lib/arm-linux-gnueabi/libicuuc.so.57 (0xb1c03000)
	libicudata.so.57 => /usr/lib/arm-linux-gnueabi/libicudata.so.57 (0xb0376000)
```

jabrena, 2018-05-20:
Testing on Debian Jessie:

```
sudo apt-get update
sudo apt-get install -t stretch --no-install-recommends --no-upgrade libopencv2.4-java
```

jabrena, 2018-05-20:
```
java -Djava.library.path=/usr/lib/jni/ -cp /home/robot/opencv-0.1.0-SNAPSHOT-all.jar:/usr/share/OpenCV/java/opencv-249.jar example.opencv.HelloWorldCV
```

jabrena, 2018-05-20:
``` 
robot@ev3dev:/usr/lib/jni$ ls -la
total 556
drwxr-xr-x  2 root root   4096 May 20 09:58 .
drwxr-xr-x 42 root root   4096 May 20 09:58 ..
-rw-r--r--  1 root root 558372 May 28  2015 libopencv_java249.so
```

jabrena, 2018-05-20:
Output:

```
2018-05-20 10:11:18 [main] INFO  example.opencv.HelloWorldCV - Welcome to OpenCV 2.4.9.1
2018-05-20 10:11:18 [main] INFO  example.opencv.HelloWorldCV - /usr/lib/jni/
2018-05-20 10:11:18 [main] INFO  example.opencv.HelloWorldCV - opencv_java249
Exception in thread "main" java.lang.UnsatisfiedLinkError: /usr/lib/jni/libopencv_java249.so: libpng16.so.16: cannot open shared object file: No such file or directory
	at java.base/java.lang.ClassLoader$NativeLibrary.load0(Native Method)
	at java.base/java.lang.ClassLoader$NativeLibrary.load(Unknown Source)
	at java.base/java.lang.ClassLoader$NativeLibrary.loadLibrary(Unknown Source)
	at java.base/java.lang.ClassLoader.loadLibrary0(Unknown Source)
	at java.base/java.lang.ClassLoader.loadLibrary(Unknown Source)
	at java.base/java.lang.Runtime.loadLibrary0(Unknown Source)
	at java.base/java.lang.System.loadLibrary(Unknown Source)
	at example.opencv.HelloWorldCV.main(HelloWorldCV.java:15)
robot@ev3dev:~$ 
```
```
robot@ev3dev:~$ ldd /usr/lib/jni/libopencv_java249.so
	libopencv_core.so.2.4 => /usr/lib/arm-linux-gnueabi/libopencv_core.so.2.4 (0xb6ce0000)
	libopencv_flann.so.2.4 => /usr/lib/arm-linux-gnueabi/libopencv_flann.so.2.4 (0xb6c70000)
	libopencv_imgproc.so.2.4 => /usr/lib/arm-linux-gnueabi/libopencv_imgproc.so.2.4 (0xb6a04000)
	libopencv_highgui.so.2.4 => /usr/lib/arm-linux-gnueabi/libopencv_highgui.so.2.4 (0xb69b1000)
	libopencv_features2d.so.2.4 => /usr/lib/arm-linux-gnueabi/libopencv_features2d.so.2.4 (0xb690b000)
	libopencv_calib3d.so.2.4 => /usr/lib/arm-linux-gnueabi/libopencv_calib3d.so.2.4 (0xb6872000)
	libopencv_ml.so.2.4 => /usr/lib/arm-linux-gnueabi/libopencv_ml.so.2.4 (0xb67eb000)
	libopencv_objdetect.so.2.4 => /usr/lib/arm-linux-gnueabi/libopencv_objdetect.so.2.4 (0xb6771000)
	libopencv_video.so.2.4 => /usr/lib/arm-linux-gnueabi/libopencv_video.so.2.4 (0xb6716000)
	libopencv_contrib.so.2.4 => /usr/lib/arm-linux-gnueabi/libopencv_contrib.so.2.4 (0xb663f000)
	libopencv_photo.so.2.4 => /usr/lib/arm-linux-gnueabi/libopencv_photo.so.2.4 (0xb6614000)
	libdl.so.2 => /lib/arm-linux-gnueabi/libdl.so.2 (0xb6601000)
	libpthread.so.0 => /lib/arm-linux-gnueabi/libpthread.so.0 (0xb65d8000)
	librt.so.1 => /lib/arm-linux-gnueabi/librt.so.1 (0xb65c1000)
	libGLU.so.1 => /usr/lib/arm-linux-gnueabi/libGLU.so.1 (0xb6554000)
	libGL.so.1 => /usr/lib/arm-linux-gnueabi/libGL.so.1 (0xb64b0000)
	libSM.so.6 => /usr/lib/arm-linux-gnueabi/libSM.so.6 (0xb6498000)
	libICE.so.6 => /usr/lib/arm-linux-gnueabi/libICE.so.6 (0xb6471000)
	libX11.so.6 => /usr/lib/arm-linux-gnueabi/libX11.so.6 (0xb6346000)
	libXext.so.6 => /usr/lib/arm-linux-gnueabi/libXext.so.6 (0xb6327000)
	libstdc++.so.6 => /usr/lib/arm-linux-gnueabi/libstdc++.so.6 (0xb6249000)
	libm.so.6 => /lib/arm-linux-gnueabi/libm.so.6 (0xb6197000)
	libgcc_s.so.1 => /lib/arm-linux-gnueabi/libgcc_s.so.1 (0xb6168000)
	libc.so.6 => /lib/arm-linux-gnueabi/libc.so.6 (0xb601f000)
	/lib/ld-linux.so.3 (0xb6fa7000)
	libz.so.1 => /lib/arm-linux-gnueabi/libz.so.1 (0xb5ff7000)
	libjpeg.so.62 => /usr/lib/arm-linux-gnueabi/libjpeg.so.62 (0xb5f9f000)
	libpng12.so.0 => /lib/arm-linux-gnueabi/libpng12.so.0 (0xb5f6d000)
	libtiff.so.5 => /usr/lib/arm-linux-gnueabi/libtiff.so.5 (0xb5ef1000)
	libjasper.so.1 => /usr/lib/arm-linux-gnueabi/libjasper.so.1 (0xb5e93000)
	libImath.so.6 => /usr/lib/arm-linux-gnueabi/libImath.so.6 (0xb5e7e000)
	libIlmImf.so.6 => /usr/lib/arm-linux-gnueabi/libIlmImf.so.6 (0xb5dc9000)
	libIex.so.6 => /usr/lib/arm-linux-gnueabi/libIex.so.6 (0xb5da1000)
	libHalf.so.6 => /usr/lib/arm-linux-gnueabi/libHalf.so.6 (0xb5d4e000)
	libIlmThread.so.6 => /usr/lib/arm-linux-gnueabi/libIlmThread.so.6 (0xb5d37000)
	libgtk-x11-2.0.so.0 => /usr/lib/arm-linux-gnueabi/libgtk-x11-2.0.so.0 (0xb595b000)
	libgdk-x11-2.0.so.0 => /usr/lib/arm-linux-gnueabi/libgdk-x11-2.0.so.0 (0xb58b3000)
	libpangocairo-1.0.so.0 => /usr/lib/arm-linux-gnueabi/libpangocairo-1.0.so.0 (0xb5896000)
	libatk-1.0.so.0 => /usr/lib/arm-linux-gnueabi/libatk-1.0.so.0 (0xb5868000)
	libcairo.so.2 => /usr/lib/arm-linux-gnueabi/libcairo.so.2 (0xb575f000)
	libgdk_pixbuf-2.0.so.0 => /usr/lib/arm-linux-gnueabi/libgdk_pixbuf-2.0.so.0 (0xb572d000)
	libgio-2.0.so.0 => /usr/lib/arm-linux-gnueabi/libgio-2.0.so.0 (0xb55d9000)
	libpangoft2-1.0.so.0 => /usr/lib/arm-linux-gnueabi/libpangoft2-1.0.so.0 (0xb55b6000)
	libpango-1.0.so.0 => /usr/lib/arm-linux-gnueabi/libpango-1.0.so.0 (0xb5562000)
	libgobject-2.0.so.0 => /usr/lib/arm-linux-gnueabi/libgobject-2.0.so.0 (0xb5505000)
	libglib-2.0.so.0 => /lib/arm-linux-gnueabi/libglib-2.0.so.0 (0xb53fd000)
	libfontconfig.so.1 => /usr/lib/arm-linux-gnueabi/libfontconfig.so.1 (0xb53b8000)
	libfreetype.so.6 => /usr/lib/arm-linux-gnueabi/libfreetype.so.6 (0xb531e000)
	libgthread-2.0.so.0 => /usr/lib/arm-linux-gnueabi/libgthread-2.0.so.0 (0xb530c000)
	libgtkglext-x11-1.0.so.0 => /usr/lib/libgtkglext-x11-1.0.so.0 (0xb5301000)
	libgdkglext-x11-1.0.so.0 => /usr/lib/libgdkglext-x11-1.0.so.0 (0xb52a9000)
[... 55 lines cut ...]
	libopus.so.0 => /usr/lib/arm-linux-gnueabi/libopus.so.0 (0xb3736000)
	libopenjpeg.so.5 => /usr/lib/arm-linux-gnueabi/libopenjpeg.so.5 (0xb3705000)
	libmp3lame.so.0 => /usr/lib/arm-linux-gnueabi/libmp3lame.so.0 (0xb3683000)
	libgsm.so.1 => /usr/lib/arm-linux-gnueabi/libgsm.so.1 (0xb366f000)
	libva.so.1 => /usr/lib/arm-linux-gnueabi/libva.so.1 (0xb364b000)
	librtmp.so.1 => /usr/lib/arm-linux-gnueabi/librtmp.so.1 (0xb3622000)
	libgnutls-deb0.so.28 => /usr/lib/arm-linux-gnueabi/libgnutls-deb0.so.28 (0xb3511000)
[... cut ...]
	liborc-0.4.so.0 => /usr/lib/arm-linux-gnueabi/liborc-0.4.so.0 (0xb335c000)
	libhogweed.so.2 => /usr/lib/arm-linux-gnueabi/libhogweed.so.2 (0xb3322000)
	libnettle.so.4 => /usr/lib/arm-linux-gnueabi/libnettle.so.4 (0xb32e0000)
	libgmp.so.10 => /usr/lib/arm-linux-gnueabi/libgmp.so.10 (0xb326a000)
	libp11-kit.so.0 => /usr/lib/arm-linux-gnueabi/libp11-kit.so.0 (0xb3226000)
	libtasn1.so.6 => /usr/lib/arm-linux-gnueabi/libtasn1.so.6 (0xb3206000)
robot@ev3dev:~$ 
``` 

It seems that OpenCV is broken for both distros. :(

jabrena, 2018-05-20:
https://linux.die.net/man/8/ldconfig

JakubVanek, 2018-05-20:
Hmmm, would symlinking `libpng16.so.16` from `libpng16-16` to /usr/lib/jni/ help?
```
sudo apt-get install libpng16-16
sudo ln -s /usr/lib/arm-linux-gnueabi/libpng16.so.16 /usr/lib/jni/libpng16.so.16
```
or, ldconfig refresh:
```
sudo apt-get install libpng16-16
sudo ldconfig
```
or, adding the libpng dir to LD_LIBRARY_PATH:
```
sudo apt-get install libpng16-16
LD_LIBRARY_PATH="/usr/lib/arm-linux-gnueabi" <java command>
```

jabrena, 2018-05-20:
I am finishing the compilation on EV3, I will inform about the progress.
Crossing fingers :D

jabrena, 2018-05-20:
Fixed!!! 

```
robot@ev3dev:~$ java -Djava.library.path=/usr/lib/jni/ -cp /home/robot/opencv-0.1.0-SNAPSHOT-all.jar:/usr/share/OpenCV/java/opencv-249.jar example.opencv.HelloWorldCV2018-05-20 16:30:04 [main] INFO  example.opencv.HelloWorldCV - Welcome to OpenCV 2.4.9.1
2018-05-20 16:30:04 [main] INFO  example.opencv.HelloWorldCV - /usr/lib/jni/
2018-05-20 16:30:04 [main] INFO  example.opencv.HelloWorldCV - opencv_java249
2018-05-20 16:30:11 [main] INFO  example.opencv.HelloWorldCV - m = [1, 0, 0;
  0, 1, 0;
  0, 0, 1]
robot@ev3dev:~$ 
```

jabrena, 2018-05-20:
Solved doubt:

``` java
package example.opencv;

import org.opencv.core.Core;
import org.opencv.core.Mat;
import org.opencv.highgui.Highgui;
import org.opencv.highgui.VideoCapture;

/**
 * Multiple webcams
 *
 */
public class MultipleWebcam {

    public static void main( String[] args ) {
        System.out.println("Welcome to OpenCV " + Core.VERSION);
        System.loadLibrary(Core.NATIVE_LIBRARY_NAME);

        for(int i = 0; i < 3; i++) {

            VideoCapture camera = new VideoCapture(i);
            if (!camera.isOpened()) {
                System.out.println("Camera Error");
            }

            Mat frame = new Mat();
            camera.read(frame);
            System.out.println("Frame Obtained");
            System.out.println("Captured Frame Width " + frame.width());

            Highgui.imwrite("camera" + i +".jpg", frame);

            try {
                Thread.sleep(1000);
            } catch (InterruptedException e) {
            }
            System.out.println("Iteration:" + i);

            camera.release();
        }
    }
}
```

jabrena, 2018-05-30:
Doubt solved

closed 2018-05-30 by jabrena

JakubVanek, 2018-12-11:
Hmmm, we experienced the segmentation fault on Stretch even back then:

> Unfortunately for Debian Stretch, I found a runtime error on the dependency:
>
> ```
> robot@ev3dev:~$ java -Djava.library.path=/usr/lib/jni/ -jar opencv-0.1.0-SNAPSHOT-all.jar
> 2018-05-19 20:44:17 [...]
This is resolved by https://github.com/ev3dev-lang-java/ev3dev-lang-java/issues/625.

Then we had the issue on Jessie with libpng. This was probably solved by adding a patch which adds more search paths for JNI libraries ([JDK-8214332](https://bugs.openjdk.java.net/browse/JDK-8214332)).

JakubVanek, 2018-12-11:
OpenCV is confirmed to work out-of-the-box on Jessie as well: #509

jabrena, 2018-12-11:
You are OnFire like Luca Doncic in the NBA!!!
https://www.youtube.com/watch?v=m5969smjFRg