ev3dev-lang-java/ev3dev-lang-java #514: Debian Strech discovered as Jessie

aleksandra-majchrzak, 2018-06-12:
I downloaded Debian Strech version **snapshot-ev3dev-stretch-ev3-generic-2018-05-05** from ev3dev website, configured it and tried to run template example (the latest version from develop branch). It all went well, until EV3DevDistros class tired to determine my OS version by running `Shell.execute("cat /etc/os-release | grep jessie");` (line 23). 

The content of this file is:

```
PRETTY_NAME="ev3dev-stretch"
NAME="ev3dev-stretch"
ID=ev3dev
ID_LIKE=debian
HOME_URL="http://www.ev3dev.org"
SUPPORT_URL="http://www.ev3dev.org/support"
BUG_REPORT_URL="https://github.com/ev3dev/ev3dev/issues"
```

and when I run this commant in my terminal, I get nothing, but when it's executed in the program it returns the whole file content, and therefore detects my system as Debian Jessie, which leads to further errors, and finally to a crash.

I really don't get why the result of bash execution is different in the terminal and the program, and would be really gratefull I anybody advised me what can be done about it.

jabrena, 2018-06-12:
Hi @aleksandra-majchrzak,

Many thanks for the issue. I will replay in detail in the next days because the issue has different questions.

In Debian Stretch, I recommend to use the stable version:

```
compile("com.github.ev3dev-lang-java:ev3dev-lang-java:2.3.0")
```

Sorry by the confusion. That version was tested with this Debian Stretch snapshot:
https://oss.jfrog.org/list/oss-snapshot-local/org/ev3dev/brickstrap/2018-03-16/snapshot-ev3dev-jessie-ev3-generic-2018-03-16.img.xz

Take a look this issue:
https://github.com/ev3dev-lang-java/ev3dev-lang-java/issues/482

Currently, I am working in a new version compatible with Debian Jessie & Debian Stretch:

```
compile("com.github.ev3dev-lang-java:ev3dev-lang-java:2.4.6-SNAPSHOT")
```

But current iteration didn´t finish. :)

Note: The default EV3Dev image is Debian Jessie.

```
compile("com.github.ev3dev-lang-java:ev3dev-lang-java:0.7.0")
```

Juan Antonio

jabrena, 2018-06-13:
Hi @aleksandra-majchrzak, did you solve your execution?

aleksandra-majchrzak, 2018-06-13:
Sorry for the delay, setting ev3dev-lang-java to 2.3.0 solved the problem. Thank you. For the future users, it would be great if you (in addition to info about using develop branch for Stretch) added info about preffered ev3dev-lang-java version in the first project tutorial.

closed 2018-06-13 by aleksandra-majchrzak

jabrena, 2018-06-13:
Hi @aleksandra-majchrzak, totally agree with your comment.
Tomorrow, I will do the change in the template project.

I would appreciate if you can test in the last snapshot a LCD example. Recently `EV3Dev` driver changed and I am not sure if something with the EV3 Display is broken.

https://github.com/ev3dev-lang-java/examples/tree/master/ev3dev-lang-java/src/main/java/ev3dev/actuators

``` java

import lejos.hardware.lcd.GraphicsLCD;
import lejos.robotics.Color;

public class LCDTest {

    public static GraphicsLCD lcd = LCD.getInstance();

    public static void main(final String[] args){

        System.out.println("EV3 LCD Example");

        lcd.setColor(Color.BLACK);
        lcd.drawRect(0,0, lcd.getWidth(), lcd.getHeight());
        lcd.fillRect(0,0, lcd.getWidth(), lcd.getHeight());

        lcd.setColor(Color.WHITE);
        lcd.drawString("Hello World", (lcd.getWidth()/ 2)-30, lcd.getHeight()/2, 0);
        lcd.refresh();
    }

    public static void clear(){
        lcd.setColor(Color.WHITE);
        lcd.fillRect(0,0, lcd.getWidth(), lcd.getHeight());
    }
}
```

I like your critic spirit. Feel free to add any new idea.
Did you install OpenJDK 10?

Juan Antonio

aleksandra-majchrzak, 2018-06-13:
Hi,
I'm not sure if you meant latest snapshot of ev3dev-lang-java (which does not work for me because of the OS detection problem) or snapshot of ev3dev (which also has some issue with host/hostnames). I use ev3dev previous version from 2018-05-05.

With my current configuration I ran the program, and nothing was displayed on the screen (I assume it should appear on the ev3 built-in sreen), but I may missed something in the configuration.

Please, specify your question, and correct me if I'm wrong with my assumptions.

I have OpenJDK 10 installed:   openjdk version "10" 2018-03-20.

jabrena, 2018-06-13:
Hi @aleksandra-majchrzak, I talked about the EV3Dev snapshot.

So... the new EV3Dev driver is not compatible with previous implementation.
https://github.com/ev3dev-lang-java/ev3dev-lang-java/issues/494

Nice, you are using latest JVM.

What is your interest in the project?

Cheers

jabrena, 2018-06-16:
Hi @aleksandra-majchrzak, 

What is the purpose of the project?
Simple curiosity.

If you need to use OpenJDK11, exist an early build ready here:
https://ci.adoptopenjdk.net/view/ev3dev/job/openjdk-11-ev3-ea/
https://ci.adoptopenjdk.net/view/ev3dev/job/openjdk-11-ev3-ea/lastSuccessfulBuild/artifact/build/jdk-ev3.tar.gz

Juan Antonio