ev3dev-lang-java/ev3dev-lang-java #516: Review the idea avbout executable jars for linux

jabrena, 2018-06-18:
https://coderwall.com/p/ssuaxa/how-to-make-a-jar-file-linux-executable

JakubVanek, 2018-06-22:
Hmmm, that is a great idea! We then wouldn't need separate shell script wrapper for each jar file.

jabrena, 2018-06-23:
Yes, exactly.
I suppose that the implementation in gradle should not be very hard

JakubVanek, 2018-11-12:
I was thinking how we can run JARs from Brickman:
1. The new Java package includes a file for a so called binfmt_misc, which, when combined with binfmt-support package from debian, can directly execute java jars (like ./hello.jar), given that they have the executable flag set.
2. openjdk-system integration, but now I'm wondering if it is not unnecessarily complicated.
3. Creating a shell script in template_project_gradle which would work as a launcher for a jar somewhere not directly in /home/robot.

(3) seems the simplest, the user could also edit the wrapper easily. There might however be problems with that the icon in brickman != the jar itself, but I'm not sure how real such a scenario is.

jabrena, 2018-11-15:
Yes, we encapsulate the functions in installer and later call from template project.

I will this weekend a prototype. Maybe later, I will need your help with the call from Installer to execute the AppCDS. 

Using this technique, any user who deploy a HelloWorld should execute in less than 5 seconds. From the current 60 seconds.

jabrena, 2018-11-15:
I will test adding the following JVM flags:

`-noverify -XX:TieredStopAtLevel=1`

Source: http://presos.dsyer.com/decks/how-fast-is-spring.html

JakubVanek, 2018-12-02:
Delivered via shell script wrapper.

closed 2018-12-02 by JakubVanek