ev3dev-lang-java/ev3dev-lang-java #517: Installer Script Fails On EV3 Due To Typo In Platform Detection

MindS1, 2018-07-25:
**Platform**: EV3
**OS**: ev3dev based on Debian Jessie
**Problem**: After setting up an EV3 install and following the instructions shown [here](https://github.com/ev3dev-lang-java/installer), running the latest installer.sh fails with "Platform detected: unknown".  

[image]  

**Solution**
I believe the issue can be resolved by fixing a minor **typo** in the "vars.sh" installer module: [https://raw.githubusercontent.com/ev3dev-lang-java/installer/develop/modules/vars.sh](https://raw.githubusercontent.com/ev3dev-lang-java/installer/develop/modules/vars.sh)  

Currently:
BATT_EV3="/sys/class/power_supply/lego-ev3-battery"
Should be:
BATT_EV3="/sys/class/power_supply/legoev3-battery"

Can be verified by navigating to that directory on the EV3:
[image]

jabrena, 2018-08-02:
Hi @MindS1, sorry by the delay I didnt see the alert.
Try to use master branch. Master is for Jessie, Develop is for Stretch.

jabrena, 2018-08-02:
And welcome to this project with your WIN environment :)

jabrena, 2018-08-03:
Did my comment solve your doubt?

MindS1, 2018-08-26:
Yes, thanks for figuring that one out. I wasn't aware there were different installer versions.

closed 2018-08-26 by MindS1