ev3dev-lang-java/ev3dev-lang-java #518: Howto java, brickpi and pi 3 model B+

mingan666, 2018-07-29:
So, I'm trying to get my new brickpi and Pi 3 model B+ to work with java.
First it seems that ev3dev jessie does not support the newest model B+.
Now it seems like ev3dev-lang-java only supports the brickpi on jessie.

So, what's the idea here? Does this mean the latest brickpi starter kit is not supported?
I mean since it comes with the latest model B+ pi.

What is missing in ev3dev to get this working?

jabrena, 2018-08-02:
Hi @mingan666, Sorry by the delay, I didn´t see the alert.

I didn´t have time, but I would like to add the support. 

Don´t worry, this weekend, I will review the latest branch to do some changes and together we will fix the minor issues to offer that support. Ok?

Cheers

Juan Antonio

mingan666, 2018-08-04:
That sounds great. 
I've tested the 2.4.6 snapshot and found that at least the platforms class contains errors (all motor ports are set to A). Also I found an error regarding the fake ev3dev tmp path (no slash, so it tries to create in the root).

I'm very busy the coming week but thereafter I should be able help out also. Just if I know where to start I might be able to contribute (when time allows). I found several snapshots but don't know what they are for or how to fork (only succeed to fork the main branch)

jabrena, 2018-08-06:
For the development, use the branch: 443 
https://github.com/ev3dev-lang-java/ev3dev-lang-java/tree/feature/%23443

This is the stable branch for Stretch with Dual support for Jessie & Stretch.

mingan666, 2018-08-13:
Which method should I use? Fork, new branch or clone and just push to the same branch?

mingan666, 2018-08-15:
Now I have fixed the problems I encountered earlier and I'll continue using that my local version to continue with my robot and see what other problems I might run into.

Please let me know any comments and how to continue with eventual merging of the fixes.

Regarding your questions:

* I use BrickPi3 together with Rasberry Pi 3 B+
* After your input regarding OpenJdk 10 I tested a tried lot and finally got it to work. Currently I'm using it both on the Pi and in my development.

```
Image file:         ev3dev-stretch-rpi2-generic-2018-07-26
Kernel version:     4.14.59-ev3dev-2.2.2-rpi2
Brickman:           0.10.0
BogoMIPS:           38.40
Board:              board0
BOARD_INFO_HW_REV=A020D3
BOARD_INFO_MODEL=Raspberry Pi 3 Model B Plus Rev 1.3
BOARD_INFO_SERIAL_NUM=000000003e6b500d
BOARD_INFO_TYPE=main
Board:              board1
BOARD_INFO_FW_VER=1.4.6
BOARD_INFO_HW_REV=3.2.1
BOARD_INFO_MODEL=Dexter Industries BrickPi3
BOARD_INFO_SERIAL_NUM=F247E348504D5741372E3120FF0E1F11
BOARD_INFO_TYPE=aux
```

jabrena, 2018-08-15:
This is the spirit!!!
Congratulations!!!

jabrena, 2018-08-15:
I will create a branch  and later you could do a pull request to merge your changes.

Cheers

jabrena, 2018-08-16:
Good morning @mingan666,

you have the branch available!
https://github.com/ev3dev-lang-java/ev3dev-lang-java/tree/feature/%23518

And thanks for your collaboration in the project.

- What is your interest with the usage of BrickPi?
- What kind of robots do you like to build?

Juan Antonio

mingan666, 2018-08-16:
I'm just playing around for now. Bought mindstorms this summer and quickly decided I wanted to use java. However, the ev3brick was way to slow for my patience.

jabrena, 2018-08-17:
Hi @mingan666, 

I was checking your previous comments, can you confirm that you are using OpenJDK10 in your brick?

mingan666, 2018-08-17:
Yes I am. However, I just started out so I haven't really used that much of the implementation yet.

HaginCodes, 2018-10-25:
I am wondering how I can install this on my brickpi 3.

jabrena, 2018-10-25:
Hi @HaginCodes,

This weekend, we will continue with the development in order to offer a snapshot to add this support.
I will explain how to use this library.

Cheers

Juan Antonio

jabrena, 2018-10-28:
@HaginCodes nice project:
https://github.com/HaginCodes/3x3x3-Rubiks-Cube-Solver

jabrena, 2018-10-28:
Hi @HaginCodes I merged main branch. Tomorrow I will add the tests for you and @mingan666 

Sorry by the delay.

Juan Antonio

jabrena, 2018-10-29:
Snapshot created for testing:
https://jitpack.io/#ev3dev-lang-java/ev3dev-lang-java
https://jitpack.io/#ev3dev-lang-java/ev3dev-lang-java/2.4.7-SNAPSHOT
https://jitpack.io/com/github/ev3dev-lang-java/ev3dev-lang-java/2.4.7-SNAPSHOT/build.log

jabrena, 2018-10-29:
@mingan666, @HaginCodes I have created repo with examples.

https://github.com/ev3dev-lang-java/ev3dev-java-systemtests

Tell me your doubts.

Juan Antonio

jabrena, 2018-11-19:
Building a testing environment for BrickPi+ & BrickPi3

jabrena, 2018-11-21:
Starting the testing process with latest EV3Dev image for Debian Stretch:

``` 
Linux ev3dev 4.14.80-ev3dev-2.3.1-rpi2 #1 SMP Tue Nov 13 11:59:48 CST 2018 armv7l
             _____     _
   _____   _|___ /  __| | _____   __
  / _ \ \ / / |_ \ / _` |/ _ \ \ / /
 |  __/\ V / ___) | (_| |  __/\ V /
  \___| \_/ |____/ \__,_|\___| \_/

Debian stretch on LEGO MINDSTORMS EV3!
```

jabrena, 2018-11-21:
http://www.hypexr.org/linux_scp_help.php

```
scp foobar.txt your_username@remotehost.edu:/some/remote/directory
```

jabrena, 2018-11-21:
```
robot@ev3dev:~$ sudo tar -zxvf OpenJDK11-jdk_arm_linux_hotspot_2018-11-09-18-48.tar.gz /opt 
robot@ev3dev:~$ cd /opt/
robot@ev3dev:/opt$ ls
jdk-11.0.1+13
robot@ev3dev:/opt$ sudo update-alternatives --install /usr/bin/java java /opt/jdk-11.0.1+13/bin/java 1011
update-alternatives: using /opt/jdk-11.0.1+13/bin/java to provide /usr/bin/java (java) in auto mode
robot@ev3dev:/opt$ java -version
openjdk version "11.0.1" 2018-10-16
OpenJDK Runtime Environment AdoptOpenJDK (build 11.0.1+13-201811091906)
OpenJDK Server VM AdoptOpenJDK (build 11.0.1+13-201811091906, mixed mode)
robot@ev3dev:/opt$ java -Xshare:dump
Allocated shared space: 268435456 bytes at 0x557b2000
Loading classes to share ...
Loading classes to share: done.
Rewriting and linking classes ...
Rewriting and linking classes: done
Number of classes 1245
    instance classes   =  1183
    obj array classes  =    54
    type array classes =     8
Updating ConstMethods ... done. 
Removing unshareable information ... done. 
Scanning all metaspace objects ... 
Allocating RW objects ... 
Allocating RO objects ... 
Relocating embedded pointers ... 
Relocating external roots ... 
Dumping symbol table ...
Relocating SystemDictionary::_well_known_klasses[] ... 
Removing java_mirror ... done. 
mc  space:      3044 [  0.0% of total] out of      4096 bytes [ 74.3% used] at 0x557b2000
rw  space:   2228892 [ 15.4% of total] out of   2232320 bytes [ 99.8% used] at 0x557b3000
ro  space:   5826752 [ 40.2% of total] out of   5828608 bytes [100.0% used] at 0x559d4000
md  space:      1280 [  0.0% of total] out of      4096 bytes [ 31.2% used] at 0x55f63000
od  space:   6416508 [ 44.3% of total] out of   6418432 bytes [100.0% used] at 0x55f64000
total    :  14476476 [100.0% of total] out of  14487552 bytes [ 99.9% used]
robot@ev3dev:/opt$ java -version
openjdk version "11.0.1" 2018-10-16
OpenJDK Runtime Environment AdoptOpenJDK (build 11.0.1+13-201811091906)
OpenJDK Server VM AdoptOpenJDK (build 11.0.1+13-201811091906, mixed mode, sharing)

jabrena, 2018-11-21:
```
ev3dev#1|<!-- Copy everything between these lines -->
ev3dev#1|**System info (from `ev3dev-sysinfo`)**
ev3dev#1|```
ev3dev#1|Image file:         ev3dev-stretch-rpi2-generic-2018-11-13
ev3dev#1|Kernel version:     4.14.80-ev3dev-2.3.1-rpi2
ev3dev#1|Brickman:           0.10.0
ev3dev#1|BogoMIPS:           38.40
ev3dev#1|Board:              board0
ev3dev#1|BOARD_INFO_HW_REV=A02082
ev3dev#1|BOARD_INFO_MODEL=Raspberry Pi 3 Model B Rev 1.2
ev3dev#1|BOARD_INFO_SERIAL_NUM=00000000b5d4af9e
ev3dev#1|BOARD_INFO_TYPE=main
ev3dev#1|Board:              board1
ev3dev#1|BOARD_INFO_FW_VER=1.4.0
ev3dev#1|BOARD_INFO_HW_REV=3.2.1
ev3dev#1|BOARD_INFO_MODEL=Dexter Industries BrickPi3
ev3dev#1|BOARD_INFO_SERIAL_NUM=9CD6576C514D355231202020FF072D17
ev3dev#1|BOARD_INFO_TYPE=aux
ev3dev#1|```
ev3dev#1|<!-- Copy everything between these lines -->
```

jabrena, 2018-11-21:
Executing full regression for Brickpi3 + Raspberry Pi 3:

**Actuators:** 

- [x] SoundDemo
- [x] LCDDrawIconsTest
- [x] LCDDrawImagesTest
- [x] LCDDrawLinesTest
- [x] LCDDrawRectanglesAnimationTest
- [x] LCDFontTest
- [x] LCDTest
- [x] LCDWriteTextTest
- [x] LEDExample
- [x] LargeMotorDemo
- [x] LargeMotorDemo2
- [x] LargeMotorDemo3 
- [x] LargeMotorStopModesExample
- [x] MediumMotorDemo
- [x] MediumMotorStopModesExample
- [x] MotorEncoderTest
- [x] MotorEventTest
- [x] MultipleMotorsDemo
- [x] MultipleMotorsStopDemo
- [x] NXTRegulatedMotorDemo
- [x] NXTRegulatedMotorStopModesExample 
- [x] RegulatedMotorRotateDemo
- [x] RegulatedMotorRotateToDemo
- [x] UnregulatedMotorDemo 
- [x] UnregulatedMotorDemo 
- [x] UnregulatedMotorDemo2 
- [x] UnregulatedMotorDemo3 
- [x] UnregulatedMotorDemo4 
- [ ] UnregulatedMutilpleMotorDemo 

**Sensors:**

- [x] BatteryDemo
- [x] ButtonExample
- [x] ButtonExample2
- [x] ButtonListenersExample
- [x] ColorSensorDemo
- [x] ColorSensorDemo2
- [x] ColorSensorDemo3
- [x] ColorSensorMultipleDemo
- [x] GyroSensorDemo1
- [x] GyroSensorDemo2
- [x] GyroSensorDemo3
- [x] IRSensorDemo1
- [x] IRSensorDemo2
- [x] IRSensorDemo3
- [x] IRSensorDemo4
- [x] USSensorDemo
- [x] USSensorDemo2
- [x] USSensorMultipleDemo
- [x] AbsoluteIMUAccelerationTest
- [x] AbsoluteIMUCompassTest
- [x] AbsoluteIMUGyroTest
- [x] AbsoluteIMUMagneticTest
- [x] AbsoluteIMUTiltTest
- [x] CreatePhotoDemo
- [x] CreatePhotoEV3Demo
- [x] FaceRecognitionDemo
- [x] FaceRecognitionDemo2
- [x] RecordVideoDemo
- [x] RectanglesDemo
- [x] TrackObjectsDemo

jabrena, 2018-11-21:
Related issue:
https://github.com/ev3dev/ev3dev/issues/1192

jabrena, 2018-11-22:
Reviewed support for BrickPi3 with Debian Stretch

jabrena, 2018-11-22:
Is pending to run the same tests with Raspberry Pi 3+

jabrena, 2018-11-24:
After some tests between BrickPi+ & BrickPi3, the behaviour is similar with Raspberry Pi 3 & Raspberry Pi 3 Model B +, the unique difference, is the speed.

In BrickPi+ appear some problems in some areas. In order to improve the quality, I am going to test again the Brick.

The test to repeat are:

- No support for Unregulated motors (RCX Motors)
- No support for rotate & rotateTo methods
- GyroSensor: Using the mode `gyroSensor.getAngleMode();` don't process the sensor properly. Example: GyroSensorDemo
- GyroSensor: Using the mode `gyroSensor.getAngleAndRateMode();` don't process the sensor properly. Example: GyroSensorDemo3
- IRSensor: Using the mode `ir1.getDistanceMode();` don't process the sensor properly. Example: IRSensorDemo
- IRSensor: Using the mode `ir1.getSeekMode();` don't process the sensor properly. Example: IRSensorDemo2
- IRSensor: Using the mode `ir1.getRemoteMode();` don't process the sensor properly. Example: IRSensorDemo3 
- Touch Sensor: returns Random values. Example: TouchSensorDemo
- USSensor: Not reliable the values returned in mode `us1.getDistanceMode();`. Example: USSensorDemo
- USSensor: Multiple USSensor. One listening the another one emitting US signal. Example: MultipleUSSensors
- I2C Sensor problem. Example: AbsoluteIMUAccelerationTest

jabrena, 2018-11-24:
Adding support for unregulated motors:
https://github.com/ev3dev-lang-java/ev3dev-lang-java/issues/591

jabrena, 2018-11-24:
Tested with success.

In the next release, we will certificate BrickPi 3 for EV3Dev Debian Stretch

closed 2018-11-24 by jabrena