ev3dev-lang-java/ev3dev-lang-java #519: Execute openjdk_regression & systemtests for OpenJDK 11 build on armv5tejl

jabrena, 2018-08-11:
It is necessary to execute openjdk_regression & systemtests for  OpenJDK 11 build.

jabrena, 2018-08-11:
OpenJDK 11, call the kernel method: `getcpu(2)`

```
[TEST]  Configuring JDK.
update-alternatives: using /opt/jdktest/jdk/bin/java to provide /usr/bin/java (java) in auto mode
qemu: Unsupported syscall: 345
Error occurred during initialization of VM
getcpu(2) system call not supported by kernel
```
But **QEmu** has not implemented yet.
https://github.com/resin-io/qemu/issues/13
https://wiki.qemu.org/ChangeLog/

jabrena, 2018-08-11:
@JakubVanek, does exist some way to provide a fake method `getcpu(2)` with a fixed value?
For EV3, the number of CPUs is 1.

jabrena, 2018-08-11:
Blocked execution of regression tests for OpenJDK11:
https://ci.adoptopenjdk.net/view/ev3dev/job/openjdk11_openjdktest_ev3_linux/

JakubVanek, 2018-08-12:
Hmmm: https://wiki.qemu.org/ChangeLog/2.12

jabrena, 2018-08-12:
Related issue:
https://github.com/ev3dev/ev3dev/issues/1148

JakubVanek, 2018-08-12:
QEMU 2.12 installed on [feature/openjdk10-full](https://github.com/ev3dev-lang-java/openjdk-ev3-test/tree/feature/openjdk10-full) branch in [2a66a63a42b382494392662a390d07c98686e97e](https://github.com/ev3dev-lang-java/openjdk-ev3-test/commit/2a66a63a42b382494392662a390d07c98686e97e). We can cherry-pick the commit to other branches too.
```
robot@cd9496e744c1:/$ qemu-arm-static --version
qemu-arm version 2.12.0 (Debian 1:2.12+dfsg-3ubuntu3)
Copyright (c) 2003-2017 Fabrice Bellard and the QEMU Project developers
```

jabrena, 2019-08-18:
Late

closed 2019-08-18 by jabrena