ev3dev-lang-java/ev3dev-lang-java #520: Test openjdk-system integration

JakubVanek, 2018-08-13:
Hi,

I think I've reached a point in openjdk-system where we can use it.

Repo URL: https://github.com/ev3dev-lang-java/openjdk-system
Help: not yet, but after installing the package you can read manpages: `man elj-run`, `man elj-shjar`. Also, the sources in `<repo>/root/usr/bin/elj-run` and `<repo>/root/usr/bin/elj-shjar` have help blocks in them.
Release URL: https://github.com/ev3dev-lang-java/openjdk-system/releases/tag/debian%2F1.0.2-2

intercept from `elj-run` manpage:
```
NAME
       elj-run - run java program in ev3dev-lang-java environment

SYNOPSIS
       elj-run  [--help]  [--program  jar]  [--mod modname] [--java-exe path]
       [--extra-jvm-flags  flags]  [--extra-class-path  paths]  [--extra-mod‐
       ule-path paths] [--cache-mode mode] [--appcds-name name] [--appcds-dir
       dir] [--appcds-flush] -- [JAR OPTIONS ...]

DESCRIPTION
       elj-run is a main  program  for  running  plain  java  programs  under
       ev3dev-lang-java.   Given the right parameters, it wraps the java com‐
       mand with appropriate flags.  When some parameters are not  specified,
       their values are taken from the system configuration.

       First  task  this  program  is for is gluing together ev3dev-lang-java
       libraries and user programs.  Library paths  are  loaded  from  system
       configuration  and  combined  with  user-provided  additional  library
       paths.

       Second problem this program solves is managing Java class caches.  EV3
       is  quite slow when starting Java, so this wrapper automatically turns
       on CDS caching.  To achieve further speedup, AppCDS  caching  mode  is
       available.   It caches all classes loaded on the first program startup
       for future loads.
```
elj-shjar:
```
NAME
       elj-shjar - create executable wrapper for a jar file

SYNOPSIS
       elj-run  [--help]  [--program jar] [--output sh] [--extra-program-args
       args] [MORE ELJ-RUN OPTIONS ...]

DESCRIPTION
       elj-shjar is an utility for wrapping  JARs  with  small  shell  script
       wrapper.  When this wrapped program is executed, the wrapper runs elj-
       run(1) on itself with flags saved in this very script. The  configura‐
       tion is embedded in from -B elj-shjar command line parameters.

       The wrapper is documented under elj-stub(7) manual entry.
```
elj-stub:
```
NAME
       stub.sh - jar wrapper for ev3dev-lang-java

SYNOPSIS
       stub.sh [JAR OPTIONS ...]

DESCRIPTION
       stub.sh  is a small shell script template that is preprocessed by elj-
       shjar(1) and then a JAR file is appended to it.  This script  provides
       minimal  interface for running and extracting its payload and flushing
       the AppCDS cache.
```

JakubVanek, 2018-08-13:
This moves us forward on #441, #444.
#445 is almost finished with appcds mode, even for fat jars.
#516 is almost fixed, but support from template-project gradle stuff is needed.

jabrena, 2018-08-13:
I will test it this week.

JakubVanek, 2018-08-13:
Note: the integration is built with Java 11 in mind. It will be necessary to test it with JRI 11. To use it, just extract it somewhere, update `JAVA_EXE` in `/etc/elj.d/01-default-flags` and then run /path/to/jri11/bin/java -Xshare:dump to initialize CDS cache.

JakubVanek, 2018-08-13:
If support for Java <11 is needed, it shouldn't be problematic (I will just need to add a few additional flags to the java invocations).
The only problem is that AppCDS won't work for modules: https://github.com/ev3dev-lang-java/ev3dev-lang-java/issues/441#issuecomment-412467906

jabrena, 2018-08-13:
Don´t worry, the project will use OpenJDK 11 JRI and fortunately last week, we could build it.

jabrena, 2018-11-03:
In the next release, we will add this feature. First we need to deliver openjdk in ev3dev.

JakubVanek, 2018-11-12:
This might not be needed, see #516

closed 2018-11-12 by JakubVanek

jabrena, 2018-11-12:
We will review in the future.
You developed a good POC.