ev3dev-lang-java/ev3dev-lang-java #521: Create an Architecture Decision Record document

jabrena, 2018-08-13:
After reading this tweet:
https://twitter.com/nicolas_frankel/status/1029093247497461761

I noticed that the project should include some kind of technical documentation to store some decision taken in the project.

JakubVanek, 2018-09-07:
ADR note: to have a simple solution for printing logs when running from Brickrun (#494), use of `System.out` is discouraged. `System.err` or SLF4J should be used instead.

jabrena, 2018-09-07:
Allrighty

jabrena, 2018-11-04:
I will follow the following structure:
https://github.com/joelparkerhenderson/architecture_decision_record

- /docs/adr/adr_name.md

JakubVanek, 2018-11-04:
I added the Sensor ADR and the Display API ADR. I also tweaked a bit the other ADRs.

jabrena, 2018-11-09:
Added

closed 2018-11-09 by jabrena