ev3dev-lang-java/ev3dev-lang-java #523: LeJOS Compatibility Code for a motor object

abaumgart, 2018-08-14:
The old uses a motor.close() method.
The method is with not available under ev3dev

jabrena, 2018-08-14:
Hi Andreas,

Can you paste the example where you use that method?

Juan Antonio

abaumgart, 2018-08-14:
package colorSensor;

import ev3dev.actuators.lego.motors.EV3LargeRegulatedMotor;
import ev3dev.sensors.ev3.EV3ColorSensor;
import lejos.hardware.Key;
import lejos.hardware.port.MotorPort;
import lejos.hardware.port.SensorPort;
import lejos.utility.Delay;

public class Ev3devColorSensorRedMode {
	public static void main(String[] args) {
		EV3ColorSensor colorSensor = new EV3ColorSensor(SensorPort.S1);	
		float[] sample = new float[colorSensor.sampleSize()];
		EV3LargeRegulatedMotor motor = new EV3LargeRegulatedMotor(MotorPort.A);
		Key esc = LocalEV3.get().getKey("Escape");			
		colorSensor.setCurrentMode("Red");
		motor.setSpeed(80);
			
		while(esc.isDown()!=true){			
			colorSensor.fetchSample(sample, 0);	
			System.out.println(sample[0]);
			if( sample[0] > 0.5 ){
				motor.stop();
			}
			else {
				motor.forward();
			}
			Delay.msDelay(1000);
		}
		motor.close();
	}
}

jabrena, 2018-08-15:
It is a good question @abaumgart, at the moment, all operations interact with `EV3Dev` runtime interface using this class:

https://github.com/ev3dev-lang-java/ev3dev-lang-java/blob/develop/src/main/java/ev3dev/utils/Sysfs.java
https://github.com/ev3dev-lang-java/ev3dev-lang-java/blob/develop/src/main/java/ev3dev/sensors/Battery.java

If you observe, the class provide some Static methods. In the future, maybe the objects provided will maintain the state of the files operated to not have to reopen the files but at the moment, we open and close. In the future, we could add that feature.

Anyway, provide this method for the next iteration will not generate high effort. 
I will add it to not break your logic.

Juan Antonio

jabrena, 2018-08-16:
Hi @abaumgart, I was checking on internet other cases. The method close is not necessary at the moment in current implementation. If I add for motors, I would have to add in all sensors too and the impact of the change will increase. We are going to add immediately.

When sensors and motors manage state, we will add that method.
At the moment, it is not necessary and if you need to run the example, you should remove that method.

**References:**
https://bricks.stackexchange.com/questions/4680/how-to-close-color-sensor-properly
https://lejos.sourceforge.io/forum/viewtopic.php?t=7827

Juan Antonio

closed 2018-08-18 by jabrena