ev3dev-lang-java/ev3dev-lang-java #524: LeJOS compatibility > LocalEV3-Class not available

abaumgart, 2018-08-14:
After transferring  the following source code from LeJOS to ev3dev the LocalEV3-Class is not compatible.

package colorSensor;

import ev3dev.actuators.lego.motors.EV3LargeRegulatedMotor;
import ev3dev.sensors.ev3.EV3ColorSensor;
import lejos.hardware.Key;
import lejos.hardware.port.MotorPort;
import lejos.hardware.port.SensorPort;
import lejos.utility.Delay;

public class Ev3devColorSensorRedMode {
	public static void main(String[] args) {
		EV3ColorSensor colorSensor = new EV3ColorSensor(SensorPort.S1);	
		float[] sample = new float[colorSensor.sampleSize()];
		EV3LargeRegulatedMotor motor = new EV3LargeRegulatedMotor(MotorPort.A);
		**Key esc = LocalEV3.get().getKey("Escape");**			
		colorSensor.setCurrentMode("Red");
		motor.setSpeed(80);
			
		while(esc.isDown()!=true){			
			colorSensor.fetchSample(sample, 0);	
			System.out.println(sample[0]);
			if( sample[0] > 0.5 ){
				motor.stop();
			}
			else {
				motor.forward();
			}
			Delay.msDelay(1000);
		}
		motor.close();
	}
}

jabrena, 2018-08-15:
Hi @abaumgart, 

At the moment, we don´t have support for the class:

`LocalEV3.get()` 

because that class normally is related with Remote operations. With current classes, you can do the same stuff.

When we have the remote support, you could use that class and others like BrickFinder, Brick or RemoteEV3.

But it is true that it could be a good enhancement.

We will add in the future but you can develop the same logic using directly the class in charge of the Keys for EV3.

Juan Antonio

jabrena, 2019-08-18:
No feedback
Rejected by LEAN

closed 2019-08-18 by jabrena