ev3dev-lang-java/ev3dev-lang-java #525: Review units for method: setSpeed(int speed)

jabrena, 2018-08-15:
http://ev3dev-lang-java.github.io/docs/api/latest/ev3dev-lang-java/ev3dev/actuators/lego/motors/BaseRegulatedMotor.html

```
public void setSpeed(int speed)
Sets desired motors speed , in degrees per second; The maximum reliably sustainable velocity is 100 x battery voltage under moderate load, such as a direct drive robot on the level.
Parameters:
speed - value in degrees/sec
```

http://docs.ev3dev.org/projects/lego-linux-drivers/en/ev3dev-jessie/motors.html#tacho-motor-subsystem

```
speed | read-only | Returns the current motor speed in tacho counts per second. Note, this is not necessarily degrees (although it is for LEGO motors). Use the count_per_rot attribute to convert this value to RPM or deg/sec.
```

jabrena, 2018-08-17:
**Business Rule:**
It is necessary to create some kind of input validation to avoid values greater than 900.

JakubVanek, 2018-08-18:
Please do not do that. It is possible to achieve higher speeds. I have tried to mix 3 NiMH batteries with 3 NiZn batteries and the motors really had higher peak speeds. I have not measured the RPM, but the buzzing noise had higher pitch. If the speed is higher than maximum, the motor regulator will probably keep them on 100 % of time, which is OK.

jabrena, 2018-08-18:
Oki.

The essence of this issue was to review the units sent from LeJOS interface to EV3Dev runtime but everything is Ok. I was checking it because exist some possible projects in the air to interact with SLAM subsystems and it is important the accuracy between `tacho counts`and other metrics.

Maybe, I should invest more time in documentation about Motor theory for Differential model.

jabrena, 2018-10-28:
Reviewed

closed 2018-10-28 by jabrena