ev3dev-lang-java/ev3dev-lang-java #528: Debian packaging

JakubVanek, 2018-08-18:
There are few things to do before we can install ev3dev-lang-java packages on the brick using apt-get:

- [ ] Provide a package build environment
- this could be AdoptOpenJDK build servers triggered by a new tag push
- [ ] Provide a hosting for the repository
- this could be our own repository, quick google for "apt repository hosting" returned https://packagecloud.io/l/apt-repository and https://www.aptly.info/
- [ ] Provide a way to upload new packages from the build environment to the hosting
- this is yet to be googled

jabrena, 2018-08-19:
Hi @JakubVanek, 

I think that if we follow the documentation provided by EV3Dev, we could create the Debian package for the JVM:
https://www.ev3dev.org/docs/devtools/packaging-for-ev3dev/
https://github.com/ev3dev/ev3dev/issues/1099

We could start with a basic stuff and later, try to evolve with more complex mechanism.

For the JVM if we create a Shell Script which download the JRI and we add a Bash script to install the JVM in the right path, we could have the solution.

Juan Antonio

jabrena, 2018-10-28:
We should return to this point

JakubVanek, 2018-11-11:
* Build environment: AdoptOpenJDK Jenkins.
* Hosting: probably ev3dev's debian repository.
* Uploading new packages: manually, because they have to be signed by a GPG key.

JakubVanek, 2018-11-11:
\+ let's see what happens in https://ci.adoptopenjdk.net/view/ev3dev/job/openjdk11_build_ev3_linux/44/

jabrena, 2018-11-11:
Later, in the next evolution, we should decouple the build with the debian package, it is a different build.

jabrena, 2018-11-12:
Today, I will test the build from Jenkins:
https://ci.adoptopenjdk.net/view/ev3dev/job/openjdk11_build_ev3_linux/

JakubVanek, 2018-11-12:
Hi,

these dependencies are required and the reason isn't bad:
* `ca-certificates-java` - these are SSL certificates for Java. We were taking them from AdoptOpenJDK until now, so they just come from a separate, maintained package.
* `java-common` - this is the code that supports switching Java versions using `update-java-alternatives`. It probably provides also some basic functionality common to all Java implementations.
* libnss3 - this is a crypto library, if I understand correctly. Maybe it is not needed, but the Debian package included it. I can remove it, if it is not needed.

To reinstall the package, you will have to uninstall it through `sudo apt purge jri-11-ev3` and then install it again through APT.

Regards

JakubVanek, 2018-11-12:
Hmmm, maybe to clarify:
The `debian.zip` package is not intended for the end-user download. It is intended for the package uploader, who can upload it to ev3dev repository. From there, only the non-dbgsym `.deb` package will be downloaded (this is around 15 MB).

jabrena, 2018-11-12:
Oki, I continue with the testing process :)

JakubVanek, 2018-11-12:
Hmmm, `libnss3` is an order of magnitude bigger than the rest of them, I will remove this dependency. Standalone JRI worked well even without it. In this case, the dependency really doesn't add much value (like I think the remaining two do).

Just beware: you might be using the JRI in /opt. You can switch to the system one using `update-java-alternatives -s jri-11-ev3`.

jabrena, 2018-11-12:
I tested that JVM installed is from the debian package:

```
robot@ev3dev:~/debian/pkg$ sudo apt install --no-install-recommends  ./jri-11-ev3_11.0.1~13-1_armel.deb 
```

```
robot@ev3dev:~/debian/pkg$ java -version
openjdk version "11.0.1" 2018-10-16
OpenJDK Runtime Environment (build 11.0.1+13-ev3)
OpenJDK Client VM (build 11.0.1+13-ev3, mixed mode, sharing)
```

jabrena, 2018-11-12:
```
robot@ev3dev:~$ sudo update-alternatives --config java
[sudo] password for robot: 
There are 3 choices for the alternative java (providing /usr/bin/java).

  Selection    Path                              Priority   Status
------------------------------------------------------------
  0            /usr/lib/jvm/jri-11-ev3/bin/java   1101      auto mode
* 1            /opt/jri-11/bin/java               11        manual mode
  2            /opt/jri-12/bin/java               12        manual mode
  3            /usr/lib/jvm/jri-11-ev3/bin/java   1101      manual mode
```

JakubVanek, 2018-11-12:
Hmmm, I noticed that jexec is broken:
```
robot@ev3dev:~/debian/pkg$ jexec
can't locate java: No such file or directory
```
I will have to look at it later.

jabrena, 2018-11-12:
Well, for the next fix.

The most important commands are working:

- java
- rmiregistry
- keytool
- rmid

jabrena, 2018-11-12:
Don't worry about `jexec` it is a Dinosaur.
https://bugs.openjdk.java.net/browse/JDK-8176066

jabrena, 2018-11-30:
Testing Snapshot version:
https://oss.jfrog.org/list/oss-snapshot-local/org/ev3dev/brickstrap/2018-11-29/

jabrena, 2018-11-30:
```
Linux ev3dev 4.14.80-ev3dev-2.3.1-ev3 #1 PREEMPT Tue Nov 13 13:08:11 CST 2018 armv5tejl
             _____     _
   _____   _|___ /  __| | _____   __
  / _ \ \ / / |_ \ / _` |/ _ \ \ / /
 |  __/\ V / ___) | (_| |  __/\ V /
  \___| \_/ |____/ \__,_|\___| \_/

Debian stretch on LEGO MINDSTORMS EV3!
robot@ev3dev:~$ java -version
-bash: java: command not found
robot@ev3dev:~$ cd /usr/lib/jvm/
-bash: cd: /usr/lib/jvm/: No such file or directory
robot@ev3dev:/etc$ dpkg --list | grep jri
robot@ev3dev:/etc$ 
```

jabrena, 2018-12-01:
Today I will test the new Snapshot:
https://oss.jfrog.org/list/oss-snapshot-local/org/ev3dev/brickstrap/2018-12-01/

jabrena, 2018-12-01:
Tested.

```
> Task :ev3devInfo
Strict host key checking is off. It may be vulnerable to man-in-the-middle attacks.
ev3dev-sysinfo -m

ev3dev#25|<!-- Copy everything between these lines -->
ev3dev#25|**System info (from `ev3dev-sysinfo`)**
ev3dev#25|```
ev3dev#25|Image file:         ev3dev-stretch-ev3-generic-2018-11-30
ev3dev#25|Kernel version:     4.14.80-ev3dev-2.3.1-ev3
ev3dev#25|Brickman:           0.10.0
ev3dev#25|BogoMIPS:           227.32
ev3dev#25|Board:              board0
ev3dev#25|BOARD_INFO_HW_REV=7
ev3dev#25|BOARD_INFO_MODEL=LEGO MINDSTORMS EV3
ev3dev#25|BOARD_INFO_ROM_REV=6
ev3dev#25|BOARD_INFO_SERIAL_NUM=001653413CAE
ev3dev#25|BOARD_INFO_TYPE=main
ev3dev#25|```
ev3dev#25|<!-- Copy everything between these lines -->
``` 

```
Linux ev3dev 4.14.80-ev3dev-2.3.1-ev3 #1 PREEMPT Tue Nov 13 13:08:11 CST 2018 armv5tejl
             _____     _
   _____   _|___ /  __| | _____   __
  / _ \ \ / / |_ \ / _` |/ _ \ \ / /
 |  __/\ V / ___) | (_| |  __/\ V /
  \___| \_/ |____/ \__,_|\___| \_/

Debian stretch on LEGO MINDSTORMS EV3!
Last login: Fri Nov 30 21:32:17 2018 from 192.168.2.1
robot@ev3dev:~$ java -version 
openjdk version "11.0.1" 2018-10-16
OpenJDK Runtime Environment (build 11.0.1+13-ev3)
OpenJDK Client VM (build 11.0.1+13-ev3, mixed mode, sharing)
```

jabrena, 2018-12-01:
Pending for the next snapshot to test profiling features.

closed 2018-12-01 by jabrena