ev3dev-lang-java/ev3dev-lang-java #534: installer.sh doesn't detect my ev3

SpyceTewan, 2018-08-25:
I'm using a normal ev3 with ev3dev, but when I run installer.sh to install everything, the program can't detect the brick

jabrena, 2018-08-25:
Hi @SpyceTewan ,

What EV3Dev version are you using?

- Jessie
- Stretch

SpyceTewan, 2018-08-25:
Jessie

SpyceTewan, 2018-08-25:
Here is the log

--2018-08-25 23:38:56--  https://raw.githubusercontent.com/ev3dev-lang-java/installer/develop/modules/utilities.sh
Resolving raw.githubusercontent.com (raw.githubusercontent.com)... 151.101.0.133, 151.101.64.133, 151.101.128.133, ...
Connecting to raw.githubusercontent.com (raw.githubusercontent.com)|151.101.0.133|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: 748 [text/plain]
Saving to: 'utilities.sh'

utilities.sh        100%[=====================>]     748  --.-KB/s   in 0.007s

Last-modified header missing -- time-stamps turned off.
2018-08-25 23:38:58 (111 KB/s) - 'utilities.sh' saved [748/748]

expr: syntax error
 PLATFORM SECTION #

converted 'https://raw.githubusercontent.com/ev3dev-lang-java/installer/develop/modules/platform.sh' (ANSI_X3.4-1968) -> 'https://raw.githubusercontent.com/ev3dev-lang-java/installer/develop/modules/platform.sh' (UTF-8)
--2018-08-25 23:38:59--  https://raw.githubusercontent.com/ev3dev-lang-java/installer/develop/modules/platform.sh
Resolving raw.githubusercontent.com (raw.githubusercontent.com)... 151.101.0.133, 151.101.64.133, 151.101.128.133, ...
Connecting to raw.githubusercontent.com (raw.githubusercontent.com)|151.101.0.133|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: 642 [text/plain]
Saving to: 'platform.sh'

platform.sh         100%[=====================>]     642  --.-KB/s   in 0.007s

Last-modified header missing -- time-stamps turned off.
2018-08-25 23:39:02 (87.0 KB/s) - 'platform.sh' saved [642/642]

Platform detected: unknown

Sorry, this platform is not recognized by the installer.
This installer was designed for EV3Dev hardware.

Open a issue if the problem continues:
https://github.com/ev3dev-lang-java/ev3dev-lang-java/issues

root@ev3dev:/home/robot/java-installer# clear
root@ev3dev:/home/robot/java-installer# ./installer.sh

##############################
# EV3Dev-lang-java Installer #
##############################
# Last update: 2017/06/18    #
##############################

Init installer
converted 'https://raw.githubusercontent.com/ev3dev-lang-java/installer/develop/modules/utilities.sh' (ANSI_X3.4-1968) -> 'https://raw.githubusercontent.com/ev3dev-lang-java/installer/develop/modules/utilities.sh' (UTF-8)
--2018-08-25 23:39:21--  https://raw.githubusercontent.com/ev3dev-lang-java/installer/develop/modules/utilities.sh
Resolving raw.githubusercontent.com (raw.githubusercontent.com)... 151.101.0.133, 151.101.64.133, 151.101.128.133, ...
Connecting to raw.githubusercontent.com (raw.githubusercontent.com)|151.101.0.133|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: 748 [text/plain]
Saving to: 'utilities.sh'

utilities.sh        100%[=====================>]     748  --.-KB/s   in 0.008s

Last-modified header missing -- time-stamps turned off.
2018-08-25 23:39:25 (97.3 KB/s) - 'utilities.sh' saved [748/748]

expr: syntax error
 PLATFORM SECTION #

converted 'https://raw.githubusercontent.com/ev3dev-lang-java/installer/develop/modules/platform.sh' (ANSI_X3.4-1968) -> 'https://raw.githubusercontent.com/ev3dev-lang-java/installer/develop/modules/platform.sh' (UTF-8)
--2018-08-25 23:39:26--  https://raw.githubusercontent.com/ev3dev-lang-java/installer/develop/modules/platform.sh
Resolving raw.githubusercontent.com (raw.githubusercontent.com)... 151.101.0.133, 151.101.64.133, 151.101.128.133, ...
Connecting to raw.githubusercontent.com (raw.githubusercontent.com)|151.101.0.133|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: 642 [text/plain]
Saving to: 'platform.sh'

platform.sh         100%[=====================>]     642  --.-KB/s   in 0.006s

Last-modified header missing -- time-stamps turned off.
2018-08-25 23:39:28 (112 KB/s) - 'platform.sh' saved [642/642]

Platform detected: unknown

Sorry, this platform is not recognized by the installer.
This installer was designed for EV3Dev hardware.

Open a issue if the problem continues:
https://github.com/ev3dev-lang-java/ev3dev-lang-java/issues

SpyceTewan, 2018-08-25:
Here is the log:
##############################
# EV3Dev-lang-java Installer #
##############################
# Last update: 2017/06/18    #
##############################

Init installer
converted 'https://raw.githubusercontent.com/ev3dev-lang-java/installer/develop/modules/utilities.sh' (ANSI_X3.4-1968) -> 'https://raw.githubusercontent.com/ev3dev-lang-java/installer/develop/modules/utilities.sh' (UTF-8)
--2018-08-25 23:41:14--  https://raw.githubusercontent.com/ev3dev-lang-java/installer/develop/modules/utilities.sh
Resolving raw.githubusercontent.com (raw.githubusercontent.com)... 151.101.0.133, 151.101.64.133, 151.101.128.133, ...
Connecting to raw.githubusercontent.com (raw.githubusercontent.com)|151.101.0.133|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: 748 [text/plain]
Saving to: 'utilities.sh'

utilities.sh        100%[=====================>]     748  --.-KB/s   in 0.007s

Last-modified header missing -- time-stamps turned off.
2018-08-25 23:41:17 (104 KB/s) - 'utilities.sh' saved [748/748]

expr: syntax error
 PLATFORM SECTION #

converted 'https://raw.githubusercontent.com/ev3dev-lang-java/installer/develop/modules/platform.sh' (ANSI_X3.4-1968) -> 'https://raw.githubusercontent.com/ev3dev-lang-java/installer/develop/modules/platform.sh' (UTF-8)
--2018-08-25 23:41:18--  https://raw.githubusercontent.com/ev3dev-lang-java/installer/develop/modules/platform.sh
Resolving raw.githubusercontent.com (raw.githubusercontent.com)... 151.101.0.133, 151.101.64.133, 151.101.128.133, ...
Connecting to raw.githubusercontent.com (raw.githubusercontent.com)|151.101.0.133|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: 642 [text/plain]
Saving to: 'platform.sh'

platform.sh         100%[=====================>]     642  --.-KB/s   in 0.01s

Last-modified header missing -- time-stamps turned off.
2018-08-25 23:41:20 (64.8 KB/s) - 'platform.sh' saved [642/642]

Platform detected: unknown

Sorry, this platform is not recognized by the installer.
This installer was designed for EV3Dev hardware.

Open a issue if the problem continues:
https://github.com/ev3dev-lang-java/ev3dev-lang-java/issues

jabrena, 2018-08-26:
Oki, are you using installer from template project?

SpyceTewan, 2018-08-26:
I followed the Getting Started guide
```shell

cd /home/robot
mkdir installer
cd installer
wget -N https://raw.githubusercontent.com/ev3dev-lang-java/installer/master/installer.sh
chmod +x installer.sh
sudo ./installer.sh help
sudo ./installer.sh
```

I'm trying to install everything manually and will come back when I'm done

jabrena, 2018-10-29:
No feedback

closed 2018-10-29 by jabrena