ev3dev-lang-java/ev3dev-lang-java #535: LeJOS compatibility using EV3 Keys

abaumgart, 2018-09-05:
On LeJOS it is possible to run the following program:

- Controlling the direction and speed of a motor with the EV3 Keys.

Under EV3DEV Java:
The left, right, up and down keys has no reaction. 
The Enter and ESC Keys are working.

In case of running a java program on LeJOS, the screen a blank and empty.
In case of running the same program on EV3DEV the screen shows the EV3DEV menu.
When using the EV3 Keys, the user moves through the menu. 

`
package example;
import ev3dev.actuators.lego.motors.EV3LargeRegulatedMotor;
import ev3dev.sensors.Battery;
import ev3dev.sensors.EV3Key;
import lejos.hardware.port.MotorPort;
import lejos.utility.Delay;
public class MyFirstRobot {

    public static void main(final String[] args){

        System.out.println("Creating Motor A & B");
        final EV3LargeRegulatedMotor motorLeft = new EV3LargeRegulatedMotor(MotorPort.A);
        final EV3LargeRegulatedMotor motorRight = new EV3LargeRegulatedMotor(MotorPort.D);

        //To Stop the motor in case of pkill java for example
        Runtime.getRuntime().addShutdownHook(new Thread(new Runnable() {
            public void run() {
                System.out.println("Emergency Stop");
                motorLeft.stop();
                motorRight.stop();
            }
        }));

        System.out.println("Defining the Stop mode");
        motorLeft.brake();
        motorRight.brake();

        System.out.println("Defining motor speed");
        int motorSpeed = 200;
        motorLeft.setSpeed(motorSpeed);
        

                EV3Key esc = new EV3Key(EV3Key.BUTTON_ESCAPE);
		EV3Key left = new EV3Key(EV3Key.BUTTON_LEFT);
		EV3Key right = new EV3Key(EV3Key.BUTTON_RIGHT);
		EV3Key enter = new EV3Key(EV3Key.BUTTON_ENTER);
		EV3Key up = new EV3Key(EV3Key.BUTTON_UP);
		EV3Key down = new EV3Key(EV3Key.BUTTON_DOWN);
        
		boolean forward = true; // true=for; false=back 
        
     
		while(esc.isUp()==true){
        		
  
            
            if(enter.isDown()==true) {
        		System.out.println("Enter");
        		if(motorLeft.isMoving()==true) 
	        		{
	        			motorLeft.stop();		
	        		}
	        		else {
	        			
		        		if(forward==true)
		        			motorLeft.forward();
		        		else 
		        			motorLeft.backward();
	        		}
        		
        		if(up.isDown()==true)	{ 
            		motorLeft.setSpeed(motorLeft.getSpeed()+50);
            		System.out.println("Hoch: "+motorLeft.getSpeed());
            	}
            	if(down.isDown()==true) {
            		motorLeft.setSpeed(motorLeft.getSpeed()-50);
            		System.out.println("Runter: "+motorLeft.getSpeed());
            	}
            	if(left.isDown()==true) {
            		System.out.println("Links");
            		forward = false;
            		motorLeft.backward();
            	}
            	if(right.isDown()==true){
            		System.out.println("Rechts");
            		forward = true;
            		motorLeft.forward();
            	}
        	}       	
		}
		System.exit(0);
    }
}
`

jabrena, 2018-09-05:
Hi @jartu, 

Do you think that Key support that we have for EV3, supported this syntax?

abaumgart, 2018-09-05:
I tried with one. But with the same result.
On the SSH Console I can see, that right or left is pressed.

> ev3dev#3|2018-09-05 19:15:33 [Thread-2] DEBUG ev3dev.sensors.EV3Key - KeyType LEFT released

`if(Button.LEFT.isDown()) {
            		System.out.println("Links");
            		forward = false;
            		motorLeft.backward();
            	}
            	if(Button.RIGHT.isDown()){
            		System.out.println("Rechts");
            		forward = true;
            		motorLeft.forward();
            	}`

jabrena, 2018-09-05:
Can you execute the command: `ev3dev-sysinfo -m`?

abaumgart, 2018-09-05:
> robot@ev3dev:~$ ev3dev-sysinfo -m?
Image file:         ev3dev-jessie-ev3-generic-2017-09-14
Kernel version:     4.4.87-22-ev3dev-ev3
Board:              LEGO MINDSTORMS EV3 Programmable Brick
Revision:           0006
Brickman:           0.8.1
ev3devKit:          0.4.3
robot@ev3dev:~$

abaumgart, 2018-09-05:
robot@ev3dev:~$ ev3dev-sysinfo -m?
Image file:         ev3dev-jessie-ev3-generic-2017-09-14
Kernel version:     4.4.87-22-ev3dev-ev3
Board:              LEGO MINDSTORMS EV3 Programmable Brick
Revision:           0006
Brickman:           0.8.1
ev3devKit:          0.4.3
robot@ev3dev:~$

jabrena, 2018-09-05:
I was reviewing the history of the support and it is something Pending.

https://github.com/ev3dev-lang-java/ev3dev-lang-java/issues/273
https://github.com/ev3dev-lang-java/ev3dev-lang-java/issues/298

Maybe we didn't put the focus because are interacting the robot without any human interaction but the  use case that you say, it is pending.

We will add @abaumgart 

Juan Antonio

JakubVanek, 2018-09-06:
Hi,

I have checked it out and it seems that there is a bug in your code:
```java
while (esc.isUp()) {
    if (enter.isDown()) {
        ...
    }
}
```

The `*.isDown()` will work with this loop _only_ when ENTER is pressed, because only then is the button checked.

JakubVanek, 2018-09-06:
Regarding the offending EV3 menu: in branch [feature/#494](https://github.com/ev3dev-lang-java/ev3dev-lang-java/tree/feature/%23494) there is a support for automatically disabling Brickman menu. The only thing that is going to be necessary is to add this to your main class:
```java
private static final display = SystemDisplay.initializeRealDisplay();
```

JakubVanek, 2018-09-06:
I have one more tip: I think that using key listeners or something like `waitForAnyPress` is a much better way how to do this.

closed 2018-10-29 by JakubVanek

robl0377, 2018-11-08:
I am running through the example programs (ev3dev-lang-java /examples) and found that the EV3 Brick buttons are seen in both my currently running program and in the Brickman Program that is still active on the EV3 Brick display. The example program is:
```
package ev3dev.sensors.ev3;

import ev3dev.hardware.display.DisplayInterface;
import ev3dev.hardware.display.SystemDisplay;
import ev3dev.sensors.Button;

public class ButtonExample2 {

	@SuppressWarnings("unused")
	private static final DisplayInterface display = SystemDisplay.initializeRealDisplay();

	public static void main(final String[] args){

    	System.out.println("Extended Button Example");

        System.out.println("Press UP");
        Button.UP.waitForPress();
        System.out.println("Pressed UP");

        System.out.println("Press DOWN");
        Button.DOWN.waitForPress();
        System.out.println("Pressed DOWN");

        System.out.println("Press LEFT");
        Button.LEFT.waitForPress();
        System.out.println("Pressed LEFT");

        System.out.println("Press RIGHT");
        Button.RIGHT.waitForPress();
        System.out.println("Pressed RIGHT");

        System.out.println("Press ENTER");
        Button.ENTER.waitForPress();
        System.out.println("Pressed ENTER");

        System.out.println("Press ESCAPE");
        Button.ESCAPE.waitForPress();
        System.out.println("Pressed ESCAPE");
    }
}
```
In my version of the program I moved the escape button after the enter to avoid shutting down the Brick through the still active Brickman program.
Also I tried to use JakubVanek's suggestion from 6 Sep (initializeRealDisplay). However, the Brickman program still reacts to the button presses. 

Ideally, a user program running on the Brick should have full control of all EV3 Brick resources. Actually in 'classic' LeJOS this is the case: For example the display is empty and the buttons are interpreted by the user program only.

jabrena, 2018-11-08:
Hi,

I read you question but I didn't understand what is the problem that you have.
In the comment, you don't explain what is the way that you execute the program, version of the library and distro.

Can you create a new issue with all details?

Juan Antonio

JakubVanek, 2018-11-08:
Hi,

I think the problem is that brickman is still alive when running the program.

Could you please try running the program with brickrun?
```bash
brickrun java -jar ...
```
Alternatively, I think there was a time window when the `initializeRealDisplay` hack was broken too - the Brickman shutdown code for the SSH case was accidentally disabled. It might be fixed now in the `develop` branch.

Also, as we reconsidered the API, I would now recommend against using `initializeRealDisplay` and would recommend using `ev3dev.actuators.LCD` and `LCD.getInstance()` instead.

Regards

JakubVanek, 2018-11-08:
Hmmm, I have realized that maybe the Brickman class is broken...
```java
    public static String execute(final String command) { // command = "sudo systemctl stop brickman"

        log.debug("Command: {}", command);
		StringBuilder output = new StringBuilder();

		Process p;
		try {
			p = Runtime.getRuntime().exec(command);
			p.waitFor();
			
			BufferedReader reader = new BufferedReader(new InputStreamReader(p.getInputStream()));
			
			String line;
			while ((line = reader.readLine())!= null) {
				output.append(line).append("\n");
			}
			reader.close();
		} catch (IOException | InterruptedException e) {
			log.warn(e.getLocalizedMessage(), e);
			return COMMAND_ERROR_MESSAGE;
		}

		return output.toString();
	}
```
I'm not sure how sudo works with this when the user is not already root.

jabrena, 2018-11-08:
We are going to create a new issue. We are mixing different contexts.

The idea about Brickman is the right one. In the next issue, we will create.

**THIS IS ISSUE IS CLOSED.** (It is a pity that we don't have the button like Spring...)