ev3dev-lang-java/ev3dev-lang-java #541: 2D Graphs with Java

jabrena, 2018-10-31:
http://math.hws.edu/graphicsbook/c2/s5.html
http://web.mit.edu/jlai321/Public/old_class_files/1.00/LectureSlides/Lecture17.pdf
http://laurel.datsi.fi.upm.es/_media/docencia/cursos/java/java2d.pdf

JakubVanek, 2018-11-01:
You can work with the underlying Java2D graphics this way:
1. Get the system framebuffer: `JavaFramebuffer fb = LCD.getInstance().getFramebuffer();`
2. create a BufferedImage suitable for the screen by calling `BufferedImage img = fb.createCompatibleBuffer();`
3. get Graphics2D for the BufferedImage: `Graphics2D gfx = img.createGraphics();`
4. do the painting off-screen via `gfx`
5. write the buffer to the screen: `fb.flushScreen(img)`

For rendering in a loop, I think it will be sufficient to repeat (4) and (5).

jabrena, 2018-11-11:
https://www.codemiles.com/java/3d-cube-t8.html?mobile=on
https://github.com/bourgesl/cube/blob/master/src/main/java/screen/Cubes.java

jabrena, 2018-11-11:
https://stackoverflow.com/questions/22549690/how-to-create-a-cube-in-swing

jabrena, 2018-11-16:
https://www.iro.umontreal.ca/~pift1025/bigjava/Ch05/ch05.html

jabrena, 2018-11-25:
https://gist.github.com/aerospark/80c60e801398fd961e3f

jabrena, 2019-08-18:
Rejected by LEAN

closed 2019-08-18 by jabrena