ev3dev-lang-java/ev3dev-lang-java #543: Release process v0.9.0

jabrena, 2018-11-04:
Once the development process was completed, it is necessary to review the Schedule planned

# Artifacts

**Debian Stretch:**

- [x] ev3dev-lang-java v2.4.0
- [x] installer (develop branch) v2.2.0
- [x] JVM: OpenJDK 11 JRI
- [x] JVM: OpenJDK 12 JRI EA

**Debian Jessie:**

# Documentation 

- [x] API upgrade 
- [x] Website upgrade https://github.com/ev3dev-lang-java/ev3dev-lang-java/issues/481
- [x] Docs  https://github.com/ev3dev-lang-java/ev3dev-lang-java/issues/408 https://github.com/ev3dev-lang-java/ev3dev-lang-java/issues/407
- [x] Readme ev3dev-lang-java (Master & Develop)

JakubVanek, 2018-11-08:
Hmmm, I now have the brick returned, but I have lost my sd card, so I will have to buy one tomorrow.

jabrena, 2018-11-08:
Tomorrow :p

jabrena, 2018-11-10:
Installer Released

jabrena, 2018-11-11:
Problem with the release:
https://github.com/jitpack/jitpack.io/issues/3552

jabrena, 2018-11-11:
Upgrading Website

jabrena, 2018-11-11:
Website upgrade

jabrena, 2018-11-11:
Pending docs for tomorrow

jabrena, 2018-11-12:
Problem with the release again:
https://github.com/jitpack/jitpack.io/issues/3555

Pending from a solution from Jitpack :(

jabrena, 2018-11-12:
Detected problem:
https://github.com/jitpack/jitpack.io/issues/3559

Jitpack need to downgrade to OpenJDK 10 to run Gradle
https://github.com/jitpack/jitpack.io/issues/3559

jabrena, 2018-11-13:
Published correctly.
https://jitpack.io/com/github/ev3dev-lang-java/ev3dev-lang-java/2.4.16/build.log

In the future, if some build fail, it is very pragmatic to write support people of Jitpack. The support is fantastic.

jabrena, 2018-11-13:
Updating docs

jabrena, 2018-11-14:
Question about choco:
https://github.com/chocolatey/choco/issues/1677

jabrena, 2018-11-15:
Documentation updated

closed 2018-11-15 by jabrena