ev3dev-lang-java/ev3dev-lang-java #547: Build Opencv and Java Bindings for EV3Dev Stretch

jabrena, 2018-11-04:


JakubVanek, 2018-11-05:
OpenCV should work out of the box when installed from Debian repositories. I haven't tested it, but I added the debian library path patch.
What are the Java Bindings?

jabrena, 2018-11-05:
In Debian Stretch with an EV3 Overclocked, the installation is fast and smooth:

```
robot@ev3dev:~$ sudo apt-get install --no-install-recommends --no-upgrade libopencv2.4-java
Reading package lists... Done
Building dependency tree       
Reading state information... Done
The following additional packages will be installed:
  libavcodec57 libavformat57 libavutil55 libbluray1 libchromaprint1
  libdc1394-22 libgme0 libgsm1 libgtk2.0-0 libgtk2.0-common libilmbase12
  libmp3lame0 libmpg123-0 libopencv-calib3d2.4v5 libopencv-contrib2.4v5
  libopencv-core2.4v5 libopencv-features2d2.4v5 libopencv-flann2.4v5
  libopencv-highgui2.4-deb0 libopencv-imgproc2.4v5 libopencv-ml2.4v5
  libopencv-objdetect2.4v5 libopencv-photo2.4v5 libopencv-video2.4v5
  libopencv2.4-jni libopenexr22 libopenjp2-7 libopenmpt0 libraw1394-11
  libshine3 libsnappy1v5 libsoxr0 libspeex1 libssh-gcrypt-4 libswresample2
  libswscale4 libtheora0 libtwolame0 libv4l-0 libv4lconvert0 libva-drm1
  libva-x11-1 libva1 libvdpau1 libvorbisfile3 libvpx4 libwavpack1 libx264-148
  libx265-95 libxvidcore4 libzvbi-common libzvbi0
Suggested packages:
  libbluray-bdj gvfs libraw1394-doc speex
Recommended packages:
  libaacs0 libgail-common libgtk2.0-bin va-driver-all | va-driver
  vdpau-driver-all | vdpau-driver
The following NEW packages will be installed:
  libavcodec57 libavformat57 libavutil55 libbluray1 libchromaprint1
  libdc1394-22 libgme0 libgsm1 libgtk2.0-0 libgtk2.0-common libilmbase12
  libmp3lame0 libmpg123-0 libopencv-calib3d2.4v5 libopencv-contrib2.4v5
  libopencv-core2.4v5 libopencv-features2d2.4v5 libopencv-flann2.4v5
  libopencv-highgui2.4-deb0 libopencv-imgproc2.4v5 libopencv-ml2.4v5
  libopencv-objdetect2.4v5 libopencv-photo2.4v5 libopencv-video2.4v5
  libopencv2.4-java libopencv2.4-jni libopenexr22 libopenjp2-7 libopenmpt0
  libraw1394-11 libshine3 libsnappy1v5 libsoxr0 libspeex1 libssh-gcrypt-4
  libswresample2 libswscale4 libtheora0 libtwolame0 libv4l-0 libv4lconvert0
  libva-drm1 libva-x11-1 libva1 libvdpau1 libvorbisfile3 libvpx4 libwavpack1
  libx264-148 libx265-95 libxvidcore4 libzvbi-common libzvbi0
0 upgraded, 53 newly installed, 0 to remove and 31 not upgraded.
Need to get 18.0 MB of archives.
After this operation, 67.4 MB of additional disk space will be used.
Do you want to continue? [Y/n] y
Get:1 http://security.debian.org stretch/updates/main armel libavutil55 armel 7:3.2.12-1~deb9u1 [239 kB]
Get:12 http://security.debian.org stretch/updates/main armel libswresample2 armel 7:3.2.12-1~deb9u1 [95.9 kB]
Get:14 http://security.debian.org stretch/updates/main armel libavcodec57 armel 7:3.2.12-1~deb9u1 [4176 kB]
Get:2 http://cdn-fastly.deb.debian.org/debian stretch/main armel libva1 armel 1.7.3-2 [46.7 kB]
Get:3 http://cdn-fastly.deb.debian.org/debian stretch/main armel libva-drm1 armel 1.7.3-2 [14.4 kB]
Get:4 http://cdn-fastly.deb.debian.org/debian stretch/main armel libva-x11-1 armel 1.7.3-2 [18.0 kB]
Get:5 http://cdn-fastly.deb.debian.org/debian stretch/main armel libvdpau1 armel 1.1.1-6 [38.9 kB]
Get:6 http://cdn-fastly.deb.debian.org/debian stretch/main armel libgsm1 armel 1.0.13-4+b2 [23.0 kB]
Get:7 http://cdn-fastly.deb.debian.org/debian stretch/main armel libmp3lame0 armel 3.99.5+repack1-9+b2 [342 kB]
Get:8 http://cdn-fastly.deb.debian.org/debian stretch/main armel libopenjp2-7 armel 2.1.2-1.1+deb9u2 [110 kB]
Get:9 http://cdn-fastly.deb.debian.org/debian stretch/main armel libshine3 armel 3.1.0-5 [25.3 kB]
Get:10 http://cdn-fastly.deb.debian.org/debian stretch/main armel libsnappy1v5 armel 1.1.3-3 [50.2 kB]
Get:11 http://cdn-fastly.deb.debian.org/debian stretch/main armel libspeex1 armel 1.2~rc1.2-1+b2 [54.8 kB]
Get:13 http://cdn-fastly.deb.debian.org/debian stretch/main armel libsoxr0 armel 0.1.2-2 [61.8 kB]
Get:15 http://cdn-fastly.deb.debian.org/debian stretch/main armel libtheora0 armel 1.1.1+dfsg.1-14+b1 [158 kB]
Get:16 http://cdn-fastly.deb.debian.org/debian stretch/main armel libtwolame0 armel 0.3.13-2 [48.8 kB]
Get:17 http://cdn-fastly.deb.debian.org/debian stretch/main armel libvpx4 armel 1.6.1-3+deb9u1 [436 kB]
Get:18 http://cdn-fastly.deb.debian.org/debian stretch/main armel libwavpack1 armel 5.0.0-2+deb9u2 [73.6 kB]
Get:19 http://cdn-fastly.deb.debian.org/debian stretch/main armel libx264-148 armel 2:0.148.2748+git97eaef2-1 [811 kB]
Get:20 http://cdn-fastly.deb.debian.org/debian stretch/main armel libx265-95 armel 2.1-2+b2 [452 kB]
Get:21 http://cdn-fastly.deb.debian.org/debian stretch/main armel libxvidcore4 armel 2:1.3.4-1+b2 [201 kB]
Get:51 http://security.debian.org stretch/updates/main armel libssh-gcrypt-4 armel 0.7.3-2+deb9u1 [144 kB]
Get:52 http://security.debian.org stretch/updates/main armel libavformat57 armel 7:3.2.12-1~deb9u1 [873 kB]
Get:22 http://cdn-fastly.deb.debian.org/debian stretch/main armel libzvbi-common all 0.2.35-13 [63.7 kB]
Get:23 http://cdn-fastly.deb.debian.org/debian stretch/main armel libzvbi0 armel 0.2.35-13 [240 kB]
Get:24 http://cdn-fastly.deb.debian.org/debian stretch/main armel libbluray1 armel 1:0.9.3-3 [122 kB]
Get:25 http://cdn-fastly.deb.debian.org/debian stretch/main armel libchromaprint1 armel 1.4.2-1 [35.2 kB]
Get:26 http://cdn-fastly.deb.debian.org/debian stretch/main armel libgme0 armel 0.6.0-4 [115 kB]
Get:27 http://cdn-fastly.deb.debian.org/debian stretch/main armel libmpg123-0 armel 1.23.8-1+b1 [121 kB]
Get:28 http://cdn-fastly.deb.debian.org/debian stretch/main armel libvorbisfile3 armel 1.3.5-4+deb9u2 [25.0 kB]
[... 218 lines cut ...]
[... cut ...]
Setting up libopencv-highgui2.4-deb0:armel (2.4.9.1+dfsg1-2) ...
Setting up libopencv-objdetect2.4v5:armel (2.4.9.1+dfsg1-2) ...
Setting up libopencv-features2d2.4v5:armel (2.4.9.1+dfsg1-2) ...
Setting up libopencv-calib3d2.4v5:armel (2.4.9.1+dfsg1-2) ...
Setting up libopencv-contrib2.4v5:armel (2.4.9.1+dfsg1-2) ...
Setting up libopencv2.4-jni (2.4.9.1+dfsg1-2) ...
Setting up libopencv2.4-java (2.4.9.1+dfsg1-2) ...
Processing triggers for libc-bin (2.24-11+deb9u3) ...
robot@ev3dev:~$ 
```

Pending run one example with OpenCV

jabrena, 2018-11-05:
The same problem that in previous issue:

```
robot@ev3dev:/usr/lib/jni$ ldd /usr/lib/jni/libopencv_java249.so
	libopencv_contrib.so.2.4 => /usr/lib/arm-linux-gnueabi/libopencv_contrib.so.2.4 (0xb6e16000)
	libopencv_photo.so.2.4 => /usr/lib/arm-linux-gnueabi/libopencv_photo.so.2.4 (0xb6ded000)
	libdl.so.2 => /lib/arm-linux-gnueabi/libdl.so.2 (0xb6dda000)
	libpthread.so.0 => /lib/arm-linux-gnueabi/libpthread.so.0 (0xb6db0000)
	librt.so.1 => /lib/arm-linux-gnueabi/librt.so.1 (0xb6d99000)
	libopencv_calib3d.so.2.4 => /usr/lib/arm-linux-gnueabi/libopencv_calib3d.so.2.4 (0xb6d0c000)
	libopencv_features2d.so.2.4 => /usr/lib/arm-linux-gnueabi/libopencv_features2d.so.2.4 (0xb6c79000)
	libopencv_flann.so.2.4 => /usr/lib/arm-linux-gnueabi/libopencv_flann.so.2.4 (0xb6c11000)
	libopencv_ml.so.2.4 => /usr/lib/arm-linux-gnueabi/libopencv_ml.so.2.4 (0xb6b90000)
	libopencv_objdetect.so.2.4 => /usr/lib/arm-linux-gnueabi/libopencv_objdetect.so.2.4 (0xb6b23000)
	libopencv_highgui.so.2.4 => /usr/lib/arm-linux-gnueabi/libopencv_highgui.so.2.4 (0xb6ada000)
	libopencv_video.so.2.4 => /usr/lib/arm-linux-gnueabi/libopencv_video.so.2.4 (0xb6a87000)
	libopencv_imgproc.so.2.4 => /usr/lib/arm-linux-gnueabi/libopencv_imgproc.so.2.4 (0xb684a000)
	libopencv_core.so.2.4 => /usr/lib/arm-linux-gnueabi/libopencv_core.so.2.4 (0xb666d000)
	libstdc++.so.6 => /usr/lib/arm-linux-gnueabi/libstdc++.so.6 (0xb6521000)
	libm.so.6 => /lib/arm-linux-gnueabi/libm.so.6 (0xb6467000)
	libgcc_s.so.1 => /lib/arm-linux-gnueabi/libgcc_s.so.1 (0xb6437000)
	libc.so.6 => /lib/arm-linux-gnueabi/libc.so.6 (0xb62ed000)
	/lib/ld-linux.so.3 (0xb6f6e000)
	libz.so.1 => /lib/arm-linux-gnueabi/libz.so.1 (0xb62c4000)
	libjpeg.so.62 => /usr/lib/arm-linux-gnueabi/libjpeg.so.62 (0xb627a000)
	libpng16.so.16 => /usr/lib/arm-linux-gnueabi/libpng16.so.16 (0xb623e000)
	libtiff.so.5 => /usr/lib/arm-linux-gnueabi/libtiff.so.5 (0xb61c1000)
	libImath-2_2.so.12 => /usr/lib/arm-linux-gnueabi/libImath-2_2.so.12 (0xb619c000)
	libIlmImf-2_2.so.22 => /usr/lib/arm-linux-gnueabi/libIlmImf-2_2.so.22 (0xb5ee8000)
	libIex-2_2.so.12 => /usr/lib/arm-linux-gnueabi/libIex-2_2.so.12 (0xb5ec1000)
	libHalf.so.12 => /usr/lib/arm-linux-gnueabi/libHalf.so.12 (0xb5e6e000)
	libIlmThread-2_2.so.12 => /usr/lib/arm-linux-gnueabi/libIlmThread-2_2.so.12 (0xb5e57000)
	libgtk-x11-2.0.so.0 => /usr/lib/arm-linux-gnueabi/libgtk-x11-2.0.so.0 (0xb5a7f000)
	libgdk-x11-2.0.so.0 => /usr/lib/arm-linux-gnueabi/libgdk-x11-2.0.so.0 (0xb59d5000)
	libpangocairo-1.0.so.0 => /usr/lib/arm-linux-gnueabi/libpangocairo-1.0.so.0 (0xb59b8000)
	libatk-1.0.so.0 => /usr/lib/arm-linux-gnueabi/libatk-1.0.so.0 (0xb598a000)
	libcairo.so.2 => /usr/lib/arm-linux-gnueabi/libcairo.so.2 (0xb5881000)
	libgdk_pixbuf-2.0.so.0 => /usr/lib/arm-linux-gnueabi/libgdk_pixbuf-2.0.so.0 (0xb584d000)
	libgio-2.0.so.0 => /usr/lib/arm-linux-gnueabi/libgio-2.0.so.0 (0xb56e1000)
	libpangoft2-1.0.so.0 => /usr/lib/arm-linux-gnueabi/libpangoft2-1.0.so.0 (0xb56be000)
	libpango-1.0.so.0 => /usr/lib/arm-linux-gnueabi/libpango-1.0.so.0 (0xb566b000)
	libgobject-2.0.so.0 => /usr/lib/arm-linux-gnueabi/libgobject-2.0.so.0 (0xb560e000)
	libglib-2.0.so.0 => /lib/arm-linux-gnueabi/libglib-2.0.so.0 (0xb5501000)
	libfontconfig.so.1 => /usr/lib/arm-linux-gnueabi/libfontconfig.so.1 (0xb54bb000)
	libfreetype.so.6 => /usr/lib/arm-linux-gnueabi/libfreetype.so.6 (0xb541c000)
	libgthread-2.0.so.0 => /usr/lib/arm-linux-gnueabi/libgthread-2.0.so.0 (0xb540a000)
	libdc1394.so.22 => /usr/lib/arm-linux-gnueabi/libdc1394.so.22 (0xb5384000)
	libv4l1.so.0 => /usr/lib/arm-linux-gnueabi/libv4l1.so.0 (0xb536e000)
	libavcodec.so.57 => /usr/lib/arm-linux-gnueabi/libavcodec.so.57 (0xb40a2000)
	libavformat.so.57 => /usr/lib/arm-linux-gnueabi/libavformat.so.57 (0xb3e7f000)
	libavutil.so.55 => /usr/lib/arm-linux-gnueabi/libavutil.so.55 (0xb3dde000)
	libswscale.so.4 => /usr/lib/arm-linux-gnueabi/libswscale.so.4 (0xb3d57000)
	liblzma.so.5 => /lib/arm-linux-gnueabi/liblzma.so.5 (0xb3d25000)
	libjbig.so.0 => /usr/lib/arm-linux-gnueabi/libjbig.so.0 (0xb3d08000)
	libgmodule-2.0.so.0 => /usr/lib/arm-linux-gnueabi/libgmodule-2.0.so.0 (0xb3cf2000)
	libX11.so.6 => /usr/lib/arm-linux-gnueabi/libX11.so.6 (0xb3bc8000)
	libXcomposite.so.1 => /usr/lib/arm-linux-gnueabi/libXcomposite.so.1 (0xb3bb5000)
	libXdamage.so.1 => /usr/lib/arm-linux-gnueabi/libXdamage.so.1 (0xb3ba2000)
	libXfixes.so.3 => /usr/lib/arm-linux-gnueabi/libXfixes.so.3 (0xb3b8d000)
	libXrender.so.1 => /usr/lib/arm-linux-gnueabi/libXrender.so.1 (0xb3b73000)
	libXinerama.so.1 => /usr/lib/arm-linux-gnueabi/libXinerama.so.1 (0xb3b60000)
	libXi.so.6 => /usr/lib/arm-linux-gnueabi/libXi.so.6 (0xb3b41000)
	libXrandr.so.2 => /usr/lib/arm-linux-gnueabi/libXrandr.so.2 (0xb3b28000)
	libXcursor.so.1 => /usr/lib/arm-linux-gnueabi/libXcursor.so.1 (0xb3b0f000)
	libXext.so.6 => /usr/lib/arm-linux-gnueabi/libXext.so.6 (0xb3aee000)
	libpixman-1.so.0 => /usr/lib/arm-linux-gnueabi/libpixman-1.so.0 (0xb3a36000)
	libxcb-shm.so.0 => /usr/lib/arm-linux-gnueabi/libxcb-shm.so.0 (0xb3a23000)
	libxcb.so.1 => /usr/lib/arm-linux-gnueabi/libxcb.so.1 (0xb39f3000)
	libxcb-render.so.0 => /usr/lib/arm-linux-gnueabi/libxcb-render.so.0 (0xb39d8000)
	libselinux.so.1 => /lib/arm-linux-gnueabi/libselinux.so.1 (0xb39a3000)
[... 53 lines cut ...]
	libmpg123.so.0 => /usr/lib/arm-linux-gnueabi/libmpg123.so.0 (0xb24dc000)
	libvorbisfile.so.3 => /usr/lib/arm-linux-gnueabi/libvorbisfile.so.3 (0xb24c3000)
	libxml2.so.2 => /usr/lib/arm-linux-gnueabi/libxml2.so.2 (0xb2329000)
[... cut ...]
	libcom_err.so.2 => /lib/arm-linux-gnueabi/libcom_err.so.2 (0xb1fa8000)
	libkrb5support.so.0 => /usr/lib/arm-linux-gnueabi/libkrb5support.so.0 (0xb1f8f000)
	libkeyutils.so.1 => /lib/arm-linux-gnueabi/libkeyutils.so.1 (0xb1f7c000)
	libicui18n.so.57 => /usr/lib/arm-linux-gnueabi/libicui18n.so.57 (0xb1d61000)
	libicuuc.so.57 => /usr/lib/arm-linux-gnueabi/libicuuc.so.57 (0xb1bf2000)
	libicudata.so.57 => /usr/lib/arm-linux-gnueabi/libicudata.so.57 (0xb0365000)
```

https://github.com/ev3dev-lang-java/ev3dev-lang-java/issues/502

JakubVanek, 2018-11-05:
Try removing the java.library.path argument.

JakubVanek, 2018-11-05:
Try removing the `java.library.path` argument. Also, try running it with `-verbose:jni` argument.

JakubVanek, 2018-11-05:
From the last issue, `-verbose:class` also seemed to be helpful.

JakubVanek, 2018-11-05:
Hmmm, if we do not find the cause of the issue, I will have to use GDB to find the cause. It is also possible that new JNI code contains something that is not compatible with EV3. That would require a new patch for the new Javas.

jabrena, 2018-11-05:
I will try with old Java 8 but if you remember the previous issues was with a dependency not with the JVM. I will try with Java 8.

```
time java -verbose:class -verbose:jni -Djava.library.path=/usr/lib/jni/ -cp /home/robot/ev3dev-lang-java-2.4.7-SNAPSHOT-all.jar:/usr/share/OpenCV/java/opencv-249.jar examples.opencv.HelloWorldCV

ev3dev#8|[0.420s][info][class,load] opened: /opt/jri-12/lib/modules
ev3dev#8|[0.691s][info][class,load] java.lang.Object source: jrt:/java.base
ev3dev#8|[0.747s][info][class,load] java.io.Serializable source: jrt:/java.base
ev3dev#8|[0.754s][info][class,load] java.lang.Comparable source: jrt:/java.base
ev3dev#8|[0.784s][info][class,load] java.lang.CharSequence source: jrt:/java.base
ev3dev#8|[0.803s][info][class,load] java.lang.String source: jrt:/java.base
ev3dev#8|[0.872s][info][class,load] java.lang.reflect.AnnotatedElement source: jrt:/java.base
ev3dev#8|[0.887s][info][class,load] java.lang.reflect.GenericDeclaration source: jrt:/java.base
ev3dev#8|[0.894s][info][class,load] java.lang.reflect.Type source: jrt:/java.base
ev3dev#8|[0.929s][info][class,load] java.lang.Class source: jrt:/java.base
ev3dev#8|[0.937s][info][class,load] java.lang.Cloneable source: jrt:/java.base
ev3dev#8|[0.990s][info][class,load] java.lang.ClassLoader source: jrt:/java.base
ev3dev#8|[1.017s][info][class,load] java.lang.System source: jrt:/java.base
ev3dev#8|[1.039s][info][class,load] java.lang.Throwable source: jrt:/java.base
ev3dev#8|[1.046s][info][class,load] java.lang.Error source: jrt:/java.base
ev3dev#8|[1.050s][info][class,load] java.lang.ThreadDeath source: jrt:/java.base
ev3dev#8|[1.061s][info][class,load] java.lang.Exception source: jrt:/java.base
ev3dev#8|[1.068s][info][class,load] java.lang.RuntimeException source: jrt:/java.base
ev3dev#8|[1.084s][info][class,load] java.lang.SecurityManager source: jrt:/java.base
ev3dev#8|[1.116s][info][class,load] java.security.ProtectionDomain source: jrt:/java.base
ev3dev#8|[1.132s][info][class,load] java.security.AccessControlContext source: jrt:/java.base
ev3dev#8|[1.151s][info][class,load] java.security.SecureClassLoader source: jrt:/java.base
ev3dev#8|[1.171s][info][class,load] java.lang.ReflectiveOperationException source: jrt:/java.base
ev3dev#8|[1.186s][info][class,load] java.lang.ClassNotFoundException source: jrt:/java.base
ev3dev#8|[1.195s][info][class,load] java.lang.LinkageError source: jrt:/java.base
ev3dev#8|[1.198s][info][class,load] java.lang.NoClassDefFoundError source: jrt:/java.base
ev3dev#8|[1.208s][info][class,load] java.lang.ClassCastException source: jrt:/java.base
ev3dev#8|[1.213s][info][class,load] java.lang.ArrayStoreException source: jrt:/java.base
ev3dev#8|[1.225s][info][class,load] java.lang.VirtualMachineError source: jrt:/java.base
ev3dev#8|[1.239s][info][class,load] java.lang.OutOfMemoryError source: jrt:/java.base
ev3dev#8|[1.248s][info][class,load] java.lang.StackOverflowError source: jrt:/java.base
ev3dev#8|[1.260s][info][class,load] java.lang.IllegalMonitorStateException source: jrt:/java.base
ev3dev#8|[1.270s][info][class,load] java.lang.ref.Reference source: jrt:/java.base
ev3dev#8|[1.290s][info][class,load] java.lang.ref.SoftReference source: jrt:/java.base
ev3dev#8|[1.297s][info][class,load] java.lang.ref.WeakReference source: jrt:/java.base
ev3dev#8|[1.310s][info][class,load] java.lang.ref.FinalReference source: jrt:/java.base
ev3dev#8|[1.315s][info][class,load] java.lang.ref.PhantomReference source: jrt:/java.base
ev3dev#8|[1.332s][info][class,load] java.lang.ref.Finalizer source: jrt:/java.base
ev3dev#8|[1.357s][info][class,load] java.lang.Runnable source: jrt:/java.base
ev3dev#8|[1.380s][info][class,load] java.lang.Thread source: jrt:/java.base
ev3dev#8|[1.397s][info][class,load] java.lang.Thread$UncaughtExceptionHandler source: jrt:/java.base
ev3dev#8|[1.410s][info][class,load] java.lang.ThreadGroup source: jrt:/java.base
ev3dev#8|[1.463s][info][class,load] java.util.Map source: jrt:/java.base
ev3dev#8|[1.484s][info][class,load] java.util.Dictionary source: jrt:/java.base
ev3dev#8|[1.493s][info][class,load] java.util.Hashtable source: jrt:/java.base
ev3dev#8|[1.507s][info][class,load] java.util.Properties source: jrt:/java.base
ev3dev#8|[1.537s][info][class,load] java.lang.Module source: jrt:/java.base
ev3dev#8|[1.555s][info][class,load] java.lang.reflect.AccessibleObject source: jrt:/java.base
ev3dev#8|[1.580s][info][class,load] java.lang.reflect.Member source: jrt:/java.base
ev3dev#8|[1.601s][info][class,load] java.lang.reflect.Field source: jrt:/java.base
ev3dev#8|[1.617s][info][class,load] java.lang.reflect.Parameter source: jrt:/java.base
ev3dev#8|[1.653s][info][class,load] java.lang.reflect.Executable source: jrt:/java.base
ev3dev#8|[1.668s][info][class,load] java.lang.reflect.Method source: jrt:/java.base
ev3dev#8|[1.691s][info][class,load] java.lang.reflect.Constructor source: jrt:/java.base
ev3dev#8|[1.701s][info][class,load] jdk.internal.reflect.MagicAccessorImpl source: jrt:/java.base
ev3dev#8|[1.712s][info][class,load] jdk.internal.reflect.MethodAccessor source: jrt:/java.base
ev3dev#8|[1.727s][info][class,load] jdk.internal.reflect.MethodAccessorImpl source: jrt:/java.base
ev3dev#8|[1.741s][info][class,load] jdk.internal.reflect.ConstructorAccessor source: jrt:/java.base
ev3dev#8|[1.752s][info][class,load] jdk.internal.reflect.ConstructorAccessorImpl source: jrt:/java.base
ev3dev#8|[1.765s][info][class,load] jdk.internal.reflect.DelegatingClassLoader source: jrt:/java.base
[... cut ...]
ev3dev#8|2018-11-05 19:54:55 [main] INFO  examples.opencv.HelloWorldCV - opencv_java249
ev3dev#8|[62.148s][info][class,load] java.lang.ClassLoaderHelper source: jrt:/java.base
ev3dev#8|bash: line 1:  1084 Segmentation fault      java -verbose:class -verbose:jni -Djava.library.path=/usr/lib/jni/ -cp /home/robot/ev3dev-lang-java-2.4.7-SNAPSHOT-all.jar:/usr/share/OpenCV/java/opencv-249.jar examples.opencv.HelloWorldCV
ev3dev#8|
ev3dev#8|real	1m12.283s
ev3dev#8|user	0m27.132s
ev3dev#8|sys	0m3.723s
```

JakubVanek, 2018-11-05:
The situation now is a bit different - now we do not get a nice UnsatisfiedLinkError, we get just a segmentation fault. This is why I believe it might not necessarily have something to do with dependencies.

Also, could you please try it without `java.library.path` argument?

JakubVanek, 2018-11-05:
Hmmm, we could also try OpenJDK 9 JRI (if that is enough for OpenCV) as this Java version is reasonably old and also contains the Debian library path patch.

jabrena, 2018-11-05:
buff, I think that I will continue with this issue tomorrow, It requires more days...

Can you explain what is the Debian library path patch?

JakubVanek, 2018-11-05:
Hmm, I have a proposal: we can share long logs via Pastebin or GitHub Gists.

jabrena, 2018-11-05:
Oki, I will next time.