ev3dev-lang-java/ev3dev-lang-java #553: ERROR ev3dev.actuators.Sound - mark/reset not supported

robl0377, 2018-11-07:
Have installed ev3dev on debian stretch on my EV3 Brick and tried the sample program 
example.robotics.ev3.actuator.PlaySoundFilesExample from yesterday's template-project-gradle:

package example.robotics.ev3.actuator;
import ev3dev.actuators.Sound;
import ev3dev.utils.JarResource;
import javax.sound.sampled.LineUnavailableException;
import javax.sound.sampled.UnsupportedAudioFileException;
import java.io.File;
import java.io.IOException;
public class PlaySoundFilesExample {
    public static void main(String[] args) throws IOException, UnsupportedAudioFileException, LineUnavailableException {
        //Sound courtesy: http://www.wavsource.com/video_games/duke_nukem.htm
        String filePath = "ready_4_action.wav";
        String result = JarResource.export(filePath);
        Sound sound = Sound.getInstance();
        File file = new File(result);
        sound.playSample(file);
        JarResource.delete(result);
    }
}`

On the ssh console I get this:
> Task :remoteRun
java -server -jar /home/robot/EV3dev_Examples-all-0.3.0.jar 
ev3dev#18|2018-11-07 10:51:04 [main] INFO  ev3dev.actuators.Sound - Providing a Sound instance
ev3dev#18|2018-11-07 10:51:05 [main] DEBUG ev3dev.hardware.EV3DevDistros - Debian Stretch detected
ev3dev#18|2018-11-07 10:51:05 [main] DEBUG ev3dev.hardware.EV3DevDistros - Debian Stretch detected
ev3dev#18|2018-11-07 10:51:05 [main] DEBUG ev3dev.hardware.EV3DevFileSystem - Root Path: /sys/class
ev3dev#18|2018-11-07 10:51:05 [main] INFO  ev3dev.actuators.Sound - Creating a instance of Sound
ev3dev#18|2018-11-07 10:51:06 [main] **ERROR ev3dev.actuators.Sound - mark/reset not supported**
ev3dev#18|java.io.IOException: mark/reset not supported
ev3dev#18|	at java.base/java.io.InputStream.reset(Unknown Source)
ev3dev#18|	at java.desktop/com.sun.media.sound.SunFileReader.getAudioFileFormat(Unknown Source)
ev3dev#18|	at java.desktop/com.sun.media.sound.WaveExtensibleFileReader.getAudioInputStream(Unknown Source)
ev3dev#18|	at java.desktop/javax.sound.sampled.AudioSystem.getAudioInputStream(Unknown Source)
ev3dev#18|	at ev3dev.actuators.Sound.playSample(Sound.java:149)
ev3dev#18|	at ev3dev.actuators.Sound.playSample(Sound.java:169)
ev3dev#18|	at example.robotics.ev3.actuator.PlaySoundFilesExample.main(PlaySoundFilesExample.java:22)
ev3dev#18|Exception in thread "main" java.lang.RuntimeException: java.io.IOException: mark/reset not supported
ev3dev#18|	at ev3dev.actuators.Sound.playSample(Sound.java:159)
ev3dev#18|	at ev3dev.actuators.Sound.playSample(Sound.java:169)
ev3dev#18|	at example.robotics.ev3.actuator.PlaySoundFilesExample.main(PlaySoundFilesExample.java:22)
ev3dev#18|Caused by: java.io.IOException: mark/reset not supported
ev3dev#18|	at java.base/java.io.InputStream.reset(Unknown Source)
ev3dev#18|	at java.desktop/com.sun.media.sound.SunFileReader.getAudioFileFormat(Unknown Source)
ev3dev#18|	at java.desktop/com.sun.media.sound.WaveExtensibleFileReader.getAudioInputStream(Unknown Source)
ev3dev#18|	at java.desktop/javax.sound.sampled.AudioSystem.getAudioInputStream(Unknown Source)
ev3dev#18|	at ev3dev.actuators.Sound.playSample(Sound.java:149)
ev3dev#18|	... 2 more

Java Runtime: 
openjdk version "10" 2018-03-20
OpenJDK Runtime Environment (build 10+46-ev3)
OpenJDK Client VM (build 10+46-ev3, mixed mode, sharing)

Main libraries:
ev3dev-lang-java-2.4.7-SNAPSHOT.jar
lejos-commons-0.7.1.jar

JakubVanek, 2018-11-07:
This might be a bug in our library: https://stackoverflow.com/questions/5529754/java-io-ioexception-mark-reset-not-supported

jabrena, 2018-11-08:
Hi @robl0377, 

this problem was solved this week.
We are releasing a new version if you like, we could release a new snapshot tomorrow.

Juan Antonio

JakubVanek, 2018-11-08:
Oh, I see, this is the problem with the InputStream version of playSample... :smile:

jabrena, 2018-11-08:
@JakubVanek You like the innovation... XD

@robl0377 I have just released a Snapshot for you in exclusive:
https://jitpack.io/#ev3dev-lang-java/ev3dev-lang-java/2.4.8-SNAPSHOT

This weekend, we will release a minor version of the library, the development is freeze. Your issue about the sound should be fixed with this snapshot. In relation to the motor issue, it is a bad lucky. In NXT we worked with a microcontroller but in EV3, we interact with a Linux system and the process execution is asyncronous. In relation with the Button doubt, create the issue and we will reply. In Debian Stretch, appear new ways to interact with the display and buttons, but it requires an explanation. :)

See you tomorrow.

Juan Antonio

robl0377, 2018-11-09:
@jabrena Playing the .wav sound file in the SoundDemo works now with the 2.4.8-SNAPSHOT 👍

closed 2018-11-09 by jabrena