ev3dev-lang-java/ev3dev-lang-java #554: ERROR ev3dev.utils.Sysfs - File: '/sys/class/tacho-motor/motor9/command' without write permissions.

robl0377, 2018-11-07:
Have installed ev3dev on debian stretch on my EV3 Brick and tried the sample program
example.robotics.ev3.actuator.motor.BasicMotorOperationExample from yesterday's template-project-gradle:

package example.robotics.ev3.actuator.motor;
import org.slf4j.Logger;
import org.slf4j.LoggerFactory;
import ev3dev.actuators.lego.motors.EV3LargeRegulatedMotor;
import lejos.hardware.port.MotorPort;
import lejos.utility.Delay;
class BasicMotorOperationExample {
    public static Logger LOGGER = LoggerFactory.getLogger(BasicMotorOperationExample.class);
    public static int DELAY = 2000; //2 Seconds
    public static final EV3LargeRegulatedMotor mA = new EV3LargeRegulatedMotor(MotorPort.B);
    public static void main(String[] args) {
        mA.brake();
        LOGGER.info("Starting motor on B");
        mA.setSpeed(500);
        LOGGER.info("Forward");
        mA.forward();
        LOGGER.info("Large Motor is moving: {}, at speed {}", mA.isMoving(), mA.getSpeed());
        Delay.msDelay(DELAY);
        LOGGER.info("Stop");
        mA.stop();
        LOGGER.info("Large Motor is moving: {}, at speed {}", mA.isMoving(), mA.getSpeed());
        Delay.msDelay(DELAY);
        LOGGER.info("Backward");
        mA.backward();
        LOGGER.info("Large Motor is moving: {}, at speed {}", mA.isMoving(), mA.getSpeed());
        Delay.msDelay(2000);
        LOGGER.info("Stop");
        mA.stop();
        LOGGER.info("Large Motor is moving: {}, at speed {}", mA.isMoving(), mA.getSpeed());
        Delay.msDelay(DELAY);
        LOGGER.info("Forward");
        mA.forward();
        Delay.msDelay(2000);
        mA.stop();
        LOGGER.info("Stop");
    }
}

On the ssh console I get this:
> Task :remoteRun
java -server -jar /home/robot/EV3dev_Examples-all-0.3.0.jar 
ev3dev#27|2018-11-07 11:51:05 [main] DEBUG ev3dev.hardware.EV3DevDistros - Debian Stretch detected
ev3dev#27|2018-11-07 11:51:06 [main] DEBUG ev3dev.hardware.EV3DevDistros - Debian Stretch detected
ev3dev#27|2018-11-07 11:51:06 [main] DEBUG ev3dev.hardware.EV3DevFileSystem - Root Path: /sys/class
ev3dev#27|2018-11-07 11:51:07 [main] INFO  e.a.lego.motors.BaseRegulatedMotor - Configuring motor connected on Port: B
ev3dev#27|2018-11-07 11:51:07 [main] DEBUG e.a.lego.motors.BaseRegulatedMotor - Detecting motor on port: ev3-ports:outB
ev3dev#27|2018-11-07 11:51:07 [main] DEBUG ev3dev.hardware.EV3DevDevice - Detected device on path: /sys/class/lego-port/port5/address, ev3-ports:outB
ev3dev#27|2018-11-07 11:51:07 [main] DEBUG e.a.lego.motors.BaseRegulatedMotor - Setting port in mode: tacho-motor
ev3dev#27|2018-11-07 11:51:08 [main] DEBUG ev3dev.hardware.EV3DevDevice - Detected device on path: /sys/class/tacho-motor/motor9/address, ev3-ports:outB
ev3dev#27|2018-11-07 11:51:09 [main] ERROR ev3dev.utils.Sysfs - File: '/sys/class/tacho-motor/motor9/command' without write permissions.
ev3dev#27|Exception in thread "main" java.lang.ExceptionInInitializerError
ev3dev#27|Caused by: java.lang.RuntimeException: Operation not executed: /sys/class/tacho-motor/motor9/command with value reset
ev3dev#27|	at ev3dev.hardware.EV3DevDevice.setStringAttribute(EV3DevDevice.java:118)
ev3dev#27|	at ev3dev.actuators.lego.motors.BaseRegulatedMotor.<init>(BaseRegulatedMotor.java:94)
ev3dev#27|	at ev3dev.actuators.lego.motors.EV3LargeRegulatedMotor.<init>(EV3LargeRegulatedMotor.java:26)
ev3dev#27|	at example.robotics.ev3.actuator.motor.BasicMotorOperationExample.<clinit>(BasicMotorOperationExample.java:16)

Java Runtime:
openjdk version "10" 2018-03-20
OpenJDK Runtime Environment (build 10+46-ev3)
OpenJDK Client VM (build 10+46-ev3, mixed mode, sharing)

Main libraries:
ev3dev-lang-java-2.4.7-SNAPSHOT.jar
lejos-commons-0.7.1.jar

**However, it works on Oracle Java 8 (
java version "1.8.0"
Java(TM) SE Embedded Runtime Environment (build 1.8.0-b132, headless)
Java HotSpot(TM) Embedded Client VM (build 25.0-b70, mixed mode)
)**

robl0377, 2018-11-07:
After a reboot of the EV3 Brick I tried it again and it worked with both JDK 8 and JDK 10 just fine!

robl0377, 2018-11-08:
Got a similar Error today when running the **ev3dev.actuators.lego.motors.UnregulatedMotorDemo** on the EV3 brick with OpenJDK 10:

```java
package ev3dev.actuators.lego.motors;
import lejos.hardware.port.MotorPort;
import lejos.utility.Delay;
public class UnregulatedMotorDemo {
	// Motor Definition
	private static UnregulatedMotor umotor1 = new UnregulatedMotor(MotorPort.C);
	// Configuration
	private final static int motorPower = 50;
	private final static int ONE_SECOND = 1000;

	public static void main(String[] args) {
		System.out.println("UnregulatedMotorDemo: Set Motor Power");
		// Set power for DC Motor
		umotor1.setPower(motorPower);
		// Testing DC-Motor 1
		umotor1.forward();
		System.out.println("forward: " + umotor1.isMoving());
		Delay.msDelay(ONE_SECOND);
		umotor1.stop();
		System.out.println("stop: " + umotor1.isMoving());
		umotor1.backward();
		System.out.println("backward: " + umotor1.isMoving());
		Delay.msDelay(ONE_SECOND);
		umotor1.stop();
		System.out.println("stop: " + umotor1.isMoving());
		umotor1.forward();
		System.out.println("forward: " + umotor1.isMoving());
		Delay.msDelay(ONE_SECOND);
		umotor1.stop();
		System.out.println("stop: " + umotor1.isMoving());
	}
}
```
On the ssh console I get this:
```
> Task :remoteRun
Host key checking is off. It may be vulnerable to man-in-the-middle attacks.
java -server -jar /home/robot/EV3dev-java-systemtests-2.4.7-SNAPSHOT-all.jar 

ev3dev#47|2018-11-08 09:49:09 [main] DEBUG ev3dev.hardware.EV3DevDistros - Debian Stretch detected
ev3dev#47|2018-11-08 09:49:10 [main] DEBUG ev3dev.hardware.EV3DevDistros - Debian Stretch detected
ev3dev#47|2018-11-08 09:49:10 [main] DEBUG ev3dev.hardware.EV3DevFileSystem - Root Path: /sys/class
ev3dev#47|2018-11-08 09:49:10 [main] TRACE ev3dev.utils.Sysfs - ls /sys/class/power_supply/lego-ev3-battery
ev3dev#47|2018-11-08 09:49:10 [main] TRACE ev3dev.hardware.EV3DevPlatformsImpl - Detected platform: EV3BRICK
ev3dev#47|2018-11-08 09:49:10 [main] TRACE ev3dev.hardware.EV3DevDevice - Retrieving devices in path: 
ev3dev#47|2018-11-08 09:49:10 [main] TRACE ev3dev.utils.Sysfs - ls /sys/class/lego-port
ev3dev#47|2018-11-08 09:49:10 [main] TRACE ev3dev.hardware.EV3DevDevice - Checking devices on: /sys/class/lego-port
ev3dev#47|2018-11-08 09:49:10 [main] TRACE ev3dev.hardware.EV3DevDevice - Device 1:
ev3dev#47|2018-11-08 09:49:10 [main] TRACE ev3dev.utils.Sysfs - cat /sys/class/lego-port/port6/address
ev3dev#47|2018-11-08 09:49:11 [main] TRACE ev3dev.utils.Sysfs - value: ev3-ports:outC
ev3dev#47|2018-11-08 09:49:11 [main] TRACE ev3dev.hardware.EV3DevDevice - Port expected: ev3-ports:outC, actual: ev3-ports:outC.
ev3dev#47|2018-11-08 09:49:11 [main] DEBUG ev3dev.hardware.EV3DevDevice - Detected device on path: /sys/class/lego-port/port6/address, ev3-ports:outC
ev3dev#47|2018-11-08 09:49:11 [main] TRACE ev3dev.utils.Sysfs - echo dc-motor > /sys/class/lego-port/port6/mode
ev3dev#47|2018-11-08 09:49:11 [main] TRACE ev3dev.hardware.EV3DevDevice - Retrieving devices in path: 
ev3dev#47|2018-11-08 09:49:11 [main] TRACE ev3dev.utils.Sysfs - ls /sys/class/dc-motor
ev3dev#47|2018-11-08 09:49:12 [main] TRACE ev3dev.hardware.EV3DevDevice - Checking devices on: /sys/class/dc-motor
ev3dev#47|2018-11-08 09:49:12 [main] TRACE ev3dev.hardware.EV3DevDevice - Device 1:
ev3dev#47|2018-11-08 09:49:12 [main] TRACE ev3dev.utils.Sysfs - cat /sys/class/dc-motor/motor2/address
ev3dev#47|2018-11-08 09:49:12 [main] TRACE ev3dev.utils.Sysfs - value: ev3-ports:outC
ev3dev#47|2018-11-08 09:49:12 [main] TRACE ev3dev.hardware.EV3DevDevice - Port expected: ev3-ports:outC, actual: ev3-ports:outC.
ev3dev#47|2018-11-08 09:49:12 [main] DEBUG ev3dev.hardware.EV3DevDevice - Detected device on path: /sys/class/dc-motor/motor2/address, ev3-ports:outC
ev3dev#47|UnregulatedMotorDemo: Set Motor Power
ev3dev#47|2018-11-08 09:49:12 [main] TRACE ev3dev.utils.Sysfs - echo 50 > /sys/class/dc-motor/motor2/duty_cycle_sp
ev3dev#47|2018-11-08 09:49:12 [main] ERROR ev3dev.utils.Sysfs - File: '/sys/class/dc-motor/motor2/duty_cycle_sp' without write permissions.
ev3dev#47|Exception in thread "main" java.lang.RuntimeException: Operation not executed: /sys/class/dc-motor/motor2/duty_cycle_sp with value 50
ev3dev#47|	at ev3dev.hardware.EV3DevDevice.setIntegerAttribute(EV3DevDevice.java:131)
ev3dev#47|	at ev3dev.actuators.lego.motors.BasicMotor.setPower(BasicMotor.java:56)
[... 20 lines cut ...]
2018-11-08 10:10:20 [main] TRACE ev3dev.utils.Sysfs - ls /sys/class/lego-port
2018-11-08 10:10:20 [main] TRACE ev3dev.hardware.EV3DevDevice - Checking devices on: /sys/class/lego-port
2018-11-08 10:10:20 [main] TRACE ev3dev.hardware.EV3DevDevice - Device 1:
2018-11-08 10:10:20 [main] TRACE ev3dev.utils.Sysfs - cat /sys/class/lego-port/port6/address
2018-11-08 10:10:20 [main] TRACE ev3dev.utils.Sysfs - value: ev3-ports:outC
2018-11-08 10:10:20 [main] TRACE ev3dev.hardware.EV3DevDevice - Port expected: ev3-ports:outC, actual: ev3-ports:outC.
2018-11-08 10:10:20 [main] DEBUG ev3dev.hardware.EV3DevDevice - Detected device on path: /sys/class/lego-port/port6/address, ev3-ports:outC
2018-11-08 10:10:20 [main] TRACE ev3dev.utils.Sysfs - echo dc-motor > /sys/class/lego-port/port6/mode
2018-11-08 10:10:21 [main] TRACE ev3dev.hardware.EV3DevDevice - Retrieving devices in path:
2018-11-08 10:10:21 [main] TRACE ev3dev.utils.Sysfs - ls /sys/class/dc-motor
2018-11-08 10:10:21 [main] TRACE ev3dev.hardware.EV3DevDevice - Checking devices on: /sys/class/dc-motor
[... cut ...]
2018-11-08 10:10:22 [main] ERROR ev3dev.utils.Sysfs - File: '/sys/class/dc-motor/motor0/duty_cycle_sp' without write permissions.
Exception in thread "main" java.lang.RuntimeException: Operation not executed: /sys/class/dc-motor/motor0/duty_cycle_sp with value 50
        at ev3dev.hardware.EV3DevDevice.setIntegerAttribute(EV3DevDevice.java:131)
        at ev3dev.actuators.lego.motors.BasicMotor.setPower(BasicMotor.java:56)
        at ev3dev.actuators.lego.motors.UnregulatedMotorDemo.main(UnregulatedMotorDemo.java:19)
robot@ev3dev:~$
```

JakubVanek, 2018-11-08:
Hi,

I did some research about this, but I'm not sure what might be causing this. Could you try running this in SSH? _If the `/sys/class/dc-motor/motor0` directory doesn't exist, please try another one for this and the next test._
```bash
if [ -w "/sys/class/dc-motor/motor0/duty_cycle_sp" ]
then echo "access writable";
else echo "access unwritable"; fi
ls -la /sys/class/dc-motor/motor0/duty_cycle_sp
```
That will check that file for writability hopefully by using `access()` system call equivalent (see below) and then print the permissions of the file. Then, could you please try running this:
```bash
echo 50 > /sys/class/dc-motor/motor0/duty_cycle_sp
```
If this fails, the problem is really with permissions.

I'm a bit puzzled that this happens only on JDK10 and not JDK8. This is a part of `Sysfs.writeString` function:
```java
final File file = new File(filePath);
if(file.canWrite()) { /* <- this fails */
    //TODO Review if it possible to improve
    PrintWriter out = new PrintWriter(file);
    out.println(value);
    out.flush();
    out.close();
//TODO Review
}else {
    log.error("File: '{}' without write permissions.", filePath); /* <- this is what we see in the log */
    return false;
}
```

I have looked into the publicly available sources of JDK8 and JDK10 and both do these things to check for writable access:
1. `java.io.File.canWrite()` is called.
2. `java.io.UnixFileSystem.checkAccess()` is called.
3. Native JNI function `Java_java_io_UnixFileSystem_checkAccess()` is called.
4. `access(path, W_OK)` system call is invoked on the file.

I looked into ev3dev's [duty_cycle_sp store for dc-motor][1] and I don't think that causes the write to fail (disclaimer: this is the first time I read the code which handles sysfs, I might be wrong). However, on my desktop Ubuntu, normal users cannot write into sysfs by default. However, I have read that ev3dev uses udev rules to combat this problem. I don't know what is the difference between JVM versions then though.

Could you please try running this while JDK8 and then JDK10 are running:
```bash
ps -u -p `pidof java`
```

This should display the user who is running the Java process.

Also, please send us your kernel version:
```
uname -a
```

@dlech what do you think about this problem? 

[1]: https://github.com/ev3dev/lego-linux-drivers/blob/ev3dev-stretch/motors/dc_motor_class.c#L358

Regards,

Jakub Vanek

dlech, 2018-11-08:
FWIW, the permission come from here: https://github.com/ev3dev/lego-linux-drivers/blob/ev3dev-stretch/motors/dc_motor_class.c#L572

I think in ev3dev-lang-python, they had to make sure that each attribute was opened with the same permissions. If a file is read-only, it has to be opened for reading only, if it is write only, it has to be opened for writing only, and if it is read/write, it has to be opened for both reading and writing.

JakubVanek, 2018-11-08:
Hi,

thanks for your response - yup, I looked there now and I see ev3dev-lang-python uses `stat()` and keeps the file descriptors open if the user needs to reaccess them:
```python
    def _attribute_file_open(self, name):
        path = os.path.join(self._path, name)
        mode = stat.S_IMODE(os.stat(path)[stat.ST_MODE])
        r_ok = mode & stat.S_IRGRP
        w_ok = mode & stat.S_IWGRP

        if r_ok and w_ok:
            mode_str = 'r+'
        elif w_ok:
            mode_str = 'w'
        else:
            mode_str = 'r'

        return io.FileIO(path, mode_str)

    def _set_attribute(self, attribute, name, value):
        """Device attribute setter"""
        try:
            if attribute is None:
                attribute = self._attribute_file_open( name )
            else:
                attribute.seek(0)

            if isinstance(value, str):
                value = value.encode()
            attribute.write(value)
            attribute.flush()
        except Exception as ex:
            self._raise_friendly_access_error(ex, name)
        return attribute

```
However I think the operation would fail here as well, if `access()` reports whether the file can be opened for writing even when only the group bit matches. The attribute fd here would be opened only for reading and then the write on file opened as read-only would probably fail.

Regards

robl0377, 2018-11-08:
> Hi @robl0377 <https://github.com/robl0377>,
>
> I have tested many times the motors and I didn't find that issue.
> In my case, I am testing now with latest build from OpenJDK12 and
> Debian Stretch. If you like, can you upgrade, I would like to help you
> and test in team.
>
> Juan Antonio

Hi Juan Antonio,

I have installed OpenJDK 10 because the installer.sh installed it as the
latest version. I try to follow the tutorials, because I am starting to
work with the EV3 as a hobby now. Although, formerly, I have developed
business applications in Java, robot control is quite new for me.

So I am currently trying out all example programs. I got the impression
that this motor issue might be not reproducible until today it did not
go away after a reboot..

Nevertheless I will install OpenJDK12 tomorrow morning and try again!

Thanks for your support!
Roland

jabrena, 2018-11-08:
I think that the problem is not related with the JVM.

In the past, sometimes, I had the same problem and this is the reason that I added the following delay with 0.5 seconds:

``` java
    /**
     * Constructor
     * @param motorPort motor port
     * @param moveP moveP
     * @param moveI moveI
     * @param moveD moveD
     * @param holdP holdP
     * @param holdI holdI
     * @param holdD holdD
     * @param offset offset
     * @param maxSpeed maxSpeed
     */
    public BaseRegulatedMotor(final Port motorPort, float moveP, float moveI, float moveD,
                              float holdP, float holdI, float holdD, int offset, int maxSpeed) {

        List<RegulatedMotorListener> list = new ArrayList<>();
        listenerList = Collections.synchronizedList(list);

        if(log.isInfoEnabled())
            log.info("Configuring motor connected on Port: {}", motorPort.getName());

        MAX_SPEED_AT_9V = maxSpeed;
        final EV3DevPlatforms ev3DevPlatforms = EV3DevPlatforms.getInstance();
        final String port = ev3DevPlatforms.getMotorPort(motorPort);

        if(log.isDebugEnabled())
            log.debug("Detecting motor on port: {}", port);
        this.detect(LEGO_PORT, port);
        if(log.isDebugEnabled())
            log.debug("Setting port in mode: {}", TACHO_MOTOR);
        this.setStringAttribute(MODE, TACHO_MOTOR);
        Delay.msDelay(500);
        this.detect(TACHO_MOTOR, port);
        Delay.msDelay(500);
        this.setStringAttribute(COMMAND, RESET);
        if(log.isDebugEnabled())
            log.debug("Motor ready to use on Port: {}",motorPort.getName());
    }
```

Maybe I could increase a bit more that delay. The reason to have the delay, is because in some way EV3Dev, change internally the folder of the motor and it is necessary to wait a bit. Maybe in the future, we could create some kind of more sophisticated system with CompletableFuture... but at the moment, this is the current solution. Normally, when appear this kind of problem disconnecting the motor and connecting again, the problem is solved. You could have the same problem with the gyro. 

But trust me that the problem is not the JVM.

If you add in the example a new delay before the method setPower a new delay, the probability of this scenario should not exist. 

```
package ev3dev.actuators.lego.motors;
import lejos.hardware.port.MotorPort;
import lejos.utility.Delay;
public class UnregulatedMotorDemo {
	// Motor Definition
	private static UnregulatedMotor umotor1 = new UnregulatedMotor(MotorPort.C);
	// Configuration
	private final static int motorPower = 50;
	private final static int ONE_SECOND = 1000;

	public static void main(String[] args) {
		System.out.println("UnregulatedMotorDemo: Set Motor Power");
		// Set power for DC Motor
		umotor1.setPower(motorPower);
		// Testing DC-Motor 1
		umotor1.forward();
		System.out.println("forward: " + umotor1.isMoving());
		Delay.msDelay(ONE_SECOND);
		umotor1.stop();
		System.out.println("stop: " + umotor1.isMoving());
		umotor1.backward();
		System.out.println("backward: " + umotor1.isMoving());
		Delay.msDelay(ONE_SECOND);
		umotor1.stop();
		System.out.println("stop: " + umotor1.isMoving());
		umotor1.forward();
		System.out.println("forward: " + umotor1.isMoving());
		Delay.msDelay(ONE_SECOND);
		umotor1.stop();
		System.out.println("stop: " + umotor1.isMoving());
	}
}

```

Another question I have is: Are you using RCX Unregulated Motor?
This object is designed for Motors without any encoder like the RCX Unregulated motor:
If you are not using this device:

http://bricker.info/parts/5292/

[image]

or

[image]

Maybe, the result is not predictable.

If you used any EV3 Motor, you should use the example:

UnregulatedMotorDemo2.java 

This example use the unregulated features from a normal motor with EV3Dev.

Tomorrow more and welcome to the project.

Juan Antonio

jabrena, 2018-11-08:
We have in the backlog the creation of a TestingLab to execute from a Jenkins all tests.
https://github.com/ev3dev-lang-java/ev3dev-lang-java/issues/479

Using this way, we could detect the kind of problem that you describe. 
In my manual tests, (Maybe in 3 years, thousand) I mitigated this case with the delay but some times, it is possible that exist this problem.

Reviewing the case, I make the question again: What kind of motor are you using to test the feature about UnregulatedMotors.

jabrena, 2018-11-08:
I think that it will be not necessary to reinstall any JVM.
I suspect that you are using a EV3 Large/Medium Motor, but I need your confirmation.

I like the interest in the project.

Cheers

Juan Antonio

jabrena, 2018-11-08:
Hi Roland,

In that case, I don't recommend to run that example use in that case the example: **UnregulatedMotorDemo2**

Please, try this example. If the example run properly, I will close this support issue.

**Note:** 
Use the latest Snapshot:
https://jitpack.io/#ev3dev-lang-java/ev3dev-lang-java/2.4.8-SNAPSHOT

Used in the last full regression for EV3 this week.

Juan Antonio

jabrena, 2018-11-08:
Indeed, maybe it is another clue about my Delay theory. OpenJDK 10 goes a bit fast than Oracle JRE 8. Indeed that JVM load less classes than the previous version.

Definitively, in the next release, we will fix this case in the proper way with this issue:
https://github.com/ev3dev-lang-java/ev3dev-lang-java/issues/555

We will do with some CompletableFuture and scheduled timeout to avoid an infinite loop.
https://medium.com/@senanayake.kalpa/fantastic-completablefuture-allof-and-how-to-handle-errors-27e8a97144a0

**Note:** You are invited to this issue, if you are brave... :p 

Cheers

jabrena, 2018-11-09:
Good morning @robl0377 

Any progress with the tests?

- BasicMotorOperationExample 
- UnregulatedMotorDemo2 

Juan Antonio

robl0377, 2018-11-09:
Good Morning @jabrena 
I'm now using the 2.4.8-SNAPSHOT.
In UnregulatedMotorDemo I have introduced a small delay of 200ms before the setPower call: still the same error:
```
ev3dev#7|UnregulatedMotorDemo: Set Motor Power
ev3dev#7|2018-11-09 08:43:33 [main] TRACE ev3dev.utils.Sysfs - echo 50 > /sys/class/dc-motor/motor2/duty_cycle_sp
ev3dev#7|2018-11-09 08:43:33 [main] ERROR ev3dev.utils.Sysfs - File: '/sys/class/dc-motor/motor2/duty_cycle_sp' without write permissions.
ev3dev#7|Exception in thread "main" java.lang.RuntimeException: Operation not executed: /sys/class/dc-motor/motor2/duty_cycle_sp with value 50
ev3dev#7|	at ev3dev.hardware.EV3DevDevice.setStringAttribute(EV3DevDevice.java:118)
ev3dev#7|	at ev3dev.hardware.EV3DevDevice.setIntegerAttribute(EV3DevDevice.java:129)
ev3dev#7|	at ev3dev.actuators.lego.motors.BasicMotor.setPower(BasicMotor.java:56)
ev3dev#7|	at ev3dev.actuators.lego.motors.UnregulatedMotorDemo.main(UnregulatedMotorDemo.java:20)
```

jabrena, 2018-11-09:
Hi @robl0377,

In relation to the example about `UnregulatedMotorDemo`, you should not use because the objects are designed for other kind of errors.

robl0377, 2018-11-09:
Well, I am running the programs on the original (slow) EV3 Brick: _ev3dev-sysinfo -m_ says:
```
Image file:         ev3dev-stretch-ev3-generic-2018-08-06
Kernel version:     4.14.71-ev3dev-2.3.0-ev3
Brickman:           0.10.0
BogoMIPS:           148.88
Board:              board0
BOARD_INFO_HW_REV=7
BOARD_INFO_MODEL=LEGO MINDSTORMS EV3
BOARD_INFO_ROM_REV=6
BOARD_INFO_SERIAL_NUM=00165351B041
BOARD_INFO_TYPE=main
```

By the way, I tried the BasicMotorOperationExample program from the other example project: the same issue again:
```
java -server -jar /home/robot/EV3dev_Examples-0.3.0-all.jar 

ev3dev#4|2018-11-09 10:42:13 [main] INFO  ev3dev.hardware.EV3DevDistros - Providing an EV3DevDistros instance
ev3dev#4|2018-11-09 10:42:13 [main] DEBUG ev3dev.utils.Shell - Command: cat /etc/os-release
ev3dev#4|2018-11-09 10:42:15 [main] DEBUG ev3dev.hardware.EV3DevDistros - Debian Stretch detected
ev3dev#4|2018-11-09 10:42:15 [main] INFO  ev3dev.hardware.EV3DevPlatforms - Providing a EV3DevPlatforms instance
ev3dev#4|2018-11-09 10:42:15 [main] DEBUG ev3dev.hardware.EV3DevPlatforms - Detecting platform with the battery approach
ev3dev#4|2018-11-09 10:42:15 [main] DEBUG ev3dev.hardware.EV3DevFileSystem - Root Path: /sys/class
ev3dev#4|2018-11-09 10:42:15 [main] INFO  e.a.lego.motors.BaseRegulatedMotor - Configuring motor connected on Port: B
ev3dev#4|2018-11-09 10:42:15 [main] DEBUG e.a.lego.motors.BaseRegulatedMotor - Detecting motor on port: ev3-ports:outB
ev3dev#4|2018-11-09 10:42:16 [main] DEBUG ev3dev.hardware.EV3DevDevice - Detected device on path: /sys/class/lego-port/port5/address, ev3-ports:outB
ev3dev#4|2018-11-09 10:42:16 [main] DEBUG e.a.lego.motors.BaseRegulatedMotor - Setting port in mode: tacho-motor
ev3dev#4|2018-11-09 10:42:18 [main] DEBUG ev3dev.hardware.EV3DevDevice - Detected device on path: /sys/class/tacho-motor/motor7/address, ev3-ports:outB
ev3dev#4|2018-11-09 10:42:18 [main] ERROR ev3dev.utils.Sysfs - File: '/sys/class/tacho-motor/motor7/command' without write permissions.
ev3dev#4|Exception in thread "main" java.lang.ExceptionInInitializerError
ev3dev#4|Caused by: java.lang.RuntimeException: Operation not executed: /sys/class/tacho-motor/motor7/command with value reset
ev3dev#4|	at ev3dev.hardware.EV3DevDevice.setStringAttribute(EV3DevDevice.java:118)
ev3dev#4|	at ev3dev.actuators.lego.motors.BaseRegulatedMotor.<init>(BaseRegulatedMotor.java:94)
ev3dev#4|	at ev3dev.actuators.lego.motors.EV3LargeRegulatedMotor.<init>(EV3LargeRegulatedMotor.java:26)
ev3dev#4|	at example.robotics.ev3.actuator.motor.BasicMotorOperationExample.<clinit>(BasicMotorOperationExample.java:16)

> Task :remoteRun FAILED
```
However, after unplugging and re-plugging of the Motor the same program did work well (on OpenJDK 10):
```
robot@ev3dev:~$ java -server -jar EV3dev_Examples-0.3.0-all.jar
2018-11-09 10:51:32 [main] INFO  ev3dev.hardware.EV3DevDistros - Providing an EV3DevDistros instance
2018-11-09 10:51:32 [main] DEBUG ev3dev.utils.Shell - Command: cat /etc/os-release
2018-11-09 10:51:33 [main] DEBUG ev3dev.hardware.EV3DevDistros - Debian Stretch detected
2018-11-09 10:51:34 [main] INFO  ev3dev.hardware.EV3DevPlatforms - Providing a EV3DevPlatforms instance
2018-11-09 10:51:34 [main] DEBUG ev3dev.hardware.EV3DevPlatforms - Detecting platform with the battery approach
2018-11-09 10:51:34 [main] DEBUG ev3dev.hardware.EV3DevFileSystem - Root Path: /sys/class
2018-11-09 10:51:34 [main] INFO  e.a.lego.motors.BaseRegulatedMotor - Configuring motor connected on Port: B
2018-11-09 10:51:34 [main] DEBUG e.a.lego.motors.BaseRegulatedMotor - Detecting motor on port: ev3-ports:outB
2018-11-09 10:51:35 [main] DEBUG ev3dev.hardware.EV3DevDevice - Detected device on path: /sys/class/lego-port/port5/address, ev3-ports:outB
2018-11-09 10:51:35 [main] DEBUG e.a.lego.motors.BaseRegulatedMotor - Setting port in mode: tacho-motor
2018-11-09 10:51:36 [main] DEBUG ev3dev.hardware.EV3DevDevice - Detected device on path: /sys/class/tacho-motor/motor8/address, ev3-ports:outB
2018-11-09 10:51:36 [main] DEBUG e.a.lego.motors.BaseRegulatedMotor - Motor ready to use on Port: B
2018-11-09 10:51:36 [main] INFO  e.r.e.a.m.BasicMotorOperationExample - Starting motor on B
2018-11-09 10:51:37 [main] INFO  e.r.e.a.m.BasicMotorOperationExample - Forward
2018-11-09 10:51:37 [main] INFO  e.r.e.a.m.BasicMotorOperationExample - Large Motor is moving: true, at speed 500
2018-11-09 10:51:39 [main] INFO  e.r.e.a.m.BasicMotorOperationExample - Stop
2018-11-09 10:51:39 [main] INFO  e.r.e.a.m.BasicMotorOperationExample - Large Motor is moving: false, at speed 500
2018-11-09 10:51:41 [main] INFO  e.r.e.a.m.BasicMotorOperationExample - Backward
2018-11-09 10:51:41 [main] INFO  e.r.e.a.m.BasicMotorOperationExample - Large Motor is moving: true, at speed 500
2018-11-09 10:51:43 [main] INFO  e.r.e.a.m.BasicMotorOperationExample - Stop
2018-11-09 10:51:43 [main] INFO  e.r.e.a.m.BasicMotorOperationExample - Large Motor is moving: false, at speed 500
2018-11-09 10:51:45 [main] INFO  e.r.e.a.m.BasicMotorOperationExample - Forward
2018-11-09 10:51:47 [main] INFO  e.r.e.a.m.BasicMotorOperationExample - Stop
```

jabrena, 2018-11-09:
Hi @robl0377,

I have released a new snapshot with more Delay.
https://jitpack.io/#ev3dev-lang-java/ev3dev-lang-java/2.4.9-SNAPSHOT

Can you test the example `BasicMotorOperationExample`

Juan Antonio

robl0377, 2018-11-10:
The new 2.4.10-SNAPSHOT now has the Source attachment in Eclipse! 👍 Thank you very much!
(Alas, the lejos-commons library still has no source attachment)

However with JDK 10 I still get the **error**:
```
robot@ev3dev:~$ java -server -jar EV3dev_Examples-0.3.0-all.jar
2018-11-10 08:14:47 [main] INFO  ev3dev.hardware.EV3DevDistros - Providing an EV3DevDistros instance
2018-11-10 08:14:47 [main] DEBUG ev3dev.utils.Shell - Command: cat /etc/os-release
2018-11-10 08:14:48 [main] DEBUG ev3dev.hardware.EV3DevDistros - Debian Stretch detected
2018-11-10 08:14:49 [main] INFO  ev3dev.hardware.EV3DevPlatforms - Providing a EV3DevPlatforms instance
2018-11-10 08:14:49 [main] DEBUG ev3dev.hardware.EV3DevPlatforms - Detecting platform with the battery approach
2018-11-10 08:14:49 [main] DEBUG ev3dev.hardware.EV3DevFileSystem - Root Path: /sys/class
2018-11-10 08:14:49 [main] INFO  e.a.lego.motors.BaseRegulatedMotor - Configuring motor connected on Port: B
2018-11-10 08:14:49 [main] DEBUG e.a.lego.motors.BaseRegulatedMotor - Detecting motor on port: ev3-ports:outB
2018-11-10 08:14:50 [main] DEBUG ev3dev.hardware.EV3DevDevice - Detected device on path: /sys/class/lego-port/port5/address, ev3-ports:outB
2018-11-10 08:14:50 [main] DEBUG e.a.lego.motors.BaseRegulatedMotor - Setting port in mode: tacho-motor
2018-11-10 08:14:51 [main] DEBUG ev3dev.hardware.EV3DevDevice - Detected device on path: /sys/class/tacho-motor/motor5/address, ev3-ports:outB
2018-11-10 08:14:51 [main] ERROR ev3dev.utils.Sysfs - File: '/sys/class/tacho-motor/motor5/command' without write permissions.
Exception in thread "main" java.lang.ExceptionInInitializerError
Caused by: java.lang.RuntimeException: Operation not executed: /sys/class/tacho-motor/motor5/command with value reset
        at ev3dev.hardware.EV3DevDevice.setStringAttribute(EV3DevDevice.java:118)
        at ev3dev.actuators.lego.motors.BaseRegulatedMotor.<init>(BaseRegulatedMotor.java:94)
        at ev3dev.actuators.lego.motors.EV3LargeRegulatedMotor.<init>(EV3LargeRegulatedMotor.java:26)
        at example.robotics.ev3.actuator.motor.BasicMotorOperationExample.<clinit>(BasicMotorOperationExample.java:16)
```

On Oracle JDK 8 there is **NO error**:
```
robot@ev3dev:~$ /opt/ejdk1.8.0/linux_arm_sflt/jre/bin/java -server -jar EV3dev_Examples-0.3.0-all.jar
2018-11-10 08:16:32 [main] INFO  ev3dev.hardware.EV3DevDistros - Providing an EV3DevDistros instance
2018-11-10 08:16:32 [main] DEBUG ev3dev.utils.Shell - Command: cat /etc/os-release
2018-11-10 08:16:33 [main] DEBUG ev3dev.hardware.EV3DevDistros - Debian Stretch detected
2018-11-10 08:16:33 [main] INFO  ev3dev.hardware.EV3DevPlatforms - Providing a EV3DevPlatforms instance
2018-11-10 08:16:38 [main] DEBUG ev3dev.hardware.EV3DevPlatforms - Detecting platform with the battery approach
2018-11-10 08:16:38 [main] DEBUG ev3dev.hardware.EV3DevFileSystem - Root Path: /sys/class
2018-11-10 08:16:38 [main] INFO  e.a.lego.motors.BaseRegulatedMotor - Configuring motor connected on Port: B
2018-11-10 08:16:38 [main] DEBUG e.a.lego.motors.BaseRegulatedMotor - Detecting motor on port: ev3-ports:outB
2018-11-10 08:16:39 [main] DEBUG ev3dev.hardware.EV3DevDevice - Detected device on path: /sys/class/lego-port/port5/address, ev3-ports:outB
2018-11-10 08:16:39 [main] DEBUG e.a.lego.motors.BaseRegulatedMotor - Setting port in mode: tacho-motor
2018-11-10 08:16:40 [main] DEBUG ev3dev.hardware.EV3DevDevice - Detected device on path: /sys/class/tacho-motor/motor6/address, ev3-ports:outB
2018-11-10 08:16:40 [main] DEBUG e.a.lego.motors.BaseRegulatedMotor - Motor ready to use on Port: B
2018-11-10 08:16:41 [main] INFO  e.r.e.a.m.BasicMotorOperationExample - Starting motor on B
2018-11-10 08:16:41 [main] INFO  e.r.e.a.m.BasicMotorOperationExample - Forward
2018-11-10 08:16:41 [main] INFO  e.r.e.a.m.BasicMotorOperationExample - Large Motor is moving: true, at speed 500
2018-11-10 08:16:43 [main] INFO  e.r.e.a.m.BasicMotorOperationExample - Stop
2018-11-10 08:16:43 [main] INFO  e.r.e.a.m.BasicMotorOperationExample - Large Motor is moving: false, at speed 500
2018-11-10 08:16:45 [main] INFO  e.r.e.a.m.BasicMotorOperationExample - Backward
2018-11-10 08:16:45 [main] INFO  e.r.e.a.m.BasicMotorOperationExample - Large Motor is moving: true, at speed 500
2018-11-10 08:16:47 [main] INFO  e.r.e.a.m.BasicMotorOperationExample - Stop
2018-11-10 08:16:47 [main] INFO  e.r.e.a.m.BasicMotorOperationExample - Large Motor is moving: false, at speed 500
2018-11-10 08:16:49 [main] INFO  e.r.e.a.m.BasicMotorOperationExample - Forward
2018-11-10 08:16:51 [main] INFO  e.r.e.a.m.BasicMotorOperationExample - Stop
```

jabrena, 2018-11-10:
Related issue:
https://github.com/ev3dev-lang-java/ev3dev-lang-java/issues/415

jabrena, 2018-11-10:
Hi @robl0377,

Can you execute the example from the library in develop branch:
https://github.com/ev3dev-lang-java/ev3dev-lang-java

``` java
package examples;

import ev3dev.actuators.lego.motors.EV3LargeRegulatedMotor;
import lejos.hardware.port.MotorPort;
import lejos.utility.Delay;
import lombok.extern.slf4j.Slf4j;

@Slf4j
public class LargeMotorDemo {

    public static void main(String[] args) throws InterruptedException {

        LOGGER.info("Starting motors on A");
        final EV3LargeRegulatedMotor mA = new EV3LargeRegulatedMotor(MotorPort.A);
        mA.setSpeed(500);
        mA.brake();
        mA.forward();
        LOGGER.info(String.format("Large Motor is moving: %s at speed %d", mA.isMoving(), mA.getSpeed()));
        Delay.msDelay(2000);
        mA.stop();
        LOGGER.info("Stopped motors");
    }

}
```

and update from your side the delay?

For some reason, switching JVM or something, appear this behavior. In my tests, I don't have that Issue. Yesterday, I sent the snapshot 2.4.9-SNAPSHOT which I incremented the delay but today, you are using 2.4.10-SNAPSHOT which only add the sources, so we are mixing issues.

Can you test with 2.4.9-SNAPSHOT?

Juan Antonio

jabrena, 2018-11-10:
I send one execution example with 500 for the delay, current value:

```
> Task :remoteRun
Host key checking is off. It may be vulnerable to man-in-the-middle attacks.
time java -jar /home/robot/ev3dev-lang-java-2.4.10-SNAPSHOT-all.jar 

ev3dev#8|2018-11-10 09:48:38 [main] INFO  examples.LargeMotorDemo - Starting motors on A
ev3dev#8|2018-11-10 09:48:39 [main] INFO  ev3dev.hardware.EV3DevDistros - Providing an EV3DevDistros instance
ev3dev#8|2018-11-10 09:48:39 [main] DEBUG ev3dev.utils.Shell - Command: cat /etc/os-release
ev3dev#8|2018-11-10 09:48:40 [main] DEBUG ev3dev.hardware.EV3DevDistros - Debian Stretch detected
ev3dev#8|2018-11-10 09:48:40 [main] INFO  ev3dev.hardware.EV3DevPlatforms - Providing a EV3DevPlatforms instance
ev3dev#8|2018-11-10 09:48:41 [main] DEBUG ev3dev.hardware.EV3DevPlatforms - Detecting platform with the battery approach
ev3dev#8|2018-11-10 09:48:41 [main] DEBUG ev3dev.hardware.EV3DevFileSystem - Root Path: /sys/class
ev3dev#8|2018-11-10 09:48:41 [main] TRACE ev3dev.utils.Sysfs - ls /sys/class/power_supply/lego-ev3-battery
ev3dev#8|2018-11-10 09:48:41 [main] TRACE ev3dev.hardware.EV3DevPlatforms - Detected platform: EV3BRICK
ev3dev#8|2018-11-10 09:48:41 [main] INFO  e.a.lego.motors.BaseRegulatedMotor - Configuring motor connected on Port: A
ev3dev#8|2018-11-10 09:48:41 [main] DEBUG e.a.lego.motors.BaseRegulatedMotor - Detecting motor on port: ev3-ports:outA
ev3dev#8|2018-11-10 09:48:41 [main] TRACE ev3dev.hardware.EV3DevDevice - Retrieving devices in path: 
ev3dev#8|2018-11-10 09:48:41 [main] TRACE ev3dev.utils.Sysfs - ls /sys/class/lego-port
ev3dev#8|2018-11-10 09:48:41 [main] TRACE ev3dev.hardware.EV3DevDevice - Checking devices on: /sys/class/lego-port
ev3dev#8|2018-11-10 09:48:41 [main] TRACE ev3dev.hardware.EV3DevDevice - Device 1:
ev3dev#8|2018-11-10 09:48:41 [main] TRACE ev3dev.utils.Sysfs - cat /sys/class/lego-port/port6/address
ev3dev#8|2018-11-10 09:48:41 [main] TRACE ev3dev.utils.Sysfs - value: ev3-ports:outC
ev3dev#8|2018-11-10 09:48:41 [main] TRACE ev3dev.hardware.EV3DevDevice - Port expected: ev3-ports:outA, actual: ev3-ports:outC.
ev3dev#8|2018-11-10 09:48:41 [main] TRACE ev3dev.hardware.EV3DevDevice - Skipped device
ev3dev#8|2018-11-10 09:48:41 [main] TRACE ev3dev.hardware.EV3DevDevice - Device 2:
ev3dev#8|2018-11-10 09:48:41 [main] TRACE ev3dev.utils.Sysfs - cat /sys/class/lego-port/port4/address
ev3dev#8|2018-11-10 09:48:41 [main] TRACE ev3dev.utils.Sysfs - value: ev3-ports:outA
ev3dev#8|2018-11-10 09:48:41 [main] TRACE ev3dev.hardware.EV3DevDevice - Port expected: ev3-ports:outA, actual: ev3-ports:outA.
ev3dev#8|2018-11-10 09:48:41 [main] DEBUG ev3dev.hardware.EV3DevDevice - Detected device on path: /sys/class/lego-port/port4/address, ev3-ports:outA
ev3dev#8|2018-11-10 09:48:41 [main] DEBUG e.a.lego.motors.BaseRegulatedMotor - Setting port in mode: tacho-motor
ev3dev#8|2018-11-10 09:48:41 [main] TRACE ev3dev.utils.Sysfs - echo tacho-motor > /sys/class/lego-port/port4/mode
ev3dev#8|2018-11-10 09:48:42 [main] TRACE ev3dev.hardware.EV3DevDevice - Retrieving devices in path: 
ev3dev#8|2018-11-10 09:48:42 [main] TRACE ev3dev.utils.Sysfs - ls /sys/class/tacho-motor
ev3dev#8|2018-11-10 09:48:42 [main] TRACE ev3dev.hardware.EV3DevDevice - Checking devices on: /sys/class/tacho-motor
ev3dev#8|2018-11-10 09:48:42 [main] TRACE ev3dev.hardware.EV3DevDevice - Device 1:
ev3dev#8|2018-11-10 09:48:42 [main] TRACE ev3dev.utils.Sysfs - cat /sys/class/tacho-motor/motor3/address
ev3dev#8|2018-11-10 09:48:42 [main] TRACE ev3dev.utils.Sysfs - value: ev3-ports:outA
ev3dev#8|2018-11-10 09:48:42 [main] TRACE ev3dev.hardware.EV3DevDevice - Port expected: ev3-ports:outA, actual: ev3-ports:outA.
ev3dev#8|2018-11-10 09:48:42 [main] DEBUG ev3dev.hardware.EV3DevDevice - Detected device on path: /sys/class/tacho-motor/motor3/address, ev3-ports:outA
ev3dev#8|2018-11-10 09:48:43 [main] TRACE ev3dev.utils.Sysfs - echo reset > /sys/class/tacho-motor/motor3/command
ev3dev#8|2018-11-10 09:48:43 [main] DEBUG e.a.lego.motors.BaseRegulatedMotor - Motor ready to use on Port: A
ev3dev#8|2018-11-10 09:48:43 [main] TRACE ev3dev.utils.Sysfs - echo 500 > /sys/class/tacho-motor/motor3/speed_sp
ev3dev#8|2018-11-10 09:48:43 [main] TRACE ev3dev.utils.Sysfs - echo brake > /sys/class/tacho-motor/motor3/stop_action
ev3dev#8|2018-11-10 09:48:43 [main] TRACE ev3dev.utils.Sysfs - echo normal > /sys/class/tacho-motor/motor3/polarity
ev3dev#8|2018-11-10 09:48:43 [main] TRACE ev3dev.utils.Sysfs - echo 500 > /sys/class/tacho-motor/motor3/speed_sp
ev3dev#8|2018-11-10 09:48:43 [main] TRACE ev3dev.utils.Sysfs - echo run-forever > /sys/class/tacho-motor/motor3/command
ev3dev#8|2018-11-10 09:48:43 [main] TRACE ev3dev.utils.Sysfs - cat /sys/class/tacho-motor/motor3/state
ev3dev#8|2018-11-10 09:48:43 [main] TRACE ev3dev.utils.Sysfs - value: running
ev3dev#8|2018-11-10 09:48:43 [main] TRACE ev3dev.utils.Sysfs - cat /sys/class/tacho-motor/motor3/speed_sp
ev3dev#8|2018-11-10 09:48:43 [main] TRACE ev3dev.utils.Sysfs - value: 500
ev3dev#8|2018-11-10 09:48:44 [main] INFO  examples.LargeMotorDemo - Large Motor is moving: true at speed 500
ev3dev#8|2018-11-10 09:48:46 [main] TRACE ev3dev.utils.Sysfs - echo stop > /sys/class/tacho-motor/motor3/command
ev3dev#8|2018-11-10 09:48:46 [main] INFO  examples.LargeMotorDemo - Stopped motors
ev3dev#8|
ev3dev#8|real	0m40.844s
ev3dev#8|user	0m26.563s
ev3dev#8|sys	0m2.134s
```

jabrena, 2018-11-10:
Can you execute this command:

``` java
ev3dev-sysinfo -m

ev3dev#11|<!-- Copy everything between these lines -->
ev3dev#11|**System info (from `ev3dev-sysinfo`)**
ev3dev#11|```
ev3dev#11|Image file:         ev3dev-stretch-ev3-generic-2018-08-06
ev3dev#11|Kernel version:     4.14.71-ev3dev-2.3.0-ev3
ev3dev#11|Brickman:           0.10.0
ev3dev#11|BogoMIPS:           227.32
ev3dev#11|Board:              board0
ev3dev#11|BOARD_INFO_HW_REV=7
ev3dev#11|BOARD_INFO_MODEL=LEGO MINDSTORMS EV3
ev3dev#11|BOARD_INFO_ROM_REV=6
ev3dev#11|BOARD_INFO_SERIAL_NUM=001653413CAE
ev3dev#11|BOARD_INFO_TYPE=main
ev3dev#11|```
ev3dev#11|<!-- Copy everything between these lines -->
```

jabrena, 2018-11-10:
I checked your previous comments and your are using latest EV3Dev kernel. So, follow the other comments:

- Test with latest Snapshot: 2.4.9-SNAPSHOT 
- Modify the delay manually in the sources and execute. Project has the gradle task to run any program from the library.

Try to check your processes, if you experiment that JVM load the program worse in every iteration. Maybe you have some weird running in the system.

```
free // To check free memory
ps aux | sort -n -k 4 // Check process consuming more memory
```

I send some examples:

``` 
robot@ev3dev:/home$ free
              total        used        free      shared  buff/cache   available
Mem:          57644       19172        2620          28       35852       34772
Swap:         98300        8704       89596

robot@ev3dev:/home$ ps aux | sort -n -k 4
USER       PID %CPU %MEM    VSZ   RSS TTY      STAT START   TIME COMMAND
root         2  0.0  0.0      0     0 ?        S    09:12   0:00 [kthreadd]
root         4  0.0  0.0      0     0 ?        I<   09:12   0:00 [kworker/0:0H]
root         6  0.0  0.0      0     0 ?        I<   09:12   0:00 [mm_percpu_wq]
root         7  0.4  0.0      0     0 ?        S    09:12   0:13 [ksoftirqd/0]
root         8  0.0  0.0      0     0 ?        I    09:12   0:03 [rcu_preempt]
root         9  0.0  0.0      0     0 ?        I    09:12   0:00 [rcu_sched]
root        10  0.0  0.0      0     0 ?        I    09:12   0:00 [rcu_bh]
root        11  0.0  0.0      0     0 ?        S    09:12   0:00 [kdevtmpfs]
root        12  0.0  0.0      0     0 ?        I<   09:12   0:00 [netns]
root        14  0.0  0.0      0     0 ?        S    09:12   0:00 [oom_reaper]
root        15  0.0  0.0      0     0 ?        I<   09:12   0:00 [writeback]
root        16  0.0  0.0      0     0 ?        S    09:12   0:00 [kcompactd0]
root        17  0.0  0.0      0     0 ?        I<   09:12   0:00 [crypto]
root        18  0.0  0.0      0     0 ?        I<   09:12   0:00 [kblockd]
root        19  0.0  0.0      0     0 ?        I<   09:12   0:00 [watchdogd]
root        20  0.1  0.0      0     0 ?        S    09:12   0:04 [kswapd0]
root        48  0.0  0.0      0     0 ?        S    09:12   0:00 [irq/20-spi_davi]
root        49  0.3  0.0      0     0 ?        S    09:12   0:12 [spi0]
root        50  0.0  0.0      0     0 ?        S    09:12   0:00 [irq/56-spi_davi]
root        51  0.0  0.0      0     0 ?        S    09:12   0:00 [spi1]
root        54  0.0  0.0      0     0 ?        S    09:12   0:00 [irq/195-da830-m]
root        56  0.0  0.0      0     0 ?        I<   09:12   0:00 [ipv6_addrconf]
root        57  0.1  0.0      0     0 ?        S    09:12   0:05 [mmcqd/0]
root        58  0.0  0.0      0     0 ?        S    09:12   0:00 [irq/200-over_cu]
root        59  0.0  0.0      0     0 ?        S    09:12   0:00 [irq/246-pru1]
root        67  3.5  0.0      0     0 ?        S    09:12   1:56 [irq/247-ti-ads7]
root       114  0.0  0.0      0     0 ?        S    09:12   0:00 [jbd2/mmcblk0p2-]
root       115  0.0  0.0      0     0 ?        I<   09:12   0:00 [ext4-rsv-conver]
root       160  0.0  0.0      0     0 ?        I<   09:12   0:00 [rpciod]
root       161  0.0  0.0      0     0 ?        I<   09:12   0:00 [xprtiod]
root       240  0.0  0.0      0     0 ?        I<   09:12   0:00 [cfg80211]
root       450  0.0  0.0      0     0 ?        I<   09:13   0:00 [kworker/0:1H]
root       532  1.6  0.0      0     0 ?        I    09:14   0:53 [kworker/u2:3]
root       953  0.3  0.0      0     0 ?        I    09:34   0:05 [kworker/0:1]
root      1419  4.5  0.0      0     0 ?        S    09:52   0:39 [irq/249-ev3-tac]
root      1492  0.7  0.0      0     0 ?        I    09:58   0:03 [kworker/u2:0]
root      1495  0.0  0.0      0     0 ?        I    09:59   0:00 [kworker/0:0]
root      1514  0.0  0.0      0     0 ?        I    10:04   0:00 [kworker/0:2]
root      1519  0.0  0.0      0     0 ?        I    10:04   0:00 [kworker/u2:1]
root      1537  0.6  0.0      0     0 ?        D    10:05   0:00 [kworker/u2:2]
robot      682  0.0  0.8  29528   484 ?        S    09:20   0:00 (sd-pam)
avahi      359  0.0  1.1   6360   672 ?        S    09:12   0:00 avahi-daemon: chroot helper
robot      692  0.0  1.1   5300   644 pts/0    Ss+  09:20   0:00 -bash
robot      679  0.0  1.2   9644   724 ?        Ss   09:20   0:00 /lib/systemd/systemd --user
robot      689  0.0  1.2  11524   700 ?        S    09:20   0:00 sshd: robot@pts/0
root       673  0.0  1.2  11524   740 ?        Ss   09:20   0:00 sshd: robot [priv]
root       477  0.2  1.5  10016   892 ?        Ss   09:13   0:07 /sbin/wpa_supplicant -u -s -O /run/wpa_supplicant
root       495  0.0  1.5   7344   884 ?        Ss   09:14   0:00 /usr/lib/bluetooth/bluetoothd
robot      381  0.0  1.8  33784  1056 tty5     Ssl+ 09:12   0:00 /usr/bin/conrun-server
avahi      337  0.1  2.2   6360  1272 ?        Ss   09:12   0:04 avahi-daemon: running [ev3dev.local]
robot     1545  0.0  2.2   6864  1320 pts/1    S+   10:06   0:00 sort -n -k 4
root       452 20.5  2.2  11116  1276 ?        Ss   09:13  10:53 /usr/sbin/connmand -n
root       370  0.2  2.3  17184  1356 ?        Ssl  09:12   0:07 /usr/sbin/brickd
message+   340  0.6  2.4   6572  1408 ?        Ss   09:12   0:19 /usr/bin/dbus-daemon --system --address=systemd: --nofork --nopidfile --systemd-activation
root       182  0.1  2.4  13424  1436 ?        Ss   09:12   0:03 /lib/systemd/systemd-udevd
[... cut ...]
root       330  0.0  3.8   7408  2216 ?        Ss   09:12   0:01 /lib/systemd/systemd-logind
robot     1544  0.0  4.5   7672  2604 pts/1    R+   10:06   0:00 ps aux
robot     1528  0.3  5.5   5300  3208 pts/1    Ss   10:04   0:00 -bash
root         1  0.3  5.6  27924  3244 ?        Ss   09:12   0:10 /sbin/init
robot     1525  0.1  6.9  11524  4016 ?        S    10:04   0:00 sshd: robot@pts/1
root       463  2.6  8.3  52256  4816 tty1     Ssl+ 09:13   1:23 /usr/sbin/brickman
root      1512  0.3  9.5  11524  5532 ?        Ss   10:04   0:00 sshd: robot [priv]

```

jabrena, 2018-11-10:
The system is designed to run only one Java process in the system. Maybe you could have more java processes living.

Juan Antonio

jabrena, 2018-11-10:
And the test with 2.4.9-SNAPSHOT?

robl0377, 2018-11-10:
OK: 2.4.9-SNAPSHOT, program LargeMotorDemo as in your comment above:
first run after boot up (the .jar was still named 2.4.8-SNAPSHOT, but it used the 2.4.9-SNAPSHOT library:
```
Working Directory: S:\JAVA\EV3\EV3dev-java-systemtests
Gradle User Home: C:\Users\roland\.gradle
Gradle Distribution: Gradle wrapper from target build
Gradle Version: 4.7
Java Home: C:\Program Files\Java\jdk1.8.0_191
JVM Arguments: None
Program Arguments: None
Build Scans Enabled: false
Offline Mode Enabled: false
Gradle Tasks: deployAndRun

> Task :clean
> Task :compileJava
> Task :processResources
> Task :classes
> Task :shadowJar
> Task :fatJar

> Task :deploy
Host key checking is off. It may be vulnerable to man-in-the-middle attacks.
scp S:\JAVA\EV3\EV3dev-java-systemtests/build/libs/EV3dev-java-systemtests-2.4.8-SNAPSHOT-all.jar robot@192.168.1.41:/home/robot

> Task :remoteRun
Host key checking is off. It may be vulnerable to man-in-the-middle attacks.
java -server -jar /home/robot/EV3dev-java-systemtests-2.4.8-SNAPSHOT-all.jar 

ev3dev#1|2018-11-10 10:33:32 [main] INFO  examples.LargeMotorDemo - Starting motors on B
ev3dev#1|2018-11-10 10:33:32 [main] INFO  ev3dev.hardware.EV3DevDistros - Providing an EV3DevDistros instance
ev3dev#1|2018-11-10 10:33:32 [main] DEBUG ev3dev.utils.Shell - Command: cat /etc/os-release
ev3dev#1|2018-11-10 10:33:34 [main] DEBUG ev3dev.hardware.EV3DevDistros - Debian Stretch detected
ev3dev#1|2018-11-10 10:33:34 [main] INFO  ev3dev.hardware.EV3DevPlatforms - Providing a EV3DevPlatforms instance
ev3dev#1|2018-11-10 10:33:35 [main] DEBUG ev3dev.hardware.EV3DevPlatforms - Detecting platform with the battery approach
ev3dev#1|2018-11-10 10:33:35 [main] DEBUG ev3dev.hardware.EV3DevFileSystem - Root Path: /sys/class
ev3dev#1|2018-11-10 10:33:36 [main] TRACE ev3dev.utils.Sysfs - ls /sys/class/power_supply/lego-ev3-battery
ev3dev#1|2018-11-10 10:33:36 [main] TRACE ev3dev.hardware.EV3DevPlatforms - Detected platform: EV3BRICK
ev3dev#1|2018-11-10 10:33:36 [main] INFO  e.a.lego.motors.BaseRegulatedMotor - Configuring motor connected on Port: B
ev3dev#1|2018-11-10 10:33:36 [main] DEBUG e.a.lego.motors.BaseRegulatedMotor - Detecting motor on port: ev3-ports:outB
ev3dev#1|2018-11-10 10:33:36 [main] TRACE ev3dev.hardware.EV3DevDevice - Retrieving devices in path: 
ev3dev#1|2018-11-10 10:33:36 [main] TRACE ev3dev.utils.Sysfs - ls /sys/class/lego-port
ev3dev#1|2018-11-10 10:33:36 [main] TRACE ev3dev.hardware.EV3DevDevice - Checking devices on: /sys/class/lego-port
ev3dev#1|2018-11-10 10:33:36 [main] TRACE ev3dev.hardware.EV3DevDevice - Device 1:
ev3dev#1|2018-11-10 10:33:36 [main] TRACE ev3dev.utils.Sysfs - cat /sys/class/lego-port/port6/address
ev3dev#1|2018-11-10 10:33:36 [main] TRACE ev3dev.utils.Sysfs - value: ev3-ports:outC
ev3dev#1|2018-11-10 10:33:36 [main] TRACE ev3dev.hardware.EV3DevDevice - Port expected: ev3-ports:outB, actual: ev3-ports:outC.
ev3dev#1|2018-11-10 10:33:36 [main] TRACE ev3dev.hardware.EV3DevDevice - Skipped device
ev3dev#1|2018-11-10 10:33:36 [main] TRACE ev3dev.hardware.EV3DevDevice - Device 2:
ev3dev#1|2018-11-10 10:33:36 [main] TRACE ev3dev.utils.Sysfs - cat /sys/class/lego-port/port4/address
ev3dev#1|2018-11-10 10:33:36 [main] TRACE ev3dev.utils.Sysfs - value: ev3-ports:outA
ev3dev#1|2018-11-10 10:33:36 [main] TRACE ev3dev.hardware.EV3DevDevice - Port expected: ev3-ports:outB, actual: ev3-ports:outA.
ev3dev#1|2018-11-10 10:33:36 [main] TRACE ev3dev.hardware.EV3DevDevice - Skipped device
ev3dev#1|2018-11-10 10:33:36 [main] TRACE ev3dev.hardware.EV3DevDevice - Device 3:
ev3dev#1|2018-11-10 10:33:36 [main] TRACE ev3dev.utils.Sysfs - cat /sys/class/lego-port/port2/address
ev3dev#1|2018-11-10 10:33:36 [main] TRACE ev3dev.utils.Sysfs - value: ev3-ports:in3
ev3dev#1|2018-11-10 10:33:36 [main] TRACE ev3dev.hardware.EV3DevDevice - Port expected: ev3-ports:outB, actual: ev3-ports:in3.
ev3dev#1|2018-11-10 10:33:36 [main] TRACE ev3dev.hardware.EV3DevDevice - Skipped device
ev3dev#1|2018-11-10 10:33:36 [main] TRACE ev3dev.hardware.EV3DevDevice - Device 4:
ev3dev#1|2018-11-10 10:33:36 [main] TRACE ev3dev.utils.Sysfs - cat /sys/class/lego-port/port0/address
ev3dev#1|2018-11-10 10:33:36 [main] TRACE ev3dev.utils.Sysfs - value: ev3-ports:in1
ev3dev#1|2018-11-10 10:33:36 [main] TRACE ev3dev.hardware.EV3DevDevice - Port expected: ev3-ports:outB, actual: ev3-ports:in1.
ev3dev#1|2018-11-10 10:33:36 [main] TRACE ev3dev.hardware.EV3DevDevice - Skipped device
ev3dev#1|2018-11-10 10:33:36 [main] TRACE ev3dev.hardware.EV3DevDevice - Device 5:
ev3dev#1|2018-11-10 10:33:36 [main] TRACE ev3dev.utils.Sysfs - cat /sys/class/lego-port/port7/address
ev3dev#1|2018-11-10 10:33:36 [main] TRACE ev3dev.utils.Sysfs - value: ev3-ports:outD
ev3dev#1|2018-11-10 10:33:36 [main] TRACE ev3dev.hardware.EV3DevDevice - Port expected: ev3-ports:outB, actual: ev3-ports:outD.
ev3dev#1|2018-11-10 10:33:36 [main] TRACE ev3dev.hardware.EV3DevDevice - Skipped device
ev3dev#1|2018-11-10 10:33:36 [main] TRACE ev3dev.hardware.EV3DevDevice - Device 6:
ev3dev#1|2018-11-10 10:33:36 [main] TRACE ev3dev.utils.Sysfs - cat /sys/class/lego-port/port5/address
[... 125 lines cut ...]
ev3dev#3|2018-11-10 10:36:16 [main] TRACE ev3dev.hardware.EV3DevDevice - Port expected: ev3-ports:outB, actual: ev3-ports:outB.
ev3dev#3|2018-11-10 10:36:16 [main] DEBUG ev3dev.hardware.EV3DevDevice - Detected device on path: /sys/class/tacho-motor/motor5/address, ev3-ports:outB
[... cut ...]
ev3dev#3|2018-11-10 10:36:18 [main] INFO  examples.LargeMotorDemo - Large Motor is moving: true at speed 500
ev3dev#3|2018-11-10 10:36:20 [main] TRACE ev3dev.utils.Sysfs - echo stop > /sys/class/tacho-motor/motor5/command
ev3dev#3|2018-11-10 10:36:20 [main] INFO  examples.LargeMotorDemo - Stopped motors

> Task :deployAndRun

Deprecated Gradle features were used in this build, making it incompatible with Gradle 5.0.
See https://docs.gradle.org/4.7/userguide/command_line_interface.html#sec:command_line_warnings

BUILD SUCCESSFUL in 1m 24s
8 actionable tasks: 8 executed
```

robl0377, 2018-11-10:
- A second run with the EV3dev-java-systemtests-2.4.9-SNAPSHOT-all.jar still worked correctly.
- Reboot ...
- Another run with the EV3dev-java-systemtests-2.4.9-SNAPSHOT-all.jar still worked correctly.
.. strange..

jabrena, 2018-11-10:
I will decrease my BogoMIPS to the original speed to check with your same configuration.

Can you increase the delay to one second manually?

jabrena, 2018-11-10:
It is good signal, but in the other hand, it is an evidence that we need to improve this method a bit with a better synchronization between ev3dev execution and the java execution. It is an asynchronous process.

robl0377, 2018-11-10:
For those tests above I did not manipulate the library code! I just ran your example program with the 2.4.9-SNAPSHOT library from github.

robl0377, 2018-11-10:
Hmm.. So I don't need to play with some internal delays in the BaseRegulatedMotor class, which could depend on environmental conditions (temperature, voltage, age of the motor..)?

jabrena, 2018-11-10:
@robl0377 No, 

I only need that your run 5 times the same example with the snapshot 2.4.9-SNAPSHOT directly. If the execution is right, I will merge the new delay. 

I decrease my EV3Brick with Stretch to the original speed:

```
ev3dev-sysinfo -m

ev3dev#4|<!-- Copy everything between these lines -->
ev3dev#4|**System info (from `ev3dev-sysinfo`)**
ev3dev#4|```
ev3dev#4|Image file:         ev3dev-stretch-ev3-generic-2018-08-06
ev3dev#4|Kernel version:     4.14.71-ev3dev-2.3.0-ev3
ev3dev#4|Brickman:           0.10.0
ev3dev#4|BogoMIPS:           148.88
ev3dev#4|Board:              board0
ev3dev#4|BOARD_INFO_HW_REV=7
ev3dev#4|BOARD_INFO_MODEL=LEGO MINDSTORMS EV3
ev3dev#4|BOARD_INFO_ROM_REV=6
ev3dev#4|BOARD_INFO_SERIAL_NUM=001653413CAE
ev3dev#4|BOARD_INFO_TYPE=main
ev3dev#4|```
ev3dev#4|<!-- Copy everything between these lines -->
```

And the code has the similar times:

With EV3 Overclocked:

```
ev3dev#8|real	0m40.844s
ev3dev#8|user	0m26.563s
ev3dev#8|sys	0m2.134s
```

With normal speed:

```
time java -jar /home/robot/ev3dev-lang-java-2.4.10-SNAPSHOT-all.jar 

ev3dev#3|2018-11-10 11:36:02 [main] INFO  examples.LargeMotorDemo - Starting motors on A
ev3dev#3|2018-11-10 11:36:03 [main] INFO  ev3dev.hardware.EV3DevDistros - Providing an EV3DevDistros instance
ev3dev#3|2018-11-10 11:36:03 [main] DEBUG ev3dev.utils.Shell - Command: cat /etc/os-release
ev3dev#3|2018-11-10 11:36:05 [main] DEBUG ev3dev.hardware.EV3DevDistros - Debian Stretch detected
ev3dev#3|2018-11-10 11:36:05 [main] INFO  ev3dev.hardware.EV3DevPlatforms - Providing a EV3DevPlatforms instance
ev3dev#3|2018-11-10 11:36:06 [main] DEBUG ev3dev.hardware.EV3DevPlatforms - Detecting platform with the battery approach
ev3dev#3|2018-11-10 11:36:06 [main] DEBUG ev3dev.hardware.EV3DevFileSystem - Root Path: /sys/class
ev3dev#3|2018-11-10 11:36:06 [main] TRACE ev3dev.utils.Sysfs - ls /sys/class/power_supply/lego-ev3-battery
ev3dev#3|2018-11-10 11:36:06 [main] TRACE ev3dev.hardware.EV3DevPlatforms - Detected platform: EV3BRICK
ev3dev#3|2018-11-10 11:36:06 [main] INFO  e.a.lego.motors.BaseRegulatedMotor - Configuring motor connected on Port: A
ev3dev#3|2018-11-10 11:36:06 [main] DEBUG e.a.lego.motors.BaseRegulatedMotor - Detecting motor on port: ev3-ports:outA
ev3dev#3|2018-11-10 11:36:06 [main] TRACE ev3dev.hardware.EV3DevDevice - Retrieving devices in path: 
ev3dev#3|2018-11-10 11:36:06 [main] TRACE ev3dev.utils.Sysfs - ls /sys/class/lego-port
ev3dev#3|2018-11-10 11:36:06 [main] TRACE ev3dev.hardware.EV3DevDevice - Checking devices on: /sys/class/lego-port
ev3dev#3|2018-11-10 11:36:06 [main] TRACE ev3dev.hardware.EV3DevDevice - Device 1:
ev3dev#3|2018-11-10 11:36:06 [main] TRACE ev3dev.utils.Sysfs - cat /sys/class/lego-port/port6/address
ev3dev#3|2018-11-10 11:36:07 [main] TRACE ev3dev.utils.Sysfs - value: ev3-ports:outC
ev3dev#3|2018-11-10 11:36:07 [main] TRACE ev3dev.hardware.EV3DevDevice - Port expected: ev3-ports:outA, actual: ev3-ports:outC.
ev3dev#3|2018-11-10 11:36:07 [main] TRACE ev3dev.hardware.EV3DevDevice - Skipped device
ev3dev#3|2018-11-10 11:36:07 [main] TRACE ev3dev.hardware.EV3DevDevice - Device 2:
ev3dev#3|2018-11-10 11:36:07 [main] TRACE ev3dev.utils.Sysfs - cat /sys/class/lego-port/port4/address
ev3dev#3|2018-11-10 11:36:07 [main] TRACE ev3dev.utils.Sysfs - value: ev3-ports:outA
ev3dev#3|2018-11-10 11:36:07 [main] TRACE ev3dev.hardware.EV3DevDevice - Port expected: ev3-ports:outA, actual: ev3-ports:outA.
ev3dev#3|2018-11-10 11:36:07 [main] DEBUG ev3dev.hardware.EV3DevDevice - Detected device on path: /sys/class/lego-port/port4/address, ev3-ports:outA
ev3dev#3|2018-11-10 11:36:07 [main] DEBUG e.a.lego.motors.BaseRegulatedMotor - Setting port in mode: tacho-motor
ev3dev#3|2018-11-10 11:36:07 [main] TRACE ev3dev.utils.Sysfs - echo tacho-motor > /sys/class/lego-port/port4/mode
ev3dev#3|2018-11-10 11:36:07 [main] TRACE ev3dev.hardware.EV3DevDevice - Retrieving devices in path: 
ev3dev#3|2018-11-10 11:36:08 [main] TRACE ev3dev.utils.Sysfs - ls /sys/class/tacho-motor
ev3dev#3|2018-11-10 11:36:08 [main] TRACE ev3dev.hardware.EV3DevDevice - Checking devices on: /sys/class/tacho-motor
ev3dev#3|2018-11-10 11:36:08 [main] TRACE ev3dev.hardware.EV3DevDevice - Device 1:
ev3dev#3|2018-11-10 11:36:08 [main] TRACE ev3dev.utils.Sysfs - cat /sys/class/tacho-motor/motor2/address
ev3dev#3|2018-11-10 11:36:08 [main] TRACE ev3dev.utils.Sysfs - value: ev3-ports:outA
ev3dev#3|2018-11-10 11:36:08 [main] TRACE ev3dev.hardware.EV3DevDevice - Port expected: ev3-ports:outA, actual: ev3-ports:outA.
ev3dev#3|2018-11-10 11:36:08 [main] DEBUG ev3dev.hardware.EV3DevDevice - Detected device on path: /sys/class/tacho-motor/motor2/address, ev3-ports:outA
ev3dev#3|2018-11-10 11:36:08 [main] TRACE ev3dev.utils.Sysfs - echo reset > /sys/class/tacho-motor/motor2/command
ev3dev#3|2018-11-10 11:36:08 [main] DEBUG e.a.lego.motors.BaseRegulatedMotor - Motor ready to use on Port: A
ev3dev#3|2018-11-10 11:36:08 [main] TRACE ev3dev.utils.Sysfs - echo 500 > /sys/class/tacho-motor/motor2/speed_sp
ev3dev#3|2018-11-10 11:36:08 [main] TRACE ev3dev.utils.Sysfs - echo brake > /sys/class/tacho-motor/motor2/stop_action
ev3dev#3|2018-11-10 11:36:08 [main] TRACE ev3dev.utils.Sysfs - echo normal > /sys/class/tacho-motor/motor2/polarity
ev3dev#3|2018-11-10 11:36:08 [main] TRACE ev3dev.utils.Sysfs - echo 500 > /sys/class/tacho-motor/motor2/speed_sp
ev3dev#3|2018-11-10 11:36:08 [main] TRACE ev3dev.utils.Sysfs - echo run-forever > /sys/class/tacho-motor/motor2/command
[... cut ...]
ev3dev#3|2018-11-10 11:36:09 [main] TRACE ev3dev.utils.Sysfs - cat /sys/class/tacho-motor/motor2/speed_sp
ev3dev#3|2018-11-10 11:36:09 [main] TRACE ev3dev.utils.Sysfs - value: 500
ev3dev#3|2018-11-10 11:36:09 [main] INFO  examples.LargeMotorDemo - Large Motor is moving: true at speed 500
ev3dev#3|2018-11-10 11:36:11 [main] TRACE ev3dev.utils.Sysfs - echo stop > /sys/class/tacho-motor/motor2/command
ev3dev#3|2018-11-10 11:36:11 [main] INFO  examples.LargeMotorDemo - Stopped motors
ev3dev#3|
ev3dev#3|real	0m52.572s
ev3dev#3|user	0m34.847s
ev3dev#3|sys	0m2.619s
```

JakubVanek, 2018-11-10:
@jabrena I propose this solution:
```java
	public static boolean writeString(final String filePath, final String value) {
		if(log.isTraceEnabled()) log.trace("echo '{}' > '{}'", value, filePath);
		
		byte[] bytes = value.getBytes();
		Path path = Paths.get(filePath);
		
		for (int tries = 1; tries <= 10; tries++) {
			try {
				Files.write(path, bytes, StandardOpenOption.WRITE, StandardOpenOption.SYNC);
				if(log.isTraceEnabled()) log.trace("-> success");
				return true;
			} catch(IOException e) {
				if(log.isTraceEnabled()) log.trace("-> failed at try '{}': '{}'", tries, e.toString());
				Thread.sleep(100);
				continue;
			}
		}
		log.error("write failed!");
		return false;
	}
```

jabrena, 2018-11-10:
Hi @JakubVanek,

We are not going to introduce any **retry mechanism** in this class.
Put this comment in this issue: https://github.com/ev3dev-lang-java/ev3dev-lang-java/issues/555

jabrena, 2018-11-10:
We only need the tests from @robl0377 with the new delay.

jabrena, 2018-11-10:
Hi @robl0377,

we are waiting for you tests.
We have created a new Snapshot with sources for the library and lejos-commons:
https://jitpack.io/#ev3dev-lang-java/ev3dev-lang-java/2.4.12-SNAPSHOT

this Snapshot don't change the delay value, it has the same value than snapshot 2.4.9-SNAPSHOT:

``` java
Delay.msDelay(700);
``` 

We need to confirm that success tests to close that issue for this release.

Juan Antonio

robl0377, 2018-11-10:
To answer your question: I certainly had it running with the original 2.4.9-SNAPSHOT more than 5 times, although I did not test with different temperatures or voltages. Also my motors are a little used.

BTW: I got the book "Beginning Robotics Programming in Java with LEGO Mindstorms" by Wei Lu / Apress 2016. It gives an introduction to robotics programming with classical LeJOS. So I'm now working with the LeJOS again and enjoy the ease of work with the LeJOS Eclipse plugin and the LCD and Buttons :-)

jabrena, 2018-11-10:
Thanks for the tests @robl0377 
We will merge the change tomorrow and we will release.

**Note:** You are invited to this project when you want to use new objects from the JVM 8+ ;)

Cheers

Juan Antonio

commit ff0e1a5f (2018-11-11): Merge motor delay PR #563

Closes #554
closed 2018-11-11 by JakubVanek