ev3dev-lang-java/ev3dev-lang-java #555: Ensure that Motor/Sensor folders was created by EV3Dev properly

jabrena, 2018-11-08:
After one user question:
https://github.com/ev3dev-lang-java/ev3dev-lang-java/issues/554

maybe it is the moment to fix this possible bad user experience in the proper way. Currently, the library wait 0.5 seconds (Fixed period of time) in order that Linux+EV3Dev process mount the folder about the motor/sensor with all information. 

```
    public BaseRegulatedMotor(final Port motorPort, float moveP, float moveI, float moveD,
                              float holdP, float holdI, float holdD, int offset, int maxSpeed) {

        List<RegulatedMotorListener> list = new ArrayList<>();
        listenerList = Collections.synchronizedList(list);

        if(log.isInfoEnabled())
            log.info("Configuring motor connected on Port: {}", motorPort.getName());

        MAX_SPEED_AT_9V = maxSpeed;
        final EV3DevPlatforms ev3DevPlatforms = EV3DevPlatforms.getInstance();
        final String port = ev3DevPlatforms.getMotorPort(motorPort);

        if(log.isDebugEnabled())
            log.debug("Detecting motor on port: {}", port);
        this.detect(LEGO_PORT, port);
        if(log.isDebugEnabled())
            log.debug("Setting port in mode: {}", TACHO_MOTOR);
        this.setStringAttribute(MODE, TACHO_MOTOR);
        Delay.msDelay(500);
        this.detect(TACHO_MOTOR, port);
        Delay.msDelay(500);
        this.setStringAttribute(COMMAND, RESET);
        if(log.isDebugEnabled())
            log.debug("Motor ready to use on Port: {}",motorPort.getName());
    }
```

Maybe, it could be interesting to create a POC to see what is the best approach to do it.

Basically, we should introduce a asynchronous process to wait for the existency of a determinated number of folders.

jabrena, 2018-11-08:
Technical links for the future development:
https://medium.com/@senanayake.kalpa/fantastic-completablefuture-allof-and-how-to-handle-errors-27e8a97144a0
http://iteratrlearning.com/java9/2016/09/13/java9-timeouts-completablefutures.html
http://ev3dev-lang-java.github.io/docs/api/latest/ev3dev-lang-java/ev3dev/utils/Sysfs.html (getElements, existPath, existFile)

jabrena, 2018-11-08:
**Analysis:**

any motor folder in EV3Dev has the following content:
```

robot@ev3dev:/sys/class/tacho-motor/motor0$ ls
address        driver_name    polarity      ramp_up_sp  stop_action
command        duty_cycle     position      speed       stop_actions
commands       duty_cycle_sp  position_sp   speed_pid   subsystem
count_per_rot  hold_pid       power         speed_sp    time_sp
device         max_speed      ramp_down_sp  state       uevent
```

- What are the folders/files used in the library?

In the class `EV3DevMotorDevice` exist the description about all files/folders used. It could be necessary to create some kind of datastructure to interate and verify the existence.

The file doesn't differenciate between files, folders and states. It is a refactoring point.

```
    protected static final String POSITION_SP = "position_sp";
    protected static final String POSITION = "position";
    protected static final String SPEED = "speed_sp";
    protected static final String DUTY_CYCLE = "duty_cycle_sp";
    protected static final String COMMAND = "command";
    protected static final String STOP_COMMAND = "stop_action";
    protected static final String STATE = "state";
    protected static final String POLARITY = "polarity";
```

For this POC, we will group this files in a map to iterate ina

jabrena, 2018-11-08:
One idea from StackOverflow to manage retry behavior:
https://stackoverflow.com/questions/40485398/retry-logic-with-completablefuture

robl0377, 2018-11-09:
Maybe the following error fits into this same issue:
I was running the ev3dev.sensors.ev3.ColorSensorMultipleDemo:
```java
package ev3dev.sensors.ev3;
import java.util.concurrent.ExecutionException;
import java.util.concurrent.ExecutorService;
import java.util.concurrent.Executors;
import java.util.concurrent.Future;
import lejos.hardware.port.SensorPort;
import lejos.robotics.SampleProvider;
import lejos.utility.Delay;

public class ColorSensorMultipleDemo {
    //Robot Configuration
    private static EV3ColorSensor color1 = new EV3ColorSensor(SensorPort.S1);
    private static EV3ColorSensor color2 = new EV3ColorSensor(SensorPort.S2);
    //Configuration
    private static int HALF_SECOND = 500;
    public static void main(String[] args) {
        System.out.println("ColorSensorMultipleDemo: Start Threads");
        for (int i = 0; i < 5; i++) {
            taskInParallel1();
            taskInParallel2();
        }
        Delay.msDelay(HALF_SECOND);
    }
    private static void taskInParallel1(){
        final ExecutorService executor = Executors.newFixedThreadPool(2);
        final Future<Integer> future1 = executor.submit(() -> {
            runAmbientMode(color2);
            return 1;
        });
        final Future<Integer> future2 = executor.submit(() -> {
            runRGBMode(color1);
            return 1;
        });
        try {
            future1.get();
            future2.get();
            executor.shutdownNow();
        } catch (InterruptedException e) {
            throw new RuntimeException(e);
        } catch (ExecutionException e) {
            throw new RuntimeException(e);
        }
    }
    private static void taskInParallel2(){
        final ExecutorService executor = Executors.newFixedThreadPool(2);
        final Future<Integer> future1 = executor.submit(() -> {
            runAmbientMode(color1);
            return 1;
        });
        final Future<Integer> future2 = executor.submit(() -> {
            runRGBMode(color2);
            return 1;
        });
        try {
            future1.get();
            future2.get();
            executor.shutdownNow();
        } catch (InterruptedException e) {
            throw new RuntimeException(e);
        } catch (ExecutionException e) {
            throw new RuntimeException(e);
        }
    }
    private static void runRGBMode(final EV3ColorSensor sensor){
        //RGB
        System.out.println("Switching to RGB Mode");
        SampleProvider sp = sensor.getRGBMode();
[... 73 lines cut ...]
        at java.base/java.util.concurrent.FutureTask.run(Unknown Source)
        at java.base/java.util.concurrent.ThreadPoolExecutor.runWorker(Unknown Source)
        at java.base/java.util.concurrent.ThreadPoolExecutor$Worker.run(Unknown Source)
        at java.base/java.lang.Thread.run(Unknown Source)
Caused by: java.io.IOException: No such device or address
        at java.base/sun.nio.ch.FileDispatcherImpl.read0(Native Method)
        at java.base/sun.nio.ch.FileDispatcherImpl.read(Unknown Source)
        at java.base/sun.nio.ch.IOUtil.readIntoNativeBuffer(Unknown Source)
        at java.base/sun.nio.ch.IOUtil.read(Unknown Source)
        at java.base/sun.nio.ch.FileChannelImpl.read(Unknown Source)
        at java.base/sun.nio.ch.ChannelInputStream.read(Unknown Source)
        at java.base/sun.nio.ch.ChannelInputStream.read(Unknown Source)
        at java.base/sun.nio.ch.ChannelInputStream.read(Unknown Source)
        at java.base/sun.nio.cs.StreamDecoder.readBytes(Unknown Source)
        at java.base/sun.nio.cs.StreamDecoder.implRead(Unknown Source)
        at java.base/sun.nio.cs.StreamDecoder.read(Unknown Source)
        at java.base/java.io.InputStreamReader.read(Unknown Source)
        at java.base/java.io.BufferedReader.fill(Unknown Source)
        at java.base/java.io.BufferedReader.readLine(Unknown Source)
        at java.base/java.io.BufferedReader.readLine(Unknown Source)
        at java.base/java.nio.file.Files.readAllLines(Unknown Source)
        at ev3dev.utils.Sysfs.readString(Sysfs.java:72)
        ... 8 more
```
running under Oracle JDK 8 gives a similar error.

JakubVanek, 2019-09-02:
List of udev events:
```
udevadm monitor
```
View files in a folder with permissions:
```
ls -la /some/path
```
What to check:
* item existence: https://stackoverflow.com/a/18371026
* item permissions: check for `ev3dev` with https://stackoverflow.com/a/8845198.