ev3dev-lang-java/ev3dev-lang-java #556: Source attachment for Dependencies (in Eclipse)

robl0377, 2018-11-09:
I am using **Eclipse 2018-09 for Java Developers**. It has Support for Gradle projects. 

A Gradle Project update automatically adds the dependent libraries to the Eclipse Gradle project.  All libraries have the class files (in a .jar) and the necessary source file archive attached, so Eclipse shows me the Source code of a class in the library on demand.

However, the **ev3dev-lang-java-2.4.8-SNAPSHOT.jar** and the **lejos-commons-0.7.1.jar** do not define any source attachment (for Eclipse), while all other libraries do. This is quite an obstacle to getting to know the inner workings of those ev3dev libraries.

I am not yet familiar with the build process of those ev3dev libraries, but I suppose this is just a small modification in the maven definition of those libraries, right?

jabrena, 2018-11-09:
Hi @robl0377 thanks for the comment. I have added the support.

https://jitpack.io/com/github/ev3dev-lang-java/ev3dev-lang-java/2.4.10-SNAPSHOT/

We will add the support in this release.

Juan Antonio

jabrena, 2018-11-09:
Fixed

closed 2018-11-09 by jabrena