ev3dev-lang-java/ev3dev-lang-java #557: Keyboard focus for EV3 Brick buttons

robl0377, 2018-11-09:
When I run a program on the EV3 brick there still is the Brickman menu visible on the LCD and the Brickman program responds to button presses. 

Nevertheless I can check the buttons in my own program, for example to wait for a key press to continue my program's run. However, I need to consider, what that key press does to the Brickman program running in parallel. 
An example program is the ev3dev.sensors.ev3.ButtonExample2 from the example programs (ev3dev-lang-java /examples)

Ideally, a user program running on the Brick should be able to grab the focus of those buttons away from the Brickman program and (automatically) give the focus back on exit.

jabrena, 2018-11-09:
Hi @robl0377, 

It is normal your comment, I will explain the solution for your doubt.

Originally, the purpose of the project is the development of robots with EV3 Brick but the brick provides some kind of actuators that it is more oriented for the user (Display, Speaker & Buttons). In EV3Dev with Debian Jessie, doesn't exist a way to disable Brickman in order to solve your behaviour, so the project provide a solution with the class Brickman.
https://github.com/ev3dev-lang-java/ev3dev-lang-java/blob/develop/src/main/java/ev3dev/utils/Brickman.java

In order to execute a program with  Brickman without any problem you have to execute the jar with sudo, (Brickman.java disable Brickman and restore later and the process require root permissions)

In order to not execute any java process with root permissions, exist a natural solution from EV3Dev, brickrun but the support is only for Debian Stretch. If you run your program with brickun, you will observe, that brickman disable brickman and all output from the logging system will be redirected to ssh. This is the clean solution.

Take a look  the tasks designed for this iteration:
https://github.com/ev3dev-lang-java/ev3dev-lang-java/blob/develop/gradle/deploy.gradle#L115

``` groovy
task remoteBrickRun {
    doLast {
        ssh.run {
            session(remotes.ev3dev) {
                println "time brickrun -- java -jar /home/robot/" + "${rootProject.name}" + "-" + version + "-all.jar "
                execute "time brickrun -- java -jar /home/robot/" + "${rootProject.name}" + "-" + version + "-all.jar "
            }
        }
    }
}
```

If you use this way, you will improve your experience.

This weekend, we will upgrade the template-project, docs and other stuff.

Cheers

Juan Antonio

robl0377, 2018-11-09:
I have added the remoteBrickRun task to my deploy.gradle and run the ev3dev.sensors.ev3.ButtonExample2 program. It did work, though it took extremely long to start. 
However, I got the stdout to the EV3 LCD and Brickman was not visible. In the end there was an error. Here is the console output:
```
> Task :remoteBrickRun
time brickrun -- java -jar /home/robot/EV3dev-java-systemtests-2.4.8-SNAPSHOT-all.jar 

ev3dev#1|Remote process ended due to signal: Killed
ev3dev#1|
ev3dev#1|real	4m47.900s
ev3dev#1|user	0m0.637s
ev3dev#1|sys	0m0.715s

> Task :remoteBrickRun FAILED
Failed command ev3dev#1 with status 1: time brickrun -- java -jar /home/robot/EV3dev-java-systemtests-2.4.8-SNAPSHOT-all.jar 

FAILURE: Build failed with an exception.
* Where:
Script 'S:\JAVA\EV3\EV3dev-java-systemtests\deploy.gradle' line: 47
* What went wrong:
Execution failed for task ':remoteBrickRun'.
> Command returned exit status 1: time brickrun -- java -jar /home/robot/EV3dev-java-systemtests-2.4.8-SNAPSHOT-all.jar 
...
BUILD FAILED in 5m 51s
```

So somehow it works but in my opinion this is no elegant solution. (No offence intended!)
Maybe I don't know enough about robot programming, but on the other hand every Linux system can handle a single keyboard and multiple programs that use it (although I do not know how this works)..
However, I understand that for usual machine programs the buttons are not really important.

jabrena, 2018-11-09:
Don't worry @robl0377 for the comment, it is constructive.

I forget that you have add in the resources a logback configuration. I will paste in few hours one example. (I am in the office)

jabrena, 2018-11-09:
Add this file in your resources:

**logback.xml**

``` java
<?xml version="1.0" encoding="UTF-8"?>
<configuration>
    <appender name="STDERR" class="ch.qos.logback.core.ConsoleAppender">
        <target>System.err</target>
        <encoder class="ch.qos.logback.classic.encoder.PatternLayoutEncoder">
            <Pattern>
                %d{yyyy-MM-dd HH:mm:ss} [%thread] %-5level %logger{36} - %msg%n
            </Pattern>
        </encoder>
    </appender>

    <logger name="ch.qos.logback.core" level="OFF" />
    <logger name="ev3dev.hardware" level="TRACE" />
    <logger name="ev3dev.utils" level="TRACE" />

    <root level="TRACE">
        <appender-ref ref="STDERR"/>
    </root>

</configuration>
```

robl0377, 2018-11-09:
Thanks!
Actually I already have a similar logback.xml, that writes to STDOUT.

jabrena, 2018-11-09:
Try this one:

```  xml
<root level="TRACE">
        <appender-ref ref="STDERR"/>
    </root>
````

JakubVanek, 2018-11-09:
The behavior with stdout/stderr is expected: https://github.com/ev3dev-lang-java/ev3dev-lang-java/blob/develop/docs/adr/adr-logging-support.md

> We recommend to move away from using System.out.println("Your message") because using stdout is not the same as using stderr or a logging interface.
>
>The reason you shouldn't use System.out is that we depend on output redirection provided by ev3dev's brickrun. stderr is redirected via SSH to the user when running remotely. If you use System.out, the message will be displayed on the brick display only. On the other hand, this might be intentional, in which case you should use System.out.

robl0377, 2018-11-09:
Ah good to know!
Also after a reboot the EV3 now runs faster. Maybe I should do this always after running several test programs..

JakubVanek, 2018-11-09:
> Maybe I don't know enough about robot programming, but on the other hand every Linux system can handle a single keyboard and multiple programs that use it (although I do not know how this works)..
However, I understand that for usual machine programs the buttons are not really important.

The way I understand it so far is that ideally, Brickman runs on one Linux VT and the program runs on another VT (this is done by `brickrun`). Kernel then redirects keyboard input to either Brickman or the program - the keystrokes are not delivered simultaneously to both.

jabrena, 2018-11-09:
Hi @robl0377,

> Also after a reboot the EV3 now runs faster. Maybe I should do this always after running several test programs..

I don't understand your comment.

EV3 is not Jetson TX2 board. In this generation LEGO released a product with capacity to run Linux and a full featured JVM but the brick doesn't huge resources.
https://developer.nvidia.com/embedded/buy/jetson-tx2

Juan Antonio

robl0377, 2018-11-10:
>EV3 is not Jetson TX2 board.

Haha, that I am aware of. Nevertheless, after several hours of testing the programs load more and more sluggishly. Especially this After a reboot, the programs load faster. Maybe some cache is clogging the memory after a while...
Especially after the _remoteBrickRun_ that took about 5 minutes to run about a 1 second user program, programs loaded sluggishly.

jabrena, 2018-11-10:
It is rare @robl0377 honestly,

the usage brickrun don't offer for me that problem. What Debian Stretch Distro are you using?

Juan Antonio

jabrena, 2018-11-10:
I recommend to execute the following commands:

```
jps // To check if you have some java program living
free // To check free memory
ps aux | sort -n -k 4 // Check process consuming more memory
```

As you say, litle by litle you experiment execution problem.
It has sense for me, if the brick has less memory maybe it could affect the java execution.

jabrena, 2018-11-10:
With this information, you could continue with your tests.

closed 2018-11-10 by jabrena