ev3dev-lang-java/ev3dev-lang-java #570: Spring Planning, Release 2.5.0

jabrena, 2018-11-11:
Related issue: #480

JakubVanek, 2018-11-11:
* OpenJDK 11 Debian Package for EV3Dev
  * this is in progress, I plan to build it on Jenkins alongside the rest of the packages
* OpenCV Support
  * we need to debug the JVM, but yesterday I haven't managed to create a working debug build
* LCD Support for PiStorms
  * this will hopefully entail only trying the new LCD code

JakubVanek, 2018-11-11:
I will need to detach for a few days after finishing the debian package, a local robotic competition is getting near: https://robosoutez.fel.cvut.cz/robosoutez-2018-pro-stredoskolske-tymy (team _Duct tape will fix it, right?_)

jabrena, 2018-11-11:
Good luck in the competition!!!

jabrena, 2018-11-11:
In my case, I would like tackle with the following issues:

- Improve Motor support https://github.com/ev3dev-lang-java/ev3dev-lang-java/issues/555
- New Website (With the help of @Suriv )

For BrickPi/BrickPi3/PiStorms, I would like to have the collaboration of @mingan666 @HaginCodes if it is possible

jabrena, 2018-11-17:
After the Spring Planning, the main goals are:

- OpenJDK 11 Debian Package for EV3Dev
- Support for BrickPi3
- POC to migrate examples as a multimodule project
- Upgrade Installer to install in the brick dependencies. (One install, fast deployment, apply AppCDS)

In the next iteration, we will discuss the pengin points:

- OpenCV Support
- LCD Support for PiStorms
- Improve Motor support
- New Website

closed 2018-11-17 by jabrena