ev3dev-lang-java/ev3dev-lang-java #575: Add Travis support for main projects

jabrena, 2018-11-13:
It is necessary to review all projects in order to add Travis support.
Sometimes, we introduce minor issues that it is possible to be detected with Travis CI.

jabrena, 2018-12-04:
https://travis-ci.org/ev3dev-lang-java/lejos-commons/builds/463262609

jabrena, 2018-12-04:
It is necessary to clean Example project to unblock this issue

jabrena, 2018-12-09:
For installer, it is necessary to create a new project to execute the all methods in a Container.
I will create a new issue only specific for that area.

JakubVanek, 2018-12-09:
This is already in progress: https://github.com/JakubVanek/elj-install/tree/travis

closed 2018-12-09 by JakubVanek

jabrena, 2018-12-09:
Added Travis support for `Installer`.
https://travis-ci.org/ev3dev-lang-java/installer/builds/465727367

Related issue:
https://github.com/ev3dev-lang-java/ev3dev-lang-java/issues/618