ev3dev-lang-java/ev3dev-lang-java #576: Review this Gradle problem in combination with OpenJDK 11

jabrena, 2018-11-13:
```
Unable to find method 'sun.misc.Unsafe.defineClass(Ljava/lang/String;[BIILjava/lang/ClassLoader;Ljava/security/ProtectionDomain;)Ljava/lang/Class;'.
Possible causes for this unexpected error include:<ul><li>Gradle's dependency cache may be corrupt (this sometimes occurs after a network connection timeout.)
Re-download dependencies and sync project (requires network)</li><li>The state of a Gradle build process (daemon) may be corrupt. Stopping all Gradle daemons may solve this problem.
Stop Gradle build processes (requires restart)</li><li>Your project may be using a third-party plugin which is not compatible with the other plugins in the project or the version of Gradle requested by the project.</li></ul>In the case of corrupt Gradle processes, you can also try closing the IDE and then killing all Java processes.
```

JakubVanek, 2018-11-13:
I think you have to somehow manually include the `jdk.unsupported` module or export the package to all unnamed modules.

jabrena, 2019-08-18:
Rejected by LEAN

closed 2019-08-18 by jabrena