ev3dev-lang-java/ev3dev-lang-java #580: Review if LCD.getInstance returns a Graphics2D object

jabrena, 2018-11-16:
Currently LCD is returning a GraphicsLCD object but in order to integrate the project with many algorithms developed by third party, it could be interesting to return a Graphics2D object. GraphicsLCD is similar but not the same.

JakubVanek, 2018-11-16:
That simply means to scrap the leJOS emulation. You can then create Graphics2D from the buffer that JavaFramebuffer creates for you. Then you can draw it through flush call (or similar, I don't remember).

We can create some sort of Java2DLCD which would encapsulate this.

You can get the JavaFramebuffer through `LCD.getInstance().getFramebuffer()`

jabrena, 2018-12-08:
Reviewing the case, I think that we have a simple solution in front of our eyes. I will explain:

**Graphics2D**, is an Abstract class so, it is not possible to use directly and it is not an Interface.
https://docs.oracle.com/javase/7/docs/api/java/awt/Graphics2D.html

In our case, we have a good Interface: GraphicsLCD with many methods:
https://github.com/ev3dev-lang-java/lejos-commons/blob/master/src/main/java/lejos/hardware/lcd/GraphicsLCD.java

But it is true, that it doesn't have all methods that Graphics2D provides. So the simple solution to grow the Display support:

- Add on demand, the pending methods that we find or the community.

In that case, we don't have to add more complexity.

In my case, I find one pending method for a Fractal example: 

``` java
    public void drawOval(int x, int y, int width, int height) {
        g2d.drawOval(x, y, width, height);
    }
```

This missing method is used in this local example:
https://www.youtube.com/watch?v=iD5OcfiN3f0

At the end, in Desktop we have AWT. In our case, we have the display to manage the Framegrabber. So adding the missing methods is enough.

@JakubVanek Ok?

JakubVanek, 2018-12-08:
No. I don't want to extend the leJOS abstraction. The user should use Graphics2D directly.

JakubVanek, 2018-12-08:
If we ensure that all non-deprecated functionality in our GraphicsLCD works, then I'm OK with adding it there - then there is no reason for me to deprecate GraphicsLCD itself.

jabrena, 2018-12-08:
I will add the support for the new method in the different parts.

```
    public void drawOval(int x, int y, int width, int height) {
        g2d.drawOval(x, y, width, height);
    }
```

jabrena, 2018-12-08:
Resolved the discussion

closed 2018-12-08 by jabrena