ev3dev-lang-java/ev3dev-lang-java #582: Develop a simple IntelliJ plugin

jabrena, 2018-11-16:


robl0377, 2018-11-24:
Please don't forget us Eclipse users! 
With classic LeJOS I was happy to use their Eclipse Plugin. It would be great if ev3dev could provide one too.

jabrena, 2018-11-24:
Hi @robl0377, 

Yes, but remember that it is a Open Source project we are collaborating in this project with our spare time. In 2018, you could see that Java developers choose IntelliJ as the main Java IDE. This is the reason to start with IntelliJ.

I paste some links:
https://www.improgrammer.net/10-best-java-ides-for-programmers/
https://www.slant.co/topics/607/~best-java-ides-or-editors

I propose something:
Why not participate in the project creating a Plugin for eclipse?
It could be nice, isn't it?

robl0377, 2018-11-24:
Hmm, I'm an old hand with Eclipse, Java 6, ANT, and Subversion. Eclipse always was the best free Java IDE and until 2013 (in my last Java job) I used it professionally for application development.

Some hint from one of your links above about IntelliJ:
> David Poulet's Experience
It's not really a free IDE. The community edition is quite limited compared to Eclipse or even Netbeans. There are lots of plugins but many of them are not available in the free edition. If you spend the money though, the commercial very is probably the best IDE out there. 

So real Open Source developers should use a free IDE, don't you think? ;-)

About creating an Eclipse plugin myself: It would be quite a challenge and I don't know whether I could finish it. I think about it, though. No promises.

Actually, originally I wanted to port a program for a 2-wheel balancing robot for the Lego NXT controller from 2009, written in LeJOS NXT to some modern platform on the EV3 as a hobby. However, I'm still struggling with the development environment of ev3dev. 
So the first step of porting this program still is classic LeJOS EV3 v0.9.1beta for me: Eclipse & Plugin. Faster development turnaround time. Faster start time of the program on the brick (even with only 300MHz EV3 brick).

JakubVanek, 2018-12-28:
I have created a repository at https://github.com/ev3dev-lang-java/intellij-plugin.

jabrena, 2018-12-28:
@miguelcordovam, @robl0377, we have a new repo about the plugin. You are invited to participate!!!

Merry Christmas

jabrena, 2019-01-03:
Related repository:
https://github.com/ev3dev-lang-java/intellij-plugin

jabrena, 2020-04-18:
Reorganize the project

closed 2020-04-18 by jabrena