ev3dev-lang-java/ev3dev-lang-java #583: Upgrade Installer to install in the brick dependencies in order to improve the deployment process

jabrena, 2018-11-17:


jabrena, 2018-11-18:
In the project, the minimum libraries to install are:

```
    compile("org.slf4j:slf4j-api:1.7.25")
    compile("com.github.ev3dev-lang-java:lejos-commons:0.7.2")
    compile("net.java.dev.jna:jna:4.5.2")

    testCompile("ch.qos.logback:logback-classic:1.2.3")
```

Libraries to install in the brick:

```
mkdir /home/robot/libs
wget https://jitpack.io/com/github/ev3dev-lang-java/ev3dev-lang-java/2.4.16/ev3dev-lang-java-2.4.16.jar
```

jabrena, 2018-11-18:
Hi @JakubVanek, 

it is right this command:

```
sudo java -Xshare:dump -Xbootclasspath/a:/home/robot/libs/ev3dev-lang-java-2.4.16.jar;/home/robot/libs/jna-4.5.2.jar;/home/robot/libs/lejos-commons-0.7.2.jar;/home/robot/libs/logback-classic-1.2.3.jar;/home/robot/libs/slf4j-api-1.7.25.jar;
``` 

I am receiving the following message in the brick:

``` 
robot@ev3dev:~/libs$ sudo java -Xshare:dump -Xbootclasspath/a:/home/robot/libs/ev3dev-lang-java-2.4.16.jar;/home/robot/libs/jna-4.5.2.jar;/home/robot/libs/lejos-commons-0.7.2.jar;/home/robot/libs/logback-classic-1.2.3.jar;/home/robot/libs/slf4j-api-1.7.25.jar;
[sudo] password for robot: 
Allocated shared space: 268435456 bytes at 0x80000000
Loading classes to share ...
Loading classes to share: done.
Rewriting and linking classes ...
Rewriting and linking classes: done
Number of classes 1215
    instance classes   =  1153
    obj array classes  =    54
    type array classes =     8
Updating ConstMethods ... done. 
Removing unshareable information ... done. 
Scanning all metaspace objects ... 
Allocating RW objects ... 
Allocating RO objects ... 
Relocating embedded pointers ... 
Relocating external roots ... 
Dumping symbol table ...
Relocating SystemDictionary::_well_known_klasses[] ... 
Removing java_mirror ... done. 
mc  space:      3004 [  0.0% of total] out of      4096 bytes [ 73.3% used] at 0x80000000
rw  space:   2156232 [ 18.9% of total] out of   2158592 bytes [ 99.9% used] at 0x80001000
ro  space:   4596200 [ 40.3% of total] out of   4599808 bytes [ 99.9% used] at 0x80210000
md  space:      1280 [  0.0% of total] out of      4096 bytes [ 31.2% used] at 0x80673000
od  space:   4635148 [ 40.6% of total] out of   4636672 bytes [100.0% used] at 0x80674000
total    :  11391864 [100.0% of total] out of  11403264 bytes [ 99.9% used]
-bash: /home/robot/libs/jna-4.5.2.jar: Permission denied
-bash: /home/robot/libs/lejos-commons-0.7.2.jar: Permission denied
-bash: /home/robot/libs/logback-classic-1.2.3.jar: Permission denied
-bash: /home/robot/libs/slf4j-api-1.7.25.jar: Permission denied
``` 

Reading the following article:
https://blog.codefx.org/java/application-class-data-sharing/

I don't find in the path:

``` 
robot@ev3dev:/opt/jri-11/lib$ java -version
openjdk version "11.0.1" 2018-10-16
OpenJDK Runtime Environment (build 11.0.1+13-ev3)
OpenJDK Client VM (build 11.0.1+13-ev3, mixed mode, sharing)
robot@ev3dev:/opt/jri-11/lib$ ls
classlist           libfontmanager.so  libjsound.so      libzip.so
client              libinstrument.so   liblcms.so        modules
jexec               libj2gss.so        libmanagement.so  psfont.properties.ja
jli                 libjava.so         libmlib_image.so  psfontj2d.properties
jrt-fs.jar          libjavajpeg.so     libnet.so         security
jvm.cfg             libjawt.so         libnio.so         tzdb.dat
libawt.so           libjdwp.so         libprefs.so
libawt_headless.so  libjimage.so       librmi.so
libdt_socket.so     libjsig.so         libverify.so
``` 

Any .jsa file.

Juan Antonio

jabrena, 2018-11-18:
I have tried to create the class file list:

```
java
    -XX:+UseAppCDS
    -Xshare:dump
    -XX:SharedClassListFile=classes.lst
    -XX:SharedArchiveFile=app-cds.jsa
    --class-path app.jar
``` 

but I get an error:

```
sudo java  -Xshare:dump -XX:SharedClassListFile=classes.lst -XX:SharedArchiveFile=app-cds.jsa --class-path /a:/home/robot/libs/ev3dev-lang-java-2.4.16.jar;/home/robot/libs/jna-4.5.2.jar;/home/robot/libs/lejos-commons-0.7.2.jar;/home/robot/libs/logback-classic-1.2.3.jar;/home/robot/libs/slf4j-api-1.7.25.jar;
robot@ev3dev:~$ sudo java -Xshare:dump -XX:SharedClassListFile=classes.lst -XX:SharedArchiveFile=app-cds.jsa --class-path /a:/home/robot/libs/ev3dev-lang-java-2.4.16.jar;/home/robot/libs/jna-4.5.2.jar;/home/robot/libs/lejos-commons-0.7.2.jar;/home/robot/libs/logback-classic-1.2.3.jar;/home/robot/libs/slf4j-api-1.7.25.jar;
Allocated shared space: 268435456 bytes at 0x80000000
Loading classes to share ...
Error occurred during initialization of VM
Loading classlist failed: No such file or directory
-bash: /home/robot/libs/jna-4.5.2.jar: Permission denied
-bash: /home/robot/libs/lejos-commons-0.7.2.jar: Permission denied
-bash: /home/robot/libs/logback-classic-1.2.3.jar: Permission denied
-bash: /home/robot/libs/slf4j-api-1.7.25.jar: Permission denied
```

What is wrong?

Juan Antonio

jabrena, 2018-11-18:
I will update template project to enable appshare:

```
robot@ev3dev:~$ time java -Xshare:off HelloWorld
Hello, World

real	0m4.907s
user	0m4.233s
sys	0m0.356s
robot@ev3dev:~$ time java -Xshare:on HelloWorld
Hello, World

real	0m3.011s
user	0m2.542s
sys	0m0.266s
```

JakubVanek, 2018-11-18:
Example commands:
Generate the loaded class list so that java knows what to dump (runs the program):
 ```bash
java -cp /home/robot/classes/ev3dev-lang-java.jar -Xshare:on -XX:DumpLoadedClassList=/home/robot/appcds/program.lst -jar /home/robot/program.jar
 ```
Dump the classes:
```bash
java -cp /home/robot/classes/ev3dev-lang-java.jar -Xshare:dump -XX:SharedClassListFile=/home/robot/appcds/program.lst -XX:SharedArchiveFile=/home/robot/appcds/program.cds
```
Run the file, this time with cached classes:
```bash
java -cp /home/robot/classes/ev3dev-lang-java.jar -Xshare:on -XX:SharedArchiveFile=/home/robot/appcds/program.cds -jar /home/robot/program.jar
```

JakubVanek, 2018-11-18:
The `-Xshare:on` on is for using normal CDS on the first run to speed it up a little bit.

JakubVanek, 2018-11-18:
Maybe to explain the reasoning behind these commands:
1. First, Java needs to run the program to get the list of classes (`.lst`) that are loaded during the program lifetime. This probably saves only their names.
2. Then, Java goes through that class list and creates `.jsa` (or something, extension doesn't matter) which contains the dumped classes itself.
3. At last, when you run the program again, Java can load the classes from `.jsa`.

jabrena, 2018-11-24:
Yes, this is the idea Jakub, integrate in the installer. I didn't finish :p

JakubVanek, 2018-11-30:
Hmm, we could store the installer/libraries versions which were last installed on the brick. If a newer installer is available, it is rerun.

jabrena, 2018-12-01:
For this release, we will decouple the activity to load a jar with libraries previously loaded and later, we will try to tackle the process to use AppCDS. This last feature is a bit complicated for the users.

jabrena, 2018-12-01:
Testing to load in the brick a folder with the libraries:

```
rm -rf /home/robot/libs
mkdir /home/robot/libs
wget https://jitpack.io/com/github/ev3dev-lang-java/ev3dev-lang-java/2.4.16/ev3dev-lang-java-2.4.16.jar
wget https://jitpack.io/com/github/ev3dev-lang-java/lejos-commons/0.7.2/lejos-commons-0.7.2.jar
wget http://central.maven.org/maven2/net/java/dev/jna/jna/4.5.2/jna-4.5.2.jar
wget http://central.maven.org/maven2/org/slf4j/slf4j-api/1.7.25/slf4j-api-1.7.25.jar
wget http://central.maven.org/maven2/ch/qos/logback/logback-classic/1.2.3/logback-classic-1.2.3.jar
```

jabrena, 2018-12-01:
If you modify the build.gradle in template project:

and you put the libraries in this way:

```
dependencies {
    compileOnly("ch.qos.logback:logback-classic:1.2.3")
    compileOnly("com.github.ev3dev-lang-java:ev3dev-lang-java:2.5.0-SNAPSHOT")
    //compile("com.github.ev3dev-lang-java:ev3dev-lang-java:0.7.0") // BrickPi3/PiStorms support
}
```

You will generate a jar without any dependency:

```
Juans-MacBook-Pro:libs jabrena$ jar tf template_project_gradle-2.4.17-all.jar 
META-INF/
META-INF/MANIFEST.MF
example/
example/MyFirstRobot$1.class
example/MyFirstRobot.class
example/programming/
example/programming/HelloWorld.class
example/robotics/
example/robotics/ev3/
example/robotics/ev3/actuator/
example/robotics/ev3/actuator/LCDShowJavaLogoExample.class
example/robotics/ev3/actuator/LEDPatternsExample.class
example/robotics/ev3/actuator/motor/
example/robotics/ev3/actuator/motor/BasicMotorOperationExample.class
example/robotics/ev3/sensor/
example/robotics/ev3/sensor/BatteryDemo.class
example/robotics/ev3/sensor/DetectBlueBall.class
example/robotics/ev3/sensor/IRSensorRemoteControlExample.class
example/robotics/ev3/sensor/TouchSensorExample.class
example/robotics/ev3/sensor/TurnWithGyroSensorExample.class
example/robotics/ev3/sensor/USSensorExample.class
logback.xml
```

jabrena, 2018-12-01:
Now, the process in the computer takes few seconds:

```
08:53:16: Executing task 'deploy'...

> Task :clean
> Task :compileJava
> Task :processResources
> Task :classes
> Task :shadowJar
> Task :fatJar

> Task :deploy
Strict host key checking is off. It may be vulnerable to man-in-the-middle attacks.

Deprecated Gradle features were used in this build, making it incompatible with Gradle 5.0.
See https://docs.gradle.org/4.7/userguide/command_line_interface.html#sec:command_line_warnings

BUILD SUCCESSFUL in 4s
6 actionable tasks: 6 executed
08:53:21: Task execution finished 'deploy'.
```

jabrena, 2018-12-01:
and the execution takes few seconds:

```
time java -jar /home/robot/template_project_gradle-2.4.17-all.jar 

ev3dev#1|Hello World
ev3dev#1|
ev3dev#1|real	0m5.412s
ev3dev#1|user	0m3.306s
ev3dev#1|sys	0m0.429s

Deprecated Gradle features were used in this build, making it incompatible with Gradle 5.0.
See https://docs.gradle.org/4.7/userguide/command_line_interface.html#sec:command_line_warnings

BUILD SUCCESSFUL in 7s
```

jabrena, 2018-12-01:
The whole process combined:

```
08:55:47: Executing task 'deployAndRun'...

> Task :clean
> Task :compileJava
> Task :processResources
> Task :classes
> Task :shadowJar
> Task :fatJar

> Task :deploy
Strict host key checking is off. It may be vulnerable to man-in-the-middle attacks.

> Task :remoteRun
Strict host key checking is off. It may be vulnerable to man-in-the-middle attacks.
time java -jar /home/robot/template_project_gradle-2.4.17-all.jar 

ev3dev#3|Hello World
ev3dev#3|
ev3dev#3|real	0m4.331s
ev3dev#3|user	0m3.073s
ev3dev#3|sys	0m0.478s

> Task :deployAndRun

Deprecated Gradle features were used in this build, making it incompatible with Gradle 5.0.
See https://docs.gradle.org/4.7/userguide/command_line_interface.html#sec:command_line_warnings

BUILD SUCCESSFUL in 9s
8 actionable tasks: 8 executed
08:55:57: Task execution finished 'deployAndRun'.
```

JakubVanek, 2018-12-01:
Hi,

I think that [this HelloWorld](https://github.com/ev3dev-lang-java/template-project-gradle/blob/master/src/main/java/example/programming/HelloWorld.java) program doesn't model the typical usage at all. You need to load some of ev3dev-lang-java classes using Logback to experience the long startup time. I have an idea regarding Logback: use instead [slf4j-simple](https://mvnrepository.com/artifact/org.slf4j/slf4j-simple/1.7.25). https://stackoverflow.com/a/14545138, https://www.slf4j.org/manual.html:
> **slf4j-simple-1.8.0-beta2.jar**
Binding for Simple implementation, which outputs all events to System.err. Only messages of level INFO and higher are printed. This binding may be useful in the context of small applications.

I think it would be good to use the directory hierarchy from https://github.com/ev3dev-lang-java/ev3dev-lang-java/issues/594 .

I think that wget shouldn't be used in the end. If it is possible, it should be downloaded by Gradle and then uploaded by SSH.

Regards,

Jakub

JakubVanek, 2018-12-01:
Hmmm, I realized that the HelloWorld program didn't have to load even one of the JARs.

JakubVanek, 2018-12-01:
Hmm, I realized that the installer is already integrated in the template project. I think the installer logic is too difficult to encode in Groovy, so I reduced the installer to a monolithic script in https://github.com/ev3dev-lang-java/installer/pull/35 and modified the existing binding in https://github.com/ev3dev-lang-java/template-project-gradle/pull/12.

jabrena, 2018-12-01:
Comparing performance between Logback vs slf4j-simple

```
> Task :remoteRun2
[slf4j-api-1.7.25.jar, logback-core-1.2.3.jar, logback-classic-1.2.3.jar, jna-4.5.2.jar, lejos-commons-0.7.2.jar, ev3dev-lang-java-2.4.16.jar]
:/home/robot/libs/slf4j-api-1.7.25.jar:/home/robot/libs/logback-core-1.2.3.jar:/home/robot/libs/logback-classic-1.2.3.jar:/home/robot/libs/jna-4.5.2.jar:/home/robot/libs/lejos-commons-0.7.2.jar:/home/robot/libs/ev3dev-lang-java-2.4.16.jar
Strict host key checking is off. It may be vulnerable to man-in-the-middle attacks.
time java -Xshare:on -cp /home/robot/template_project_gradle-2.4.17-all.jar:/home/robot/libs/slf4j-api-1.7.25.jar:/home/robot/libs/logback-core-1.2.3.jar:/home/robot/libs/logback-classic-1.2.3.jar:/home/robot/libs/jna-4.5.2.jar:/home/robot/libs/lejos-commons-0.7.2.jar:/home/robot/libs/ev3dev-lang-java-2.4.16.jar example.robotics.ev3.actuator.LEDPatternsExample

ev3dev#0|2018-12-01 11:34:15 [main] INFO  e.r.ev3.actuator.LEDPatternsExample - Example with EV3 Led patterns
ev3dev#0|2018-12-01 11:34:15 [main] INFO  e.r.ev3.actuator.LEDPatternsExample - Left patterns
ev3dev#0|2018-12-01 11:34:15 [main] INFO  ev3dev.hardware.EV3DevDistros - Providing an EV3DevDistros instance
ev3dev#0|2018-12-01 11:34:15 [main] DEBUG ev3dev.utils.Shell - Command: cat /etc/os-release
ev3dev#0|2018-12-01 11:34:16 [main] DEBUG ev3dev.hardware.EV3DevDistros - Debian Stretch detected
ev3dev#0|2018-12-01 11:34:16 [main] INFO  ev3dev.hardware.EV3DevPlatforms - Providing a EV3DevPlatforms instance
ev3dev#0|2018-12-01 11:34:16 [main] DEBUG ev3dev.hardware.EV3DevPlatforms - Detecting platform with the battery approach
ev3dev#0|2018-12-01 11:34:16 [main] DEBUG ev3dev.hardware.EV3DevFileSystem - Root Path: /sys/class
ev3dev#0|2018-12-01 11:34:16 [main] TRACE ev3dev.utils.Sysfs - ls /sys/class/power_supply/lego-ev3-battery
ev3dev#0|2018-12-01 11:34:16 [main] TRACE ev3dev.hardware.EV3DevPlatforms - Detected platform: EV3BRICK
ev3dev#0|2018-12-01 11:34:16 [main] TRACE ev3dev.utils.Sysfs - echo 0 > /sys/class/leds/led0:red:brick-status/brightness
ev3dev#0|2018-12-01 11:34:16 [main] TRACE ev3dev.utils.Sysfs - echo 0 > /sys/class/leds/led0:green:brick-status/brightness
ev3dev#0|2018-12-01 11:34:18 [main] TRACE ev3dev.utils.Sysfs - echo 0 > /sys/class/leds/led0:red:brick-status/brightness
ev3dev#0|2018-12-01 11:34:18 [main] TRACE ev3dev.utils.Sysfs - echo 255 > /sys/class/leds/led0:green:brick-status/brightness
ev3dev#0|2018-12-01 11:34:19 [main] TRACE ev3dev.utils.Sysfs - echo 255 > /sys/class/leds/led0:red:brick-status/brightness
ev3dev#0|2018-12-01 11:34:19 [main] TRACE ev3dev.utils.Sysfs - echo 0 > /sys/class/leds/led0:green:brick-status/brightness
ev3dev#0|2018-12-01 11:34:20 [main] TRACE ev3dev.utils.Sysfs - echo 255 > /sys/class/leds/led0:red:brick-status/brightness
ev3dev#0|2018-12-01 11:34:20 [main] TRACE ev3dev.utils.Sysfs - echo 255 > /sys/class/leds/led0:green:brick-status/brightness
ev3dev#0|2018-12-01 11:34:21 [main] TRACE ev3dev.utils.Sysfs - echo 0 > /sys/class/leds/led0:red:brick-status/brightness
ev3dev#0|2018-12-01 11:34:21 [main] TRACE ev3dev.utils.Sysfs - echo 0 > /sys/class/leds/led0:green:brick-status/brightness
ev3dev#0|2018-12-01 11:34:22 [main] INFO  e.r.ev3.actuator.LEDPatternsExample - Right patterns
ev3dev#0|2018-12-01 11:34:22 [main] TRACE ev3dev.utils.Sysfs - echo 0 > /sys/class/leds/led1:red:brick-status/brightness
ev3dev#0|2018-12-01 11:34:22 [main] TRACE ev3dev.utils.Sysfs - echo 0 > /sys/class/leds/led1:green:brick-status/brightness
ev3dev#0|2018-12-01 11:34:23 [main] TRACE ev3dev.utils.Sysfs - echo 0 > /sys/class/leds/led1:red:brick-status/brightness
ev3dev#0|2018-12-01 11:34:23 [main] TRACE ev3dev.utils.Sysfs - echo 255 > /sys/class/leds/led1:green:brick-status/brightness
ev3dev#0|2018-12-01 11:34:24 [main] TRACE ev3dev.utils.Sysfs - echo 255 > /sys/class/leds/led1:red:brick-status/brightness
ev3dev#0|2018-12-01 11:34:24 [main] TRACE ev3dev.utils.Sysfs - echo 0 > /sys/class/leds/led1:green:brick-status/brightness
ev3dev#0|2018-12-01 11:34:25 [main] TRACE ev3dev.utils.Sysfs - echo 255 > /sys/class/leds/led1:red:brick-status/brightness
ev3dev#0|2018-12-01 11:34:25 [main] TRACE ev3dev.utils.Sysfs - echo 255 > /sys/class/leds/led1:green:brick-status/brightness
ev3dev#0|2018-12-01 11:34:26 [main] TRACE ev3dev.utils.Sysfs - echo 0 > /sys/class/leds/led1:red:brick-status/brightness
ev3dev#0|2018-12-01 11:34:26 [main] TRACE ev3dev.utils.Sysfs - echo 0 > /sys/class/leds/led1:green:brick-status/brightness
ev3dev#0|
ev3dev#0|real	0m46.563s
ev3dev#0|user	0m31.332s
ev3dev#0|sys	0m1.404s
```

```
> Task :remoteRun2
[slf4j-api-1.7.25.jar, slf4j-simple-1.7.25.jar, jna-4.5.2.jar, lejos-commons-0.7.2.jar, ev3dev-lang-java-2.4.16.jar]
:/home/robot/libs/slf4j-api-1.7.25.jar:/home/robot/libs/slf4j-simple-1.7.25.jar:/home/robot/libs/jna-4.5.2.jar:/home/robot/libs/lejos-commons-0.7.2.jar:/home/robot/libs/ev3dev-lang-java-2.4.16.jar
Strict host key checking is off. It may be vulnerable to man-in-the-middle attacks.
time java -Xshare:on -cp /home/robot/template_project_gradle-2.4.17-all.jar:/home/robot/libs/slf4j-api-1.7.25.jar:/home/robot/libs/slf4j-simple-1.7.25.jar:/home/robot/libs/jna-4.5.2.jar:/home/robot/libs/lejos-commons-0.7.2.jar:/home/robot/libs/ev3dev-lang-java-2.4.16.jar example.robotics.ev3.actuator.LEDPatternsExample

ev3dev#0|[main] INFO example.robotics.ev3.actuator.LEDPatternsExample - Example with EV3 Led patterns
ev3dev#0|[main] INFO example.robotics.ev3.actuator.LEDPatternsExample - Left patterns
ev3dev#0|[main] INFO ev3dev.hardware.EV3DevDistros - Providing an EV3DevDistros instance
ev3dev#0|[main] INFO ev3dev.hardware.EV3DevPlatforms - Providing a EV3DevPlatforms instance
ev3dev#0|[main] INFO example.robotics.ev3.actuator.LEDPatternsExample - Right patterns
ev3dev#0|
ev3dev#0|real	0m19.912s
ev3dev#0|user	0m8.223s
ev3dev#0|sys	0m0.689s
```

JakubVanek, 2018-12-01:
I think that we used `-Xlog:class+path` in the past to view which classes are loaded - there might be more interesting things.

JakubVanek, 2018-12-01:
> In the next version, for normal users, we will recommend slf4j-simple as default logger implementation. If any user require advanced features, it is possible to use logback. Only, it is necessary to load in the classpath to run.

Yup, exactly. In the past I tried to remove even SLF4J, but that would be a bad decision - it allows to swap logging backends easily.

jabrena, 2018-12-01:
New configuration:

``` groovy
//Configuration
project.ext {

    //Main program
    mainClass = "example.programming.HelloWorld"

    //Brick
    brickHost = "192.168.2.3"
    brickUser = "robot"
    brickPassword = "maker"

    //JVM
    //jvmFlags = "-verbose:class"
    jvmFlags = "-Xshare:on"
    //jvmFlags = "-Dcom.sun.management.jmxremote=true -Dcom.sun.management.jmxremote.port=7091 -Dcom.sun.management.jmxremote.authenticate=false -Dcom.sun.management.jmxremote.ssl=false"

    //Dependencies
    externalJarsPath = "/home/robot/libs/"
    externalJars = [
            "slf4j-api-1.7.25.jar",
            "slf4j-simple-1.7.25.jar",
            //"logback-core-1.2.3.jar",
            //"logback-classic-1.2.3.jar",
            "jna-4.5.2.jar",
            "lejos-commons-0.7.2.jar",
            "ev3dev-lang-java-2.4.16.jar"
    ]
}
```

Now, it is not necessary to touch 2 files: MANIFEST.MF and config.gradl
everything will be managed by config.gradle

JakubVanek, 2018-12-02:
Delivered.

closed 2018-12-02 by JakubVanek