ev3dev-lang-java/ev3dev-lang-java #585: Add task to shutdown brick

jabrena, 2018-11-21:
`shutdown -h now`

JakubVanek, 2018-11-24:
It should be possible to configure PolicyKit for sudo-less power operations and then just use:
```
systemctl reboot
systemctl poweroff
```

closed 2018-11-24 by jabrena

jabrena, 2018-11-24:
Hi @JakubVanek, 

With one line, we have the solution, it is simple...

I don't know anything about `PolicyKit` but if we have to configure something in the brick, we increase the complexity. Maybe it is not perfect, but it is not a bad solution.

JakubVanek, 2018-11-24:
Oki, it is true that this way we are not touching the system.

Some links to what I meant:
https://askubuntu.com/questions/1190/how-can-i-make-shutdown-not-require-admin-password
https://wiki.archlinux.org/index.php/allow_users_to_shutdown

jabrena, 2018-11-24:
But essentially, this is the power of a complete Linux distro. We can do it, but what is the limit?

At the end we have:

- a stable complete Linux Distro
- a stable jvm build
- a stable library
- <- Roadmap (ROS, Remote, etc...)

After some issues, we could continue developing again new features.
Later, I will show the Testing lab... Many wires...