ev3dev-lang-java/ev3dev-lang-java #594: Gradle changes proposal for installer and AppCDS

JakubVanek, 2018-11-24:
Proposal (paths need to be replaced by variables later):

Directories on the brick:
* `/home/robot/` - location for user JAR wrappers.
* `/home/robot/java/programs/` - location for user JARs.
* `/home/robot/java/libraries/` - location for ev3dev-lang-java JARs.
* `/home/robot/java/modules/` - location for future ev3dev-lang-java JMODs.
* `/home/robot/java/appcds` - location for AppCDS caches

New/changed Gradle tasks:
* java installer - install `jri-11-ev3` or another java
* rxtx installer - install `librxtx-java`
* opencv installer - install `libopencv2.4-java`
* ev3dev-lang-java installer - installs maven/another JARs to `/home/robot/java/libraries` and eventually `/home/robot/java/modules`
* manifest creation for user JARs - the libraries should not be included in the JAR, they should be picking them from the system, i.e. `/usr/share/java` and `/home/robot/java/libraries`
* user JAR uploader
  * uploads JARs to `/home/robot/java/programs`
  * creates a non-appcds shell script wrapper in `/home/robot/`
* user JAR uploader+optimizer
  * deletes appropriate cache files in `/home/robot/java/appcds`
  * uploads JARs to `/home/robot/java/programs`
  * creates an appcds-aware shell script wrapper in `/home/robot/`
  * and then two situations are possible
    * AppCDS is refreshed on the first startup
      * the script wrapper will generate the class list and dump the classes
    * AppCDS is refreshed on upload
      * a helper script called from Gradle will generate class list from specified sources and then immediately generate the AppCDS cache
        * JRI class list
        * ev3dev-lang-java libraries class list
        * user program class list

JakubVanek, 2018-11-24:
We can alternatively integrate it into the IDE plugin, but I think that even command-line users should have a chance to work with this.

jabrena, 2018-11-24:
Don't worry for the plugin, we have to wait for the initiative of the community. It will take some months. 

The folder design is **Ok**. Maybe we could start without the `AppCDS`, only the local libraries and later add that support in another iteration. For `OpenCV`, if you like, try to experiment in your side. For me, it is not working in `EV3Dev Stretch`. Try in you case to see my point of view to take the control.

For the modules support, I think that in another iteration.

The quick win, is the installation from installer the main libraries included `ev3dev-lang-java`
This idea is related to this issue: 
https://github.com/ev3dev-lang-java/ev3dev-lang-java/issues/583

We could use your folder structure: `/home/robot/java/libraries/` and use it for the execution.

We could have 2 ways to deploy: `FatJar` or `Fast`. The fast requires a previous installation of classic libraries.

For **AppCDS**, we could create a list of classes per jar and later copy to the brick and execute the statement to cache. the `.lst` should include: `jdk`, `ev3dev-lang-java`, etc... we should create a small **POC** with 2 libraries and check benefits. With this feature with get from the JVM the maximum performance but I am not sure if many people will use. If you use **BrickPi3 + RaspberryPi 3 Model B+**, the execution is immediate without any optimisation except the initial `-Xshare:dump` but I didn't enable in template project. (This another issue, I will create tomorrow to get 1 second with no effort)

The fast deployment is interesting, is a quick win for **EV3**.

JakubVanek, 2018-12-01:
Review of current solution:
* Integrating `installer.sh` in Gradle:
  * java installer - this would call `installer.sh java` (already done)
  * native library installer - this would call `installer.sh libs` (new)
* Small JARs
  * ev3dev-lang-java installer - we can iterate over the `externalJars` in [`project.ext`][1] and upload them to the library directory on the brick
  * manifest creation for user JARs - this will not be needed, arguments will be supplied at run time - `mainClass` property of [`project.ext`][1]
  * user JAR uploader/wrapper generator - [launcher-simple.sh.template][2] can be expanded with variables in [`project.ext`][1] and uploaded to the brick.

[1]: https://github.com/ev3dev-lang-java/ev3dev-lang-java/issues/583#issuecomment-443420966
[2]: https://github.com/ev3dev-lang-java/template-project-gradle/blob/8a574553ab93f59ef3e20b756075c33d9ce2a037/gradle/launcher-simple.sh.template

JakubVanek, 2018-12-02:
Delivered.

closed 2018-12-02 by JakubVanek