ev3dev-lang-java/ev3dev-lang-java #603: Review Kalman filter for sensor data.

jabrena, 2018-12-04:
Grove system apply Kalman filter for his sensor data:
http://wiki.seeedstudio.com/Grove-Barometer-High-Accuracy/

How to use current LeJOS support on sensor data?

JakubVanek, 2019-04-01:
It could either be reimplemented for this 1D purpose or mapped to the existing class.
- This is their filter with hardcoded values: https://github.com/Seeed-Studio/Grove_Barometer_HP20x/blob/master/KalmanFilter.cpp.
- LeJOS has [lejos.utility.KalmanFilter](https://github.com/ev3dev-lang-java/lejos-commons/blob/master/src/main/java/lejos/utility/KalmanFilter.java) which works directly with matrices, in this case they'd be just 1x1.

Their filter generates the noise matrices on each filtering. I don't understand the theory behind it, but technically it could be generated in advance.

jabrena, 2019-08-18:
Rejected by LEAN

closed 2019-08-18 by jabrena