ev3dev-lang-java/ev3dev-lang-java #607: Test that LCDStretch and LCDJessie work

JakubVanek, 2018-12-08:
Hi,

if we want to keep LCDStretch and LCDJessie, we should ensure that these classes are not broken. It is necessary to properly implement missing functinality and test if everything works. If something is broken and it makes no sense to preserve it in our implementation, we should deprecate it.

Regards,

Jakub

jabrena, 2018-12-09:
Hi @JakubVanek, 

If you remember, we have multiple tests in the project about LCD.
What method do you think that it is necessary to review?

Sources:
https://github.com/ev3dev-lang-java/examples/tree/master/ev3dev-lang-java/src/main/java/ev3dev/actuators/ev3

- LCDDrawIconsTest
- LCDDrawImagesTest
- LCDDrawLinesTest
- LCDDrawRectanglesAnimationTest
- LCDDrawRectanglesTest
- LCDFontTest
- LCDTest
- LCDWriteTextTest
- LCDCircles

In previous regression tests, I execute all of them.

What features, do you think that it is necessary to test?

Juan Antonio

JakubVanek, 2018-12-09:
I'm in doubt a bit about `translate(int x, int y)` and then abot various `drawImage`, `drawRegionRop`, `bitBlt` and similar.

I'm going to go against what I argued for that much - I'm actually not sure how much `drawImage` and similar are good for deprecation. If we want to preserve complete compatibility with leJOS, these could be useful - I used it once to replace a character on an on-screen-keyboard with a picture. But new programs should not rely on this at all.

jabrena, 2018-12-09:
Ideas for the examples:
https://docs.oracle.com/javase/tutorial/2d/geometry/strokeandfill.html
https://github.com/cwbollinger/highschool_java/blob/master/samples/src/org/lejos/sample/lcdui/LCDUI.java
https://github.com/dave5177/hangzhou/blob/master/game/%E4%B8%96%E7%95%8C%E5%A4%A7%E6%88%98/code_AH/WroldWar_AH/src/com/dave/worldWar/model/Soldier.java
http://www.java2s.com/Code/JavaAPI/java.awt/Graphics2Dtranslateintxinty.htm

jabrena, 2018-12-13:
Defining a checklist to track the progress of this issue.

jabrena, 2018-12-13:
``` java
    @Override
    public void drawString(String str, int x, int y, int anchor, boolean inverted) {
        Color oldFg = g2d.getColor();
        Color oldBg = g2d.getBackground();
        g2d.setColor(inverted ? Color.WHITE : Color.BLACK);
        g2d.setBackground(inverted ? Color.BLACK : Color.WHITE);

        drawString(str, x, y, anchor);

        g2d.setColor(oldFg);
        g2d.setBackground(oldBg);
    }

    @Override
    public void drawString(String str, int x, int y, int anchor) {
        FontMetrics metrics = g2d.getFontMetrics();
        int w = metrics.stringWidth(str);
        int h = metrics.getHeight();
        x = adjustX(x, w, anchor);
        y = adjustY(y, h, anchor);

        g2d.drawString(str, x, y);
    }
``` 
both methods are very different. It is necessary to review this implementation `drawString(String str, int x, int y, int anchor, boolean inverted)`

jabrena, 2018-12-13:
```
ev3dev#9| --- Running command: "time brickrun --  java -Xms64m -Xmx64m -XX:+UseSerialGC -Xshare:on -noverify -Djava.security.egd=file:/dev/./urandom   -jar /home/robot/java/programs/template_project_gradle-2.5.3-all.jar" ---
ev3dev#10|2018-12-13 20:01:50 [main] INFO ev3dev.hardware.EV3DevDistros - Providing an EV3DevDistros instance
ev3dev#10|2018-12-13 20:01:53 [main] INFO ev3dev.hardware.EV3DevPlatforms - Providing a EV3DevPlatforms instance
ev3dev#10|2018-12-13 20:01:53 [main] INFO ev3dev.actuators.LCDStretch - Instancing LCD for Stretch
ev3dev#10|2018-12-13 20:02:15 [main] INFO example.robotics.ev3.actuator.LCDWriteTextTest3 - EV3 LCD Example
ev3dev#10|Exception in thread "main" java.lang.StringIndexOutOfBoundsException: begin 2, end 9, length 7
ev3dev#10|	at java.base/java.lang.String.checkBoundsBeginEnd(Unknown Source)
```

It is necessary to review the method: `drawSubstring(String str, int offset, int len, int x, int y, int anchor)`

jabrena, 2018-12-16:
https://docs.oracle.com/javase/tutorial/2d/advanced/transforming.html

jabrena, 2018-12-16:
The pending methods doesn’t come from Graphics2D, I will check with Andy. Maybe, it is legacy code.

Next week, I will fix the errors found.

jabrena, 2018-12-19:
Question pending:
https://sourceforge.net/p/lejos/ev3/code/ci/master/tree/ev3classes/src/lejos/hardware/lcd/GraphicsLCD.java

jabrena, 2018-12-19:
Until next week, Andy will not available. I will continue with other issue.

jabrena, 2018-12-27:
Added questions:
https://lejos.sourceforge.io/forum/viewtopic.php?f=18&t=9430

jabrena, 2019-01-05:
In relation to the question, we are using normal Image class for the development.
In LeJOS, the library uses a custom implementation.
https://sourceforge.net/p/lejos/ev3/code/ci/master/tree/ev3samples/src/org/lejos/ev3/sample/graphicssample/GraphicsSample.java
https://sourceforge.net/p/lejos/ev3/code/ci/master/tree/ev3classes/src/lejos/hardware/lcd/Image.java#l132

We could deprecate that low level methods:

```
public void drawRegionRop(Image src, int sx, int sy, int w, int h, int x, int y, int anchor, int rop); PENDING
public void drawRegionRop(Image src, int sx, int sy, int w, int h, int transform, int x, int y, int anchor, int rop); PENDING
public void drawRegion(Image src, int sx, int sy, int w, int h, int transform, int x, int y, int anchor); PENDING
```

Juan Antonio

jabrena, 2019-01-05:
Later, I wil fix the issues detected in the iteration

jabrena, 2019-09-01:
Returning to this issue

jabrena, 2019-09-01:
Actions:

- [ ] Deprecate non used methods
- [ ] Review implementation for method `public void drawString(String str, int x, int y, int anchor, boolean inverted); PENDING`
- [ ] Review implementation for method `drawSubstring(String str, int offset, int len, int x, int y, int anchor)`

```
public void drawRegionRop(Image src, int sx, int sy, int w, int h, int x, int y, int anchor, int rop); PENDING
public void drawRegionRop(Image src, int sx, int sy, int w, int h, int transform, int x, int y, int anchor, int rop); PENDING
public void drawRegion(Image src, int sx, int sy, int w, int h, int transform, int x, int y, int anchor); PENDING
```

jabrena, 2019-09-01:
Example:

``` java
public class LCDDrawStringTest {

    public static Logger LOGGER = LoggerFactory.getLogger(LCDDrawStringTest.class);

    public static GraphicsLCD lcd = LCD.getInstance();

    public static void main(String[] args) {

        final int height = lcd.getHeight();
        final int width = lcd.getWidth();

        LOGGER.info("Display dimensions: W:{} H:{}", width , height);
        LOGGER.info("StrokeStyle: {}", lcd.getStrokeStyle());

        lcd.setColor(Color.BLACK);

        lcd.drawString("Hello World", 50, 50, 50);

        lcd.refresh();

        LOGGER.info("Wait");
        Delay.msDelay(10000);

        lcd.setColor(0,0,0);
        lcd.fillRect(0, 0, width, height);
        lcd.drawString("Hello World", 50, 50, 50, true);

        lcd.refresh();

        LOGGER.info("Wait");
        Delay.msDelay(10000);
    }

}
```

jabrena, 2019-09-01:
The exception in some escenarios are valid:

```
Exception in thread "main" java.lang.StringIndexOutOfBoundsException: begin 10, end 20, length 11
	at java.base/java.lang.String.checkBoundsBeginEnd(String.java:3720)
	at java.base/java.lang.String.substring(String.java:1909)
	at ev3dev.actuators.LCDStretch.drawSubstring(LCDStretch.java:181)
	at example.display.LCDDrawSubstringTest.main(LCDDrawSubstringTest.java:28)
``` 

The user need to learn to use the method substring:

``` java
    @Override
    public void drawSubstring(String str, int offset, int len,
                              int x, int y, int anchor) {
        String sub = str.substring(offset, offset + len);
        drawString(sub, x, y, anchor);
    }

Example:

``` java

public class LCDDrawSubstringTest {

    public static Logger LOGGER = LoggerFactory.getLogger(LCDDrawSubstringTest.class);

    public static GraphicsLCD lcd = LCD.getInstance();

    public static void main(String[] args) {

        final int height = lcd.getHeight();
        final int width = lcd.getWidth();

        LOGGER.info("Display dimensions: W:{} H:{}", width , height);

        lcd.setColor(Color.BLACK);
        lcd.drawSubstring("Hello World", 1, 1,20, 20, 50);
        lcd.refresh();

        //This statement will fail. The developer need to learn to use the method.
        lcd.drawSubstring("Hello World", 10, 10,60, 60, 50);
        lcd.refresh();

        LOGGER.info("Wait");
        Delay.msDelay(10000);
    }

}

```

jabrena, 2019-09-02:
Fixed

closed 2019-09-02 by jabrena