ev3dev-lang-java/ev3dev-lang-java #615: The Gradle task to install the common libraries is broken.

jabrena, 2018-12-09:
It is necessary to review the following method:

```
task getInstaller {
    group installerGroup
    description "Download component installer on the brick."
    doLast {
        ssh.run {
            session(remotes.ev3dev) {
                project.sshPrint(delegate, "mkdir -p /home/robot/java")
                project.sshPrint(delegate, "/bin/sh -c 'if grep -i jessie /etc/os-release; then wget https://raw.githubusercontent.com/ev3dev-lang-java/installer/master/installer-jessie.sh -O /home/robot/java/installer.sh; else wget https://raw.githubusercontent.com/ev3dev-lang-java/installer/master/installer.sh -O /home/robot/java/installer.sh; fi'")
                project.sshPrint(delegate, "chmod +x /home/robot/java/installer.sh")
            }
        }
    }
}
```

It generate a syntax error. It is necessary to change.

jabrena, 2018-12-09:
Restoring original code for Debian Stretch atleast:

```
 project.sshPrint(delegate, "wget https://raw.githubusercontent.com/ev3dev-lang-java/installer/master/installer.sh -O /home/robot/java/installer.sh")
```

JakubVanek, 2018-12-09:
Hmmm, it may be due to the single quote I added - it breaks the echo code.