ev3dev-lang-java/ev3dev-lang-java #619: Create a Demo which is able to replicate in another Brick

jabrena, 2018-12-09:
This is concept is inspired in `Nintendo Switch`:
https://www.youtube.com/watch?v=f5uik5fgIaI

The programs multi brick, should be auto configured.

jabrena, 2018-12-09:
Ideas:

[image]
[image]
[image]

jabrena, 2018-12-09:
https://www.zedge.net/find/ringtones/grinch

jabrena, 2018-12-09:
https://maken.wikiwijs.nl/bestanden/642289/how-the-grinch-stole-christmas%20(2)%20(1).pdf

jabrena, 2018-12-09:
https://examples.javacodegeeks.com/desktop-java/sound/play-midi-audio/
http://area512.htmlplanet.com/xmas/grinchxmas.html

jabrena, 2018-12-10:
http://stevydoos.com

jabrena, 2018-12-10:
https://gist.github.com/shmert/3859200

jabrena, 2018-12-10:
https://gist.github.com/AnimeshShaw/8e6ba43f2cfb75331f40

JakubVanek, 2019-01-14:
Hmm, there are multiple ways how to do this. The way the software is uploaded to the brick is over SSH. We have two options:
* SSH from Java - Gradle plugin bypasses Hidetake's SSH plugin and uses JSch directly ([SSH.groovy](https://github.com/ev3dev-lang-java/gradle-plugin/blob/master/src/main/groovy/com/github/elj/gradle/SSH.groovy), [SFTP.groovy](https://github.com/ev3dev-lang-java/gradle-plugin/blob/master/src/main/groovy/com/github/elj/gradle/SFTP.groovy)).
* Alternatively, we can just use `scp` and `ssh`.

Then to establish an SSH connection, there at least two options:
 * Connect the bricks via WiFi/Ethernet. They can either know each other's address, or they can discover it via Avahi (though I don't know how to do that yet). The connection will already be there, so this is the easier path.
 * Connect the bricks via Bluetooth BNEP. The only way I know now how to do it is via graphical `blueman-manager`, so from my side this is not workable yet. However there might be a way how to hack through this via NetworkManager and bluetoothctl. When we have a proper Bluetooth support, we can add a similar demo for it.

There seems to be some a library for discovering via DNS-SD (=mDNS=Avahi) - https://github.com/jmdns/jmdns.
Related ev3dev issue: https://github.com/ev3dev/ev3dev/issues/852

Regards,

Jakub