ev3dev-lang-java/ev3dev-lang-java #620: Develop a library to build User Interfaces for EV3 Display

jabrena, 2018-12-09:
Some applications require a User Interface to interact with the user.

JakubVanek, 2018-12-09:
We could base it on my old library and configuration tool:
https://gitlab.com/robovanek/robo2015-ss/tree/master/program/microgui
https://gitlab.com/robovanek/robo2015-ss/tree/master/program/cfgtool

JakubVanek, 2018-12-09:
I recommend trying it out on real leJOS EV3. However, it works also on the desktop via Swing.

JakubVanek, 2018-12-09:
It may need some reachitecting, I remember regretting some design decisions later (like the way that user has to write the UI screens).

jabrena, 2018-12-09:
Well, the idea is to develop some kind of Util classes to build and interface combining Buttons and the display. It is an incubator project in this stage.

jabrena, 2019-08-18:
Rejected by LEAN

closed 2019-08-18 by jabrena