ev3dev-lang-java/ev3dev-lang-java #625: Add support for OpenCV on Debian Stretch for EV3

jabrena, 2018-12-11:
sudo apt-get install ./libopenmpt0_0.2.10933beta36.1-1_armel.deb

jabrena, 2018-12-11:
```
robot@ev3dev:~$ sudo apt-get update
Get:1 http://security.debian.org stretch/updates InRelease [94.3 kB]
Get:3 http://archive.ev3dev.org/debian stretch InRelease [8431 B]                                                    
Ign:2 http://cdn-fastly.deb.debian.org/debian stretch InRelease
Get:4 http://cdn-fastly.deb.debian.org/debian stretch Release [118 kB]
Get:5 http://cdn-fastly.deb.debian.org/debian stretch Release.gpg [2434 B]
Get:6 http://security.debian.org stretch/updates/main armel Packages [443 kB]                                                                                            
Get:7 http://archive.ev3dev.org/debian stretch/main armel Packages [70.7 kB]                                                                                             
Get:8 http://cdn-fastly.deb.debian.org/debian stretch/non-free armel Packages [51.2 kB]                                                                                  
Get:9 http://cdn-fastly.deb.debian.org/debian stretch/main armel Packages [6889 kB]                                                                                      
Get:10 http://cdn-fastly.deb.debian.org/debian stretch/contrib armel Packages [39.8 kB]                                                                                  
Fetched 7716 kB in 55s (140 kB/s)                                                                                                                                        
Reading package lists... Done
robot@ev3dev:~$ sudo apt-get install libopencv2.4-java
Reading package lists... Done
Building dependency tree       
Reading state information... Done
The following additional packages will be installed:
  libaacs0 libavcodec57 libavformat57 libavutil55 libbdplus0 libbluray1 libchromaprint1 libdc1394-22 libdrm-freedreno1 libdrm-nouveau2 libdrm-radeon1 libgail-common
  libgail18 libgl1-mesa-dri libgl1-mesa-glx libglapi-mesa libgme0 libgsm1 libgtk2.0-0 libgtk2.0-bin libgtk2.0-common libilmbase12 libmp3lame0 libmpg123-0
  libopencv-calib3d2.4v5 libopencv-contrib2.4v5 libopencv-core2.4v5 libopencv-features2d2.4v5 libopencv-flann2.4v5 libopencv-highgui2.4-deb0 libopencv-imgproc2.4v5
  libopencv-ml2.4v5 libopencv-objdetect2.4v5 libopencv-photo2.4v5 libopencv-video2.4v5 libopencv2.4-jni libopenexr22 libopenjp2-7 libopenmpt0 libraw1394-11 libsensors4
  libshine3 libsnappy1v5 libsoxr0 libspeex1 libssh-gcrypt-4 libswresample2 libswscale4 libtheora0 libtwolame0 libtxc-dxtn-s2tc libv4l-0 libv4lconvert0 libva-drm1
  libva-x11-1 libva1 libvdpau-va-gl1 libvdpau1 libvorbisfile3 libvpx4 libwavpack1 libx264-148 libx265-95 libxcb-glx0 libxvidcore4 libxxf86vm1 libzvbi-common libzvbi0
  mesa-va-drivers mesa-vdpau-drivers va-driver-all vdpau-driver-all
Suggested packages:
  libbluray-bdj gvfs libraw1394-doc lm-sensors speex
The following NEW packages will be installed:
  libaacs0 libavcodec57 libavformat57 libavutil55 libbdplus0 libbluray1 libchromaprint1 libdc1394-22 libdrm-freedreno1 libdrm-nouveau2 libdrm-radeon1 libgail-common
  libgail18 libgl1-mesa-dri libgl1-mesa-glx libglapi-mesa libgme0 libgsm1 libgtk2.0-0 libgtk2.0-bin libgtk2.0-common libilmbase12 libmp3lame0 libmpg123-0
  libopencv-calib3d2.4v5 libopencv-contrib2.4v5 libopencv-core2.4v5 libopencv-features2d2.4v5 libopencv-flann2.4v5 libopencv-highgui2.4-deb0 libopencv-imgproc2.4v5
  libopencv-ml2.4v5 libopencv-objdetect2.4v5 libopencv-photo2.4v5 libopencv-video2.4v5 libopencv2.4-java libopencv2.4-jni libopenexr22 libopenjp2-7 libopenmpt0
  libraw1394-11 libsensors4 libshine3 libsnappy1v5 libsoxr0 libspeex1 libssh-gcrypt-4 libswresample2 libswscale4 libtheora0 libtwolame0 libtxc-dxtn-s2tc libv4l-0
  libv4lconvert0 libva-drm1 libva-x11-1 libva1 libvdpau-va-gl1 libvdpau1 libvorbisfile3 libvpx4 libwavpack1 libx264-148 libx265-95 libxcb-glx0 libxvidcore4 libxxf86vm1
  libzvbi-common libzvbi0 mesa-va-drivers mesa-vdpau-drivers va-driver-all vdpau-driver-all
0 upgraded, 73 newly installed, 0 to remove and 7 not upgraded.
Need to get 24.0 MB of archives.
After this operation, 159 MB of additional disk space will be used.
Do you want to continue? [Y/n] y
Get:1 http://cdn-fastly.deb.debian.org/debian stretch/main armel libaacs0 armel 0.8.1-2 [44.7 kB]
Get:2 http://cdn-fastly.deb.debian.org/debian stretch/main armel libva1 armel 1.7.3-2 [46.7 kB]
Get:3 http://cdn-fastly.deb.debian.org/debian stretch/main armel libva-drm1 armel 1.7.3-2 [14.4 kB]
Get:4 http://cdn-fastly.deb.debian.org/debian stretch/main armel libva-x11-1 armel 1.7.3-2 [18.0 kB]
Get:5 http://cdn-fastly.deb.debian.org/debian stretch/main armel libvdpau1 armel 1.1.1-6 [38.9 kB]
Get:6 http://cdn-fastly.deb.debian.org/debian stretch/main armel libavutil55 armel 7:3.2.12-1~deb9u1 [239 kB]
Get:7 http://cdn-fastly.deb.debian.org/debian stretch/main armel libgsm1 armel 1.0.13-4+b2 [23.0 kB]
Get:8 http://cdn-fastly.deb.debian.org/debian stretch/main armel libmp3lame0 armel 3.99.5+repack1-9+b2 [342 kB]
Get:9 http://cdn-fastly.deb.debian.org/debian stretch/main armel libopenjp2-7 armel 2.1.2-1.1+deb9u2 [110 kB]
Get:10 http://cdn-fastly.deb.debian.org/debian stretch/main armel libshine3 armel 3.1.0-5 [25.3 kB]
Get:11 http://cdn-fastly.deb.debian.org/debian stretch/main armel libsnappy1v5 armel 1.1.3-3 [50.2 kB]
Get:12 http://cdn-fastly.deb.debian.org/debian stretch/main armel libspeex1 armel 1.2~rc1.2-1+b2 [54.8 kB]
Get:13 http://cdn-fastly.deb.debian.org/debian stretch/main armel libsoxr0 armel 0.1.2-2 [61.8 kB]
[... cut ...]
Setting up libavformat57:armel (7:3.2.12-1~deb9u1) ...
Setting up libopencv-highgui2.4-deb0:armel (2.4.9.1+dfsg1-2) ...
Setting up libopencv-objdetect2.4v5:armel (2.4.9.1+dfsg1-2) ...
Setting up libopencv-features2d2.4v5:armel (2.4.9.1+dfsg1-2) ...
Setting up libopencv-calib3d2.4v5:armel (2.4.9.1+dfsg1-2) ...
Setting up libopencv-contrib2.4v5:armel (2.4.9.1+dfsg1-2) ...
Setting up libopencv2.4-jni (2.4.9.1+dfsg1-2) ...
Setting up libopencv2.4-java (2.4.9.1+dfsg1-2) ...
Processing triggers for libc-bin (2.24-11+deb9u3) ...
```

jabrena, 2018-12-11:
```
java  libopenmpt0_0.2.10933beta36.1-1_armel.deb
robot@ev3dev:~$ sudo apt-get install ./libopenmpt0_0.2.10933beta36.1-1_armel.deb 
Reading package lists... Done
Building dependency tree       
Reading state information... Done
Note, selecting 'libopenmpt0' instead of './libopenmpt0_0.2.10933beta36.1-1_armel.deb'
The following packages will be upgraded:
  libopenmpt0
1 upgraded, 0 newly installed, 0 to remove and 7 not upgraded.
Need to get 0 B/432 kB of archives.
After this operation, 11.3 kB of additional disk space will be used.
Get:1 /home/robot/libopenmpt0_0.2.10933beta36.1-1_armel.deb libopenmpt0 armel 0.2.10933~beta36.1-1 [432 kB]
(Reading database ... 35331 files and directories currently installed.)
Preparing to unpack .../libopenmpt0_0.2.10933beta36.1-1_armel.deb ...
Unpacking libopenmpt0:armel (0.2.10933~beta36.1-1) over (0.2.7386~beta20.3-3+deb9u3) ...
Setting up libopenmpt0:armel (0.2.10933~beta36.1-1) ...
Processing triggers for libc-bin (2.24-11+deb9u3) ...
```

JakubVanek, 2018-12-11:
I hope that we won't upload the package to ev3dev repos. I tried it, but luckily I made an error in preparing the package (it didn't upload) and so I could think it through again. Ideally, this bug will be fixed in Debian.

JakubVanek, 2018-12-12:
Hi Juan,

Yaay! Now https://github.com/ev3dev/ev3dev/issues/1204 is resolved - could you please try running:
```bash
sudo apt purge libopenmpt0 # to remove the version installed via dpkg
sudo apt update
sudo apt install --no-install-recommends libopencv2.4-java
```
It should work out of the box now - it works at least in my `armel` chroot.

Cheers,

Jakub

jabrena, 2018-12-27:
Hi @JakubVanek, 

maybe in the installer, we could update the process to add upgrade the debian package that we fixed:

```
sudo apt purge libopenmpt0 # to remove the version installed via dpkg
sudo apt update
sudo apt install --no-install-recommends libopencv2.4-java
```

Currently, we only install `libopencv2.4-java`

```
ev3dev#10| --- Running command: "echo -e "********" | sudo -S /home/robot/java/installer.sh opencv" ---
ev3dev#11|
ev3dev#11|##############################
ev3dev#11|# EV3Dev-lang-java Installer #
ev3dev#11|##############################
ev3dev#11|
ev3dev#11|Platform detected: ev3
ev3dev#11|
ev3dev#11|[Thu, 27 Dec 2018 15:49:07 +0000] Installing OpenCV
ev3dev#11|Reading package lists...
ev3dev#11|Building dependency tree...
ev3dev#11|Reading state information...
ev3dev#11|libopencv2.4-java is already the newest version (2.4.9.1+dfsg1-2).
ev3dev#11|0 upgraded, 0 newly installed, 0 to remove and 7 not upgraded.
ev3dev#11|[sudo] password for robot: 

BUILD SUCCESSFUL in 27s
1 actionable task: 1 executed
16:49:29: Task execution finished 'installOpenCV'.
```

What do you think?

jabrena, 2018-12-27:
Hi @JakubVanek, why it is not necessary when we use Opencv, to send the path about native libraries:

```
ev3dev#28| --- Running command: "time brickrun -- java -Xms64m -Xmx64m -XX:+UseSerialGC -Xshare:on -noverify -cp "/home/robot/java/libraries/slf4j-simple-1.7.25.jar:/home/robot/java/libraries/ev3dev-lang-java-2.5.3.jar:/home/robot/java/libraries/slf4j-api-1.7.25.jar:/home/robot/java/libraries/lejos-commons-0.7.3.jar:/home/robot/java/libraries/jna-4.5.2.jar:/usr/share/java/opencv.jar:/home/robot/java/programs/template_project_gradle-2.5.3.jar" example.programming.HelloWorldCV" ---
``` 

Originally, we always set the path manually:

```
ev3dev#28| --- Running command: "time brickrun -- java -Xms64m -Xmx64m -XX:+UseSerialGC -Xshare:on -noverify -cp "/home/robot/java/libraries/slf4j-simple-1.7.25.jar:/home/robot/java/libraries/ev3dev-lang-java-2.5.3.jar:/home/robot/java/libraries/slf4j-api-1.7.25.jar:/home/robot/java/libraries/lejos-commons-0.7.3.jar:/home/robot/java/libraries/jna-4.5.2.jar:/usr/share/java/opencv.jar:/home/robot/java/programs/template_project_gradle-2.5.3.jar" example.programming.HelloWorldCV" ---
```

https://github.com/ev3dev-lang-java/ev3dev-lang-java/issues/33

JakubVanek, 2018-12-27:
I'd rather choose another approach -- explicitly name the `libopenmpt0` package on the initial commandline, which causes its automatic update.

It is not necessary to specify java library path because I incorporated the JNI patch from Debian, so the libraries are found automatically in the system directories.

jabrena, 2018-12-27:
> It is not necessary to specify java library path because I incorporated the JNI patch from Debian, so the libraries are found automatically in the system directories.

How did you do it? In the Debian package process?

> I'd rather choose another approach -- explicitly name the libopenmpt0 package on the initial commandline, which causes its automatic update.

What is the new set of commands to fix it with that approach?

jabrena, 2018-12-27:
In relation to JNI Path, using the new flag, is it compatible with the original flag: `-Djava.library.path=/usr/lib/jni/` why the new flag?

JakubVanek, 2018-12-27:
Hmmm, the missing APT cache will be a problem only on EV3, because only there is the update phase skipped. On other platforms it is performed before installing the libraries. (see [here](https://github.com/ev3dev-lang-java/template-project-gradle/blob/70ddd8f88962e474aeb1766ab2e01e1c5339df05/README.md#ev3)).

We do not support package uninstallation yet. Once the user installs the library, it stays there.

JakubVanek, 2018-12-27:
Because the new flag is compile time only. You should not specify `java.library.path` nor `--with-jni-libpath` flag at run time now. It is done in advance.

jabrena, 2018-12-27:
No, I am saying in a manual way. What is the command to uninstall that packages?

JakubVanek, 2018-12-27:
```
sudo apt-get autoremove --purge libopencv2.4-java libopenmpt0
```

JakubVanek, 2018-12-27:
Hmmm, I was wrong:

README.md for both EV3 and other platforms:
> If you want to use RXTX or OpenCV, you can install them this way:
> ```bash
> ./gradlew updateAPT installOpenCV installRXTX
> ```

So if the user follows the instructions, the libopenmpt0 package will be automatically updated when installing OpenCV.

jabrena, 2018-12-27:
> Because the new flag is compile time only. You should not specify java.library.path nor --with-jni-libpath flag at run time now. It is done in advance.

What happen if the user add a native library in another path?
How to configure it?
Is the flag `--with-jni-libpath` part of JDK flags?

JakubVanek, 2018-12-27:
> > Because the new flag is compile time only. You should not specify java.library.path nor --with-jni-libpath flag at run time now. It is done in advance.
> 
> What happen if the user add a native library in another path?

Then they will have to list the whole library path on command line anew.

> How to configure it?

At runtime, they will have to use `-Djava.library.path=...`. At compile time, we can add it to the build scripts - we can define a directory `/home/robot/java/native` specifically for native libraries.

> Is the flag `--with-jni-libpath` part of JDK flags?

Yes, it is part of OpenJDK 12+ configure process: https://hg.openjdk.java.net/jdk/jdk12/raw-file/de9fd809bb47/doc/building.html

> On Linux, BSD and AIX, it is possible to override where Java by default searches for runtime/JNI libraries. This can be useful in situations where there is a special shared directory for system JNI libraries. This setting can in turn be overriden at runtime by setting the java.library.path property.
>
>* `--with-jni-libpath=<path>` - Use the specified path as a default when searching for runtime libraries.

JakubVanek, 2018-12-27:
> Then they will have to list the whole library path on command line anew.

At least I think so. It think that the default path is completely overwritten with the one specified on command line.

jabrena, 2018-12-27:
We should test the behavior.

We provide some support for some libraries like OpenCV or rxtx but the user could execute this kind of example without any direct technical support:
https://www3.ntu.edu.sg/home/ehchua/programming/java/JavaNativeInterface.html

JakubVanek, 2018-12-27:
Solution for `/home/robot/java/natives` is here:
https://github.com/ev3dev-lang-java/installer/pull/43
https://github.com/ev3dev-lang-java/openjdk-ev3/pull/32

JakubVanek, 2018-12-27:
I closed this issue accidentally - I clicked the first button automatically before realizing that there is one more.
[image]

jabrena, 2018-12-27:
Don't worryyyy :p

jabrena, 2018-12-27:
@JakubVanek Excellent Research job with OpenJDK 12 the flag analysis. After this merge request in Installer, we could close it.

Ok?

jabrena, 2018-12-28:
Fixed support

closed 2018-12-28 by jabrena