ev3dev-lang-java/ev3dev-lang-java #629: Transition to in-JAR classpath

JakubVanek, 2018-12-13:
This should simplify the java commandline code slightly. It will also make it easier when I add the opencv/rxtx classpath code.

JakubVanek, 2018-12-15:
Hmm, we can go both ways, it could be switched through a bool flag `useEmbeddedPaths`. There is nothing wrong with including the classpath in jar manifest, now the only thing is what to use in the deployer/runner.

jabrena, 2018-12-15:
Using cp, is more readable for the user. See everything. With jar, the user dont see the jar paths and they dont learn the folders, etc...

JakubVanek, 2018-12-15:
But yes, that is the intent of hiding it behind `-jar` - to hide unnecessary cruft from the java command line. One can the orient better in the resulting command line.

JakubVanek, 2018-12-15:
Cmdline with `-cp`:
```
ev3dev#7| --- Running command: "time  java -Xms64m -Xmx64m -XX:+UseSerialGC -Xshare:on -noverify   -cp "/home/robot/java/libraries/slf4j-simple-1.7.25.jar:/home/robot/java/libraries/ev3dev-lang-java-2.5.3.jar:/home/robot/java/libraries/slf4j-api-1.7.25.jar:/home/robot/java/libraries/lejos-commons-0.7.3.jar:/home/robot/java/libraries/jna-4.5.2.jar:file:///usr/share/java/opencv.jar:file:///usr/share/java/RXTXcomm.jar:/home/robot/java/programs/template_project_gradle-2.5.3.jar" example.programming.HelloWorld" ---
```
Cmdline with `-jar`:
```
ev3dev#16| --- Running command: "time  java -Xms64m -Xmx64m -XX:+UseSerialGC -Xshare:on -noverify   -jar /home/robot/java/programs/template_project_gradle-2.5.3.jar" ---
```

JakubVanek, 2018-12-15:
Commit https://github.com/ev3dev-lang-java/template-project-gradle/commit/c63b75beb79362524a839da60f9daa5ae11e810d implements this switching. I have enabled `-jar` by default, but we can disable it instead.