ev3dev-lang-java/ev3dev-lang-java #632: Package template_project_gradle to a plugin

JakubVanek, 2018-12-15:
Maybe most of the build machinery should be hidden in a plugin. At least `ev3dev.gradle` would be probably duplicated everywhere almost unchanged.
https://docs.gradle.org/current/userguide/custom_plugins.html

jabrena, 2018-12-15:
With a plugin, the project is more flexible and it is possible to use with maven or gradle kotlin

JakubVanek, 2018-12-16:
Alpha version implemented:
https://github.com/ev3dev-lang-java/template-project-gradle/tree/plugined
https://github.com/ev3dev-lang-java/gradle-plugin

It seems to work pretty well so far.

I will need to clean up the gradle plugin a little. Also, a publishing infrastructure needs to be set up.

jabrena, 2018-12-29:
Hi @JakubVanek, yesterday, we had a good conversation about this topic. 
I will post the summary:

Currently, Template project tries to help the newbies in Java with EV3 using a project example ready to use. Using `Gradle` or another build system like `Maven`, we can execute the software life cycle by command line, but in any case, it is necessary to edit the file `config.gradle` and normally, the user click in the list of tasks that it is possible to use in the IDE. If we evolve current implementation using a Gradle plugin, it is Ok, although we increase the complexity in the way to understand the whole Gradle project, but at the end, we are hiding details for the final user, so in perspective the benefit is the same.

So, using a Gradle plugin or using current implementation based on Gradle script files 
https://github.com/ev3dev-lang-java/template-project-gradle/commit/529f53436c9d4978df455f693fb3151451cdbf7e 
we increase project complexity but we win in the way to maintain multiple projects, but I don't have a negative feeling.

**If we add a Gradle Plugin, we need to create the same approach for a Template Project based on Maven, in order to close the gap. Maven in general is more used than Gradle in a general way.**

At the end, we need a `Intellij` plugin simply a bit more the tasks to deploy a jar in a brick and some external activities.
https://github.com/ev3dev-lang-java/ev3dev-lang-java/issues/582

But in terms of priorities, it is more urgent to have Template project based on Maven.
https://github.com/ev3dev-lang-java/ev3dev-lang-java/issues/636

So, Ok with the evolution of `Gradle` project but we will have to create a **Maven** project and later the plugin.

Juan Antonio

jabrena, 2019-01-06:
Reviewing project:
https://github.com/ev3dev-lang-java/gradle-plugin

jabrena, 2019-01-06:
Initial Travis execution:
https://travis-ci.org/ev3dev-lang-java/gradle-plugin

jabrena, 2019-01-06:
Hi @JakubVanek,
 
Why we have gradle 5 in the wrapper?
https://github.com/ev3dev-lang-java/gradle-plugin/blob/master/gradle/wrapper/gradle-wrapper.properties
In the rest of the project with 4.7 version

JakubVanek, 2019-01-06:
Hi, I did so to have the upgrade path to Gradle 5 open. It should be possible to just downgrade it to 4.7.

closed 2019-01-12 by jabrena