ev3dev-lang-java/ev3dev-lang-java #633: Template project Gradle failure Cannot add task 'wrapper'

jartu, 2018-12-26:


jartu, 2018-12-26:
It's actually enough to just rename the task from "wrapper" to anything else, e.g. "wrapper2" and the build succeeds.

JakubVanek, 2018-12-26:
I can reproduce this issue via console on Linux with Gradle 5.0. Renaming the task is probably the way to go.

jabrena, 2018-12-26:
Merged.

@jartu Many thanks to report it. 
It is weird a bit. Thanks

Merry Christmas to everyone 

Cheers

jartu, 2018-12-26:
Merry Christmas to everyone and a Happy New Year!
J.

jabrena, 2018-12-26:
Nice to know you again.
Cheers

jabrena, 2018-12-27:
Hi mates,

Maybe we could remove the Gradle task `wrapper/2`, at the moment, everyone are using the default behaviour:
https://docs.gradle.org/current/userguide/gradle_wrapper.html

and if you observe, if we rename the task as `wrapper2`, we are not customising the default behaviour:

``` groovy
wrapper {
    distributionType = Wrapper.DistributionType.ALL
}
```

When the user, import the project, everyone have 2 alternatives:

- Use local gradle version
- Use gradle wrapper

currently, with wrapper2 task, we are not adding no value to the template project.
It is better to remove it.

@jartu, @JakubVanek Ok?

Cheers

Juan Antonio

jabrena, 2018-12-27:
Tomorrow,  I will create a `PR` to remove that not used `Gradle` task. 
Less lines to maintain :)

jartu, 2018-12-27:
Sure, makes sense. Let's remove it.

jabrena, 2018-12-27:
Oki doki.

jabrena, 2018-12-28:
Added PR:
https://github.com/ev3dev-lang-java/template-project-gradle/pull/29/files

jabrena, 2018-12-28:
Fixed

jabrena, 2018-12-28:
@jartu, when are you going to return to the party? do you have some time slot to play? any area of interest?

closed 2018-12-28 by jabrena

jartu, 2018-12-30:
Hi @jabrena,

I had just a little bit of time over the holidays to install the ev3dev-lang-java on my daughter's computer (hoping she'd become interested in programming over time) and I came across this issue while doing so.

I have to be honest, I am stretched pretty thin with my job and my other extracurricular activities. However, this project is very close to my heart and I'd love to keep contributing, however little I can. 

Would you have some relatively small and non-critical programming task for me?

Best, J.

jabrena, 2018-12-30:
Hi @jartu, I like so much your personal message, honestly :)

Any new hand is welcome, super welcome. For example @JakubVanek, this year did an amazing job in multiple areas, more related to Software Engineering than Robotics but in general, his contribution was amazing. Now, the whole architecture is better:

**2018 Architecture:**

- 1. OS: Debian
- 2. EV3 Runtime: EV3Dev
- 3. JDK: OpenJDK 12
- 4. ELJ: Partial LeJOS port + new developments

**2019 Architecture:**

- 1,2,3 Preinstalled. The user in 10 minutes has a brick ready to deploy
- 4. ELJ: Partial LeJOS port + new developments

Maybe the first task, is a review of EV3Key.java
https://github.com/ev3dev-lang-java/ev3dev-lang-java/blob/master/src/main/java/ev3dev/sensors/EV3Key.java

_**We have a Compilation warning. Can you review the class to remove the warning?**_

I think that next year, we should put the focus on **Robotics** features. 
The base code is stable.

Ideas to discuss:

- Remote library EV3-EV3 (RMI)
- Review and test all cases for Navigation library (Pending to put the whole library in the stable section)
- ROS Support (This is a real big milestone for us.) for ROS2, @esteve will help us in the future Q3 2019
- Better Arduino support
- Review synchronised motor features (EV3Dev research: https://github.com/ev3dev/ev3dev/issues/853)
- More OpenCV real examples.

In terms off Software engineering, some ideas ultra important are:

- Jar loader (New iteration to improve the performance) https://github.com/ev3dev-lang-java/ev3dev-lang-java/issues/600
- Maven template + Pluggin @JakubVanek 
- Intellij Plugin (Easy way to deploy a program, current way is more customised): @miguelcordovam, @robl0377  

**Jar Loader could be a super hit.**

In my case, I will develop a **Cloud compiler for EV3**. I suppose that for the next **_LEGO Mindstorms_** generation, we could reuse many parts of current job if `EV3Dev` respect the interfaces (I suppose). Any internal change requires 4 months, with more hands, we could deliver before.

We should put emphasis to be used for **WRO** and **FLL** next year. Officially, next year, anyone could use this project, so it is a big opportunity for Java ecosystem. So, we will have to elaborate examples and documentation.

Merry christmas & Happy new year.

**Note:** This project is Open Source, we enjoy and play with technology. In my case, I learn many new stuff using this domain to learn about JVM technologies and of course, I love robotics. :) 

Juan Antonio

jabrena, 2018-12-30:
@jartu,

Tell me if you like the challenge with the warning as an introduction.

Cheers

jabrena, 2018-12-30:
We need to have a library to interact with ROS and make SLAM:

[image]

Later, we will try to solve another classic problem with manipulators. LeJOS didn't develop that support.

jartu, 2019-01-03:
Hi @jabrena,

thanks for the detailed roadmap description. 

Sure, I'll look at the warning first.

Thanks
J.

jabrena, 2019-01-03:
Perfect!!! I will create the issue later.

Cheers