ev3dev-lang-java/ev3dev-lang-java #634: Test and merge plugin-based template project

JakubVanek, 2018-12-28:
Hi,

I have focused on the [gradle plugin](https://github.com/ev3dev-lang-java/gradle-plugin) today. I have rewritten it to Groovy and it seems to work well. It can now be tested on the [`plugined`](https://github.com/ev3dev-lang-java/template-project-gradle/tree/plugined) branch of the template project ([PR here](https://github.com/ev3dev-lang-java/template-project-gradle/pull/28)).

Directly related to https://github.com/ev3dev-lang-java/ev3dev-lang-java/issues/632.
Allows for easier https://github.com/ev3dev-lang-java/ev3dev-lang-java/issues/606.
Fixes https://github.com/ev3dev-lang-java/ev3dev-lang-java/issues/576 because now newest Gradle can be used.

Cheers,

Jakub

JakubVanek, 2018-12-28:
> Fixes #576 because now newest Gradle can be used.

At least on template project Gradle 5.0 works, although I have seen once or twice a freeze after a remote command completed. I will have to catch that and identify what is wrong. The crash might be caused by a different dependency of the main project.

JakubVanek, 2018-12-28:
Hmm, this is the issue that was problematic due to too new Gradle version:
https://github.com/ev3dev-lang-java/ev3dev-lang-java/issues/529

jabrena, 2018-12-28:
It is better, if we put the focus on the plugin only @JakubVanek  and maintain in the current way.
Maybe other users don't have the latest Gradle version. Current implementation is stable.

JakubVanek, 2018-12-28:
There shouldn't be any problems with using old Gradle versions.

I think that not everyone will use IntelliJ. There are some people using Eclipse or a different IDE. I'm often using Geany and I build the project in terminal in a separate window. These use cases are handled even now, but this unifies the build code.

jabrena, 2019-01-12:
Duplicated issue: https://github.com/ev3dev-lang-java/ev3dev-lang-java/issues/632

closed 2019-01-12 by jabrena