ev3dev-lang-java/ev3dev-lang-java #639: Remove compilation warning in class EV3Key.java

jabrena, 2019-01-03:
Remove the compilation warning in the class EV3Key.java

jabrena, 2019-01-03:
@jartu, issue created. :)

jartu, 2019-01-04:
Thanks for creating the issue.

Are you sure the warnings are in the EV3Key.java ? The only problematic thing I see there (although it doesn't show during the "gradle :compileJava" task) is the reference "{@link Button#ALL}" from the javadoc of the addKeyListener(KeyListener) method.

However, I see quite a few warnings from the "gradle build" task from the EV3KeyTest.java .

Do you see some other warning in the EV3Key.java ?

jabrena, 2019-01-04:
Sorry, you are right :)

```
> Task :compileTestJava
/home/jabrena/IdeaProjects/ev3dev-lang-java/src/test/java/ev3dev/sensors/EV3KeyTest.java:46: warning: [rawtypes] found raw type: Pair
                final Pair cmp = (Pair)obj;
                      ^
  missing type arguments for generic class Pair<L,R>
  where L,R are type-variables:
    L extends Object declared in class Pair
    R extends Object declared in class Pair
/home/jabrena/IdeaProjects/ev3dev-lang-java/src/test/java/ev3dev/sensors/EV3KeyTest.java:214: warning: [unchecked] unchecked generic array creation for varargs parameter of type Pair<TestKeyEventWaiter,Boolean>[]
        assertWaiters(toList(
                            ^
/home/jabrena/IdeaProjects/ev3dev-lang-java/src/test/java/ev3dev/sensors/EV3KeyTest.java:222: warning: [unchecked] unchecked generic array creation for varargs parameter of type Pair<TestKeyEventWaiter,Boolean>[]
        assertWaiters(toList(
                            ^
/home/jabrena/IdeaProjects/ev3dev-lang-java/src/test/java/ev3dev/sensors/EV3KeyTest.java:230: warning: [unchecked] unchecked generic array creation for varargs parameter of type Pair<TestKeyEventWaiter,Boolean>[]
        assertWaiters(toList(   // no change
                            ^
/home/jabrena/IdeaProjects/ev3dev-lang-java/src/test/java/ev3dev/sensors/EV3KeyTest.java:238: warning: [unchecked] unchecked generic array creation for varargs parameter of type Pair<TestKeyEventWaiter,Boolean>[]
        assertWaiters(toList(   // no change
                            ^
/home/jabrena/IdeaProjects/ev3dev-lang-java/src/test/java/ev3dev/sensors/EV3KeyTest.java:247: warning: [unchecked] unchecked generic array creation for varargs parameter of type Pair<EV3Key,KeyEvent>[]
        Assert.assertEquals(toList(
                                  ^
/home/jabrena/IdeaProjects/ev3dev-lang-java/src/test/java/ev3dev/sensors/EV3KeyTest.java:250: warning: [unchecked] unchecked generic array creation for varargs parameter of type Pair<EV3Key,KeyEvent>[]
        Assert.assertEquals(toList(
                                  ^
/home/jabrena/IdeaProjects/ev3dev-lang-java/src/test/java/ev3dev/sensors/EV3KeyTest.java:255: warning: [unchecked] unchecked generic array creation for varargs parameter of type Pair<EV3Key,KeyEvent>[]
        Assert.assertEquals(toList(
                                  ^
/home/jabrena/IdeaProjects/ev3dev-lang-java/src/test/java/ev3dev/sensors/EV3KeyTest.java:263: warning: [unchecked] Possible heap pollution from parameterized vararg type Pair<L,R>
    private static <L, R> List<Pair<L, R>> toList(final Pair<L, R>... pairs) {
                                                                      ^
  where L,R are type-variables:
    L extends Object declared in method <L,R>toList(Pair<L,R>...)
    R extends Object declared in method <L,R>toList(Pair<L,R>...)
9 warnings
```

The warning appear in the Tests. Can you fix it?

jartu, 2019-01-04:
Pull request created: https://github.com/ev3dev-lang-java/ev3dev-lang-java/pull/641

jabrena, 2019-01-05:
Fixed

closed 2019-01-05 by jabrena

jabrena, 2019-01-05:
Now, what do you prefer?:

**Infrastructure:**

- D-Bus: https://github.com/ev3dev-lang-java/ev3dev-lang-java/issues/571
- Bluetooth: https://github.com/ev3dev-lang-java/ev3dev-lang-java/issues/446

**Software engineering:**

- Run Jar in runtime [POC]: https://github.com/ev3dev-lang-java/ev3dev-lang-java/issues/600

**Robotics:**

- Local Navigation: https://github.com/ev3dev-lang-java/ev3dev-lang-java/issues/630

Juan Antonio

jartu, 2019-01-06:
I think I'll go with the #600. More questions in there.
J.