ev3dev-lang-java/ev3dev-lang-java #648: '/sys/class/tacho-motor/motor2/command' without write permission

viniciusccarvalho, 2019-02-20:
Hi there, I found #554 but it seems that I still see this issue when using an NXT regulated motor

I'm not sure if it's the case of using an NXT regulated motor, but the same error seems to happen with me.

Java version

```
openjdk version "11.0.2" 2019-01-15
OpenJDK Runtime Environment (build 11.0.2+9-ev3)
OpenJDK Client VM (build 11.0.2+9-ev3, mixed mode, sharing)
```

Kernel:
```
Linux ev3dev 4.14.96-ev3dev-2.3.2-ev3 #1 PREEMPT Sun Jan 27 21:27:35 CST 2019 armv5tejl GNU/Linux```

ev3dev-lang-java: 2.5.3

jabrena, 2019-02-21:
Hi @viniciusccarvalho, 

this weekend, we will review the case together. 

Welcome to the project.

Cheers

Juan Antonio

jabrena, 2019-02-25:
I am going to close this issue. It is duplicated.

Juan Antonio

closed 2019-02-25 by jabrena

JakubVanek, 2019-02-26:
Hmmm. I think that #651 and #649 are related to this. NXT motors use the same ev3dev drivers as EV3 motors use.