ev3dev-lang-java/ev3dev-lang-java #649: Unregulated motors are implemented incorrectly

JakubVanek, 2019-02-23:
Hi,

I am exploring how devices are handled in ev3dev and in this library and I noticed that unregulated motors should be handled differently. Currently, the port mode is switched to `dc-motor`. IMHO this is wrong: it is still an EV3 motor. By switching the port to the dumb mode you lose the ability to query the encoder position. Manual power-setting can be achieved with standard tacho-motor by running command `run-direct` and writing the power to `duty_cycle_sp`.

Regards,

Jakub

JakubVanek, 2019-02-23:
Hmm, it seems that it is more complicated. This is the way I see it:

Original leJOS UnregulatedMotor can read out tacho values and is intended for direct PWM control. For example: I once wanted to create a "reading head" for a color sensor and wanted to control it directly [[link][1]]. I used UnregulatedMotor to do this.

ev3dev-lang-java changes meaning of these APIs. UnregulatedMotor is instead intended for non-tacho motors and unregulated motor control is achieved through BaseRegulatedMotor. However, it also changes a meaning of this function, which is newly used for the switch: `suspendRegulation()`. The original behavior is even stated in the documentation for the new function:
https://github.com/ev3dev-lang-java/ev3dev-lang-java/blob/9d02a050b3baaf37a2e8ae7cce7a1c9d1afa2589/src/main/java/ev3dev/actuators/lego/motors/BaseRegulatedMotor.java#L100-L106

Looking into leJOS sources, `suspendRegulation()` is equal to calling `flt()`.
However, in this library, no automatic reenabling is done, rather the motor switches to what effectively is the behavior of UnregulatedMotor on leJOS.

[1]: https://gitlab.com/robovanek/robo2016-zs/blob/master/programs/java/alternative/src/tux/alter/mechanics/PID2Mover.java

JakubVanek, 2019-02-23:
The problem I see with it is that UnregulatedMotor cannot read the tacho sensor, because the kernel doesn't try to read it, because it makes no assumptions about what is on the port, because it was configured to see the port as a `dc-motor`.

JakubVanek, 2019-02-23:
Then there is [RCXMotor](https://lejos.sourceforge.io/ev3/docs/lejos/hardware/motor/RCXMotor.html) in leJOS. *This* is the equivalent of a `dc-motor` - the sysfs kernel device is even getting named like that: `ev3-ports:outA:rcx-motor`. Indeed, RCX motor has no way of reporting tacho values back, so this way it would be consistent.

JakubVanek, 2019-02-23:
TL;DR: the current UnregulatedMotor should be RCXMotor and instead the BaseRegulatedMotor logic behind `suspendRegulation/regulationFlag` should be in UnregulatedMotor. `suspendRegulation` should essentially call `flt()`.

jabrena, 2019-02-25:
Hi @JakubVanek, 

The support is running fine for DC Motors. Post here one example that it doesn´t have the normal behaviour. Did you tests a DC Motor with your Brick?

Juan Antonio

JakubVanek, 2019-02-26:
Hi @jabrena,

no, my doubt is about the API. The UnregulatedMotor is not supposed to be for DC Motors. RCXMotor is. In particular, leJOS UnregulatedMotor can read tacho count, while UnregulatedMotor in this library cannot.

Regards,

Jakub Vanek

JakubVanek, 2019-02-26:
There is also another reason why I'd like fix the API to be like on leJOS: because then UnregulatedMotors will *not* need to wait for a switch to `dc-motor` mode - they will be handled by the `tacho-motor` mode (+ see note below). RCXMotors will however still have the problem and this can be avoided only by udev, retry mechanism or a delay.

`suspendRegulation()` in leJOS was intended to be like `flt()`. It was not related to UnregulatedMotor switching.

Note: If and after https://github.com/ev3dev-lang-java/ev3dev-lang-java/issues/651#issuecomment-466885443 is implemented:

> I think that reading the port mode before writing it is the way to go. It should not be changed if it is in `auto` or `tacho-motor` mode, only when it is in a different mode. If the mode attribute is not readable, there is also a status attribute which should be.

the permission denied issue will disappear for BaseRegulatedMotor descendants and the fixed UnregulatedMotor.

jabrena, 2019-02-26:
Please, lets me consider if it is a bug or not.

jabrena, 2019-09-01:
Current implementation is working fine for RCXMotor and EV3Motor as Unregulated motors.

closed 2019-09-01 by jabrena