ev3dev-lang-java/ev3dev-lang-java #651: Call to canWrite is redundant on Sysfs

viniciusccarvalho, 2019-02-24:
See #650

JakubVanek, 2019-02-25:
Hi,

I looked into BaseRegulatedMotor constructor and indeed, it switches the port mode. This is causing the mount remapping.

By default, the motor port is in `auto` mode. This means that the kernel identifies the type of the motor by reading the voltage across an ID resistor. However this doesn't work for "dumb" motors which are just electric motors connected to the PWM power lines, like the RCX motors (however see #649). For this reason the port can be switched from the `auto` mode to the `dc-motor` mode. However it can also be switched to the `tacho-motor` mode (EV3 motors). This is what is happening in the BaseRegulatedMotor constructor.

A switch makes the kernel remove the old motor device and then add the new one. If one eliminates the switch, everything should work.

I think that reading the port mode before writing it is the way to go. It should not be changed if it is in `auto` or `tacho-motor` mode, only when it is in a different mode. If the mode attribute is not readable, there is also a status attribute which should be.

Regards,

Jakub

jabrena, 2019-02-25:
Hi @JakubVanek, 

the method was designed and implemented in that way to be compatible with all platforms.

This case was discussed few times ago. It is asynchronous operation and this is the source of the discussion. 

@viniciusccarvalho, I created this issue to solve your problem.
https://github.com/ev3dev-lang-java/ev3dev-lang-java/issues/555

viniciusccarvalho, 2019-02-25:
@jabrena Why is the PR closed? Doesn't make sense to remove the redundant check?

I tested with stretch. The mode switching seems to be the cause, but the check is redundant, so why not accept the PR?

jabrena, 2019-02-25:
Good night @viniciusccarvalho, 

Give me the opportunity to review the whole conversation.

Juan Antonio

JakubVanek, 2019-02-26:
> Hi @JakubVanek,
> 
> the method was designed and implemented in that way to be compatible with all platforms.
> 
> This case was discussed few times ago. It is asynchronous operation and this is the source of the discussion.
> 
> @viniciusccarvalho, I created this issue to solve your problem.
> #555
> 
> Did you test with EV3Dev Stretch? Few months ago, we had the same problem and we increased the fixed time.
> 
> Juan Antonio

Indeed, it is an asynchronous process. However, we can still make it write it to the kernel only when it is not in the mode already. This reduces the impact of the delay and fixes the problem for situations where the mode wasn't switched previously (i.e. when `dc-motor` wasn't used).

viniciusccarvalho, 2019-03-12:
@jabrena Any chances this will be merged?

jabrena, 2019-03-13:
Yes, this week, we are going to work in this issue. :)

viniciusccarvalho, 2019-03-13:
Do you plan to accept the PR? It at least solves the issue when the file is readable, and the if is redundant anyways

JakubVanek, 2019-03-14:
Hi,

https://github.com/ev3dev-lang-java/ev3dev-lang-java/pull/654 changes the `writeString()` method and also avoids the explicit permission checks.
https://github.com/ev3dev-lang-java/ev3dev-lang-java/pull/655 implements the write-if-different mechanism for EV3. Other platforms do not support motor autodetection, so there the mode switch is unavoidable the first time.

Regards,

Jakub

jabrena, 2019-03-14:
Hi mates,

I am going to review the pull request from @viniciusccarvalho and the current analysis. Sorry the delay but I was pretty busy in previous weeks. When we have any issue, it is not necessary to change everything because, in that case, we need to do a full regression of the library and it is not the case.

@viniciusccarvalho, did you continue with the problem with **Debian Stretch** and **OpenJDK 11** using the latest snapshot recommended?

Juan Antonio

jabrena, 2019-09-01:
Hi @viniciusccarvalho @JakubVanek,

Does exist an example to raise the issue that you say?

In the PR or in this issue, I didn´t see any example.

Juan Antonio

viniciusccarvalho, 2019-09-01:
I honestly thought this project was dead, I stopped using it and went back to lejos, so I'm not sure I would have an example, all I recall is that the check is unnecessary

jabrena, 2019-09-01:
@viniciusccarvalho, the project goes slow, nothing more. LeJOS is stopped in 2015, this project continue running slowly. We are not **Spring** we don´t have any Sponsorship behind, we do our best in our spare time. :)

If you have one example, we could review the case.

Sorry the delay.

Juan Antonio

viniciusccarvalho, 2019-09-01:
It's not about sponsorship, but accepting PRs and contribution of others as opposed to wait 6 months is a sign of project health. If this is a one man project that's fine, I'm just not interested in using projects that can suffer from the hit by the bus syndrome. I saw a few other PRs were people were keen to help and put forward their best effort, and it went to void with just no acknowledge from the owners. Good luck with the project,not a good way to foster a community but perhaps that was never the intention.

jabrena, 2019-09-01:
I understand that you are not happy, it is a fact. If next week, you find the example, I will test.

Good night 

Juan Antonio

jabrena, 2019-12-02:
Increasing the delay, it was possible to mitigate the issue.

closed 2019-12-02 by jabrena