ev3dev-lang-java/ev3dev-lang-java #656: Deliver OpenJDK 12 for EV3

jabrena, 2019-03-15:
The release of Java 12 is inminent:
http://www.java-countdown.xyz/

so we need to move the build from AdoptOpenJDK to be used by the users:
https://ci.adoptopenjdk.net/view/ev3dev/job/openjdk12_build_ev3_linux/

It is necessary to review the testing for that build and do some regresssion tests with it.

Juan Antonio

JakubVanek, 2019-03-15:
Debian package from the [openjdk12_build_ev3_linux#78](https://ci.adoptopenjdk.net/view/ev3dev/job/openjdk12_build_ev3_linux/78/) build: [jri-12-ev3_12.0.0~33-1_armel.deb](https://drive.google.com/open?id=1POtEEQnr7ye820kZOTR_VQiPiqso1BX_)

I only had to modify the main mainfest to change the version string from `12` to `12.0.0`, as the parsing method was developed around `11.0.1` version string.

jabrena, 2019-08-18:
Late

closed 2019-08-18 by jabrena