ev3dev-lang-java/ev3dev-lang-java #657: java.lang.NoClassDefFoundError

lucas3600, 2019-03-20:
Hello everyone,
I’ve been trying for several days to run the example : FractalTree on my robot. 
But the problem is that I get the following error when I entered the command gradlew deployRun.
Error: Unable to initialize main class example.programming.FractalTree
Caused by: java.lang.NoClassDefFoundError: Lejos/hardware/lcd/GraphicsLCD.
I don’t know what can cause the problem because I followed the tutorial to the letter.
I tried with the example MyFirstRobot and it's the same error for class EV3LargeRegulatedMotor.
Below: build.gradle.

[build.gradke.txt](https://github.com/ev3dev-lang-java/ev3dev-lang-java/files/2987678/build.gradke.txt)

Thank a lot

JakubVanek, 2019-03-20:
Hi,

could you please run this command?
```sh
./gradlew uploadGradleLibraries
```

I think the ev3dev-lang-java libraries are not uploaded to the brick. This causes the issue with the missing libraries - Java doesn't fail when a library on classpath is missing. The command above will look up all dependencies and upload their JARs to the library directory.

There is also another way to install the basic libraries:
```sh
./gradlew getInstaller installJavaLibraries
```

Regards,

Jakub

lucas3600, 2019-03-21:
Thank you so much Jakub.
It's working !

closed 2019-03-21 by lucas3600